The-OpenROAD-Project/OpenROAD #3055: [ERROR PPL-0013] Internal error, placed more pins than exist (844 out of 583).

oharboe, 2023-03-21:
### Describe the bug

When I use the new -group and -order set_io_constrains feature, I get the error above.

I believe that it is always a bug if the above error is reported, so just a heads up that there is an issue here, even if I don't have a test-case I can share yet.

### Expected Behavior

No error

### Environment

```shell
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.0-67-generic
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
cmake version 3.22.1
```

### To Reproduce

I think this can be reproduced by updating designs/asap7/mock-array-big to use -order and -group

oharboe, 2023-03-21:
@eder-matheus @maliberty I was able to reproduce it, see build log for https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/905

eder-matheus, 2023-03-24:
@oharboe Sorry for the delay. I was having trouble to reproduce it on my Mac. Something is weird with the `make issue` for this new design.

Anyway, this PR should fix the issue: https://github.com/The-OpenROAD-Project/OpenROAD/pull/3075
Could you check with your internal testcase?

maliberty, 2023-03-24:
There was a bug in make issue that I fixed this week.

closed 2023-03-24 by maliberty