The-OpenROAD-Project/OpenROAD #3061: Compile the Standalone OpenROAD ending up with errors

JiaDYuan, 2023-03-22:
### Describe the bug

When compiling  OpenROAD, it ended up with errors after installing essential dependencies and running make command.  It is almost finished (97%).

collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/openroad.dir/build.make:410: src/openroad] Error 1
make[2]: Leaving directory '/home/yuanjiadong/OpenROAD/build'
make[1]: *** [CMakeFiles/Makefile2:1666: src/CMakeFiles/openroad.dir/all] Error 2
make[1]: Leaving directory '/home/yuanjiadong/OpenROAD/build'
make: *** [Makefile:149: all] Error 2

Here is what was printed after I run  cmake .. -DCMAKE_INSTALL_PREFIX=/home/yuanjiadong/openroad:

-- OpenROAD version: v2.0-7328-g3dc4848b3
-- System name: Linux
-- Compiler: GNU 10.2.1
-- Build type: RELEASE
-- Install prefix: /home/yuanjiadong/openroad
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Could NOT find GTest (missing: GTEST_LIBRARY GTEST_INCLUDE_DIR GTEST_MAIN_LIBRARY)
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- TCL readline library: /usr/lib/x86_64-linux-gnu/libtclreadline.so
-- TCL readline header: /usr/include/x86_64-linux-gnu
-- Found Boost: /home/yuanjiadong/openroad/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")
-- boost: 1.80.0
-- spdlog: 1.8.1
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
-- STA version: 2.4.0
-- STA git sha: 20805c83cf4817f6d2ca7867a52dac226515a21e
-- System name: Linux
-- Compiler: GNU 10.2.1
-- Build type: RELEASE
-- Build CXX_FLAGS: -O3 -DNDEBUG
-- Install prefix: /home/yuanjiadong/openroad
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- SSTA: 0
-- STA executable: /home/yuanjiadong/OpenROAD/src/sta/app/sta
-- GPU is not enabled
-- GUI is enabled
-- Found Boost: /home/yuanjiadong/openroad/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS)
-- Found Boost: /home/yuanjiadong/openroad/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")
-- Found Boost: /home/yuanjiadong/openroad/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
CMake Warning at src/CMakeLists.txt:269 (add_executable):
  Cannot generate a safe runtime search path for target openroad because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libtcl8.6.so] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/yuanjiadong/mambaforge/lib
    runtime library [libz.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/yuanjiadong/mambaforge/lib
    runtime library [libgomp.so.1] in /usr/lib/gcc/x86_64-linux-gnu/10 may be hidden by files in:
      /home/yuanjiadong/mambaforge/lib

  Some of these libraries may not be found correctly.

CMake Warning at src/odb/src/swig/python/CMakeLists.txt:28 (add_executable):
  Cannot generate a safe runtime search path for target odb_py-bin because
  files in some directories may conflict with libraries in implicit
  directories:

[... 202 lines cut ...]
  in some directories may conflict with libraries in implicit directories:

    runtime library [libtcl8.6.so] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/yuanjiadong/mambaforge/lib
    runtime library [libgomp.so.1] in /usr/lib/gcc/x86_64-linux-gnu/10 may be hidden by files in:
      /home/yuanjiadong/mambaforge/lib
    runtime library [libz.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/yuanjiadong/mambaforge/lib

  Some of these libraries may not be found correctly.

nents: serialization system thread
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.vAkC7YP8nL
```

### To Reproduce

[CMakeError.log](https://github.com/The-OpenROAD-Project/OpenROAD/files/11041072/CMakeError.log)
[CMakeOutput.log](https://github.com/The-OpenROAD-Project/OpenROAD/files/11041073/CMakeOutput.log)
[openroad-env.log](https://github.com/The-OpenROAD-Project/OpenROAD/files/11041074/openroad-env.log)

vvbandeira, 2023-03-22:
@JiaDYuan looks like you are using an env manager (i.e., `mambaforge`). Could you disable it and compile it again with:
```
./etc/Build.sh -clean
```

JiaDYuan, 2023-03-22:
/usr/bin/ld: /home/yuanjiadong/mambaforge/lib/libQt5Core.so.5.15.6: undefined reference to `std::__exception_ptr::exception_ptr::_M_addref()@CXXABI_1.3.13'
collect2: error: ld returned 1 exit status
gmake[2]: *** [src/CMakeFiles/openroad.dir/build.make:410: src/openroad] Error 1
gmake[2]: Leaving directory '/home/yuanjiadong/OpenROAD/build'
gmake[1]: *** [CMakeFiles/Makefile2:1666: src/CMakeFiles/openroad.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
[100%] Linking CXX executable trTest
cd /home/yuanjiadong/OpenROAD/build/src/drt && /home/yuanjiadong/openroad/bin/cmake -E cmake_link_script CMakeFiles/trTest.dir/link.txt --verbose=1
/usr/bin/g++ -O3 -DNDEBUG CMakeFiles/trTest.dir/test/gcTest.cpp.o CMakeFiles/trTest.dir/test/fixture.cpp.o CMakeFiles/trTest.dir/test/stubs.cpp.o CMakeFiles/trTest.dir/__/gui/src/stub.cpp.o -o trTest  -Wl,-rpath,/home/yuanjiadong/mambaforge/lib drt.a ../stt/stt.a ../dst/dst.a /home/yuanjiadong/openroad/lib/libboost_serialization.a /home/yuanjiadong/openroad/lib/libboost_system.a /home/yuanjiadong/openroad/lib/libboost_thread.a ../dbSta/src/dbSta.a ../gui/gui.a ../odb/src/cdl/libcdl.a ../odb/src/defin/libdefin.a ../odb/src/def/libdef.a ../odb/src/def/libdefzlib.a ../odb/src/lefin/liblefin.a ../odb/src/lef/liblef.a ../odb/src/lef/liblefzlib.a ../odb/src/db/libdb.a ../odb/src/defout/libdefout.a ../odb/src/lefout/liblefout.a ../odb/src/zutil/libzutil.a ../odb/src/db/libdb.a ../odb/src/defout/libdefout.a ../odb/src/lefout/liblefout.a ../odb/src/zutil/libzutil.a ../odb/src/tm/libtm.a /usr/lib/x86_64-linux-gnu/libtcl.so /home/yuanjiadong/mambaforge/lib/libQt5Widgets.so.5.15.6 /home/yuanjiadong/mambaforge/lib/libQt5Gui.so.5.15.6 /home/yuanjiadong/mambaforge/lib/libQt5Core.so.5.15.6 /home/yuanjiadong/OpenROAD/src/sta/app/libOpenSTA.a ../utl/utl.a /home/yuanjiadong/openroad/lib/libspdlog.a -lpthread /usr/lib/gcc/x86_64-linux-gnu/10/libgomp.so /usr/lib/x86_64-linux-gnu/libpthread.so /home/yuanjiadong/openroad/lib/libboost_serialization.a /usr/lib/x86_64-linux-gnu/libz.so
gmake[2]: Leaving directory '/home/yuanjiadong/OpenROAD/build'
[100%] Built target trTest
gmake[1]: Leaving directory '/home/yuanjiadong/OpenROAD/build'
gmake: *** [Makefile:149: all] Error 2

It ended up with the same errors