The-OpenROAD-Project/OpenROAD #3067: Wierd connection with sroute command after setting the ndr rule.

Wen-Tian-Pineapple, 2023-03-23:
### Description

This problem is related to #[191] (https://github.com/idea-fasoc/OpenFASOC/issues/191). When using sroute and the ndr rule below to connect the ring oscillator with the power ring. We see some weird routing wires. Extra figure are showing below to describe the issue.

-------------------------ndr rule-------------------------------
create_ndr -name NDR_5W_5S \
           -spacing { *4 } \
           -width { *4 }
set ndr [$block findNonDefaultRule NDR_5W_5S]
$ndr setHardSpacing 1
------------------------------------------------------------------

In the figure below, you can see there is couple weird routing behaviors. During the connection from cell to power ring, there is sudden change of wire width even though at the same layer. Also at some of the via cut, there is extra wire added which is way more than enough.
[image]

### Suggested Solution

Maybe the detailed router can be changed to adapt to the sroute and the ndr rule. This problem might be cause by the incorrect setting of parameters in ndr rule functions.

maliberty, 2023-03-23:
Please provide a test case (preferably on a public PDK)

Wen-Tian-Pineapple, 2023-03-27:
[global_route.txt](https://github.com/The-OpenROAD-Project/OpenROAD/files/11073845/global_route.txt)

In OpenRoad Flow script, I changed global_route.tcl to the same as the file above (can't attach tcl file so I attached the txt) and run the design of "./designs/nangate45/gcd/config.mk" (to run the make file faster, you can set the wire width/spacing smaller)

The weird connection is in the figure below(on net 36):
[image]
[image]

maliberty, 2023-03-27:
From the picture it looks like the wire is connecting to that standard cell pin.  The standard cell pins are very dense and using a wide wire would block access to the other pins.  So we automatically apply tapering to the wire when you approach them.

I'm guessing for your sroute purposes you don't want tapering but I'm not certain.

michahn01, 2023-04-17:
> From the picture it looks like the wire is connecting to that standard cell pin. The standard cell pins are very dense and using a wide wire would block access to the other pins. So we automatically [...]

I'm looking at this issue right now, and I'm wondering how I might disable tapering for sroute (and sroute only). Is there a function somewhere in a file in OpenROAD's src directory that controls this behavior? I apologize in advance if this question doesn't make sense.

maliberty, 2025-08-28:
no updates