The-OpenROAD-Project/OpenROAD #3069: Slow generate_abstract

oharboe, 2023-03-23:
### Description

I have a macro that takes "Elapsed time: 2:47:31[h:]min:sec.",  nearly 3 hours to generate abstract. I was expecting minutes to do this. The .lib and .lef are 21 and 97 mbyte respectively.

I can't share the design in question, but I will look into creating a test-case based on mock-array-big.

### Suggested Solution

Improve performance.

tspyrou, 2023-03-23:
Is it the .lib of the .lef generation that is taking the time?

oharboe, 2023-03-28:
For a 2 minute generate mock abstract, we get the following perf report:

```
   4,79%  openroad  openroad                [.] odb::_dbObject::getOID                                    ◆
   4,55%  openroad  openroad                [.] odb::_dbObject::getType                                   ▒
   3,09%  openroad  openroad                [.] sta::dbNetwork::staToDb                                   ▒
   2,01%  openroad  openroad                [.] odb::dbTable<odb::_dbInst>::getPtr                        ▒
   1,85%  openroad  openroad                [.] odb::_dbObject::getOwner                                  ▒
   1,60%  openroad  openroad                [.] odb::dbTable<odb::_dbITerm>::getPtr                       ▒
   1,17%  openroad  openroad                [.] sta::NetworkNameAdapter::id                               ▒
   1,13%  openroad  openroad                [.] sta::VisitPathEnds::visitCheckEnd                         ▒
   1,11%  openroad  openroad                [.] std::_Identity<sta::Pin const*>::operator()               ▒
   1,06%  openroad  openroad                [.] odb::dbTable<odb::_dbMaster>::getPtr                      ▒
   1,05%  openroad  openroad                [.] odb::dbId<odb::_dbITerm>::operator unsigned int           ▒
   1,03%  openroad  openroad                [.] odb::dbTable<odb::_dbLib>::getPtr                         ▒
   1,03%  openroad  openroad                [.] odb::dbITerm::getMTerm                                    ▒
   1,01%  openroad  openroad                [.] odb::dbObject::getImpl                                    ▒
   1,00%  openroad  openroad                [.] sta::PinIdLess::operator()                                ▒
   0,99%  openroad  openroad                [.] odb::dbTable<odb::_dbMTerm>::getPtr                       ▒
   0,97%  openroad  openroad                [.] sta::dbNetwork::id                                        ▒
   0,94%  openroad  openroad                [.] odb::dbTable<odb::_dbInstHdr>::getPtr                     ▒
   0,94%  openroad  openroad                [.] odb::dbSet<odb::dbBTerm>::size                            ▒
   0,93%  openroad  openroad                [.] odb::dbId<odb::_dbInst>::operator unsigned int            ▒
   0,84%  openroad  openroad                [.] sta::VertexIterator::findNextPin                          ▒
   0,81%  openroad  openroad                [.] std::_Rb_tree<sta::Pin const*, sta::Pin const*, std::_Iden▒
   0,74%  openroad  openroad                [.] sta::VertexPathIterator::findNext                         ▒
   0,74%  openroad  openroad                [.] sta::VisitPathEnds::visitClkedPathEnds                    ▒
   0,74%  openroad  libc.so.6               [.] ____wcstof_l_internal                                     ▒
   0,69%  openroad  openroad                [.] odb::dbInstITermItr::next     
```

oharboe, 2023-03-28:
5 minute run:

```
   4,61%  openroad  openroad                 [.] odb::_dbObject::getType
   4,26%  openroad  openroad                 [.] odb::_dbObject::getOID
   3,10%  openroad  openroad                 [.] sta::dbNetwork::staToDb
   2,03%  openroad  openroad                 [.] odb::dbTable<odb::_dbInst>::getPtr
   1,70%  openroad  openroad                 [.] odb::_dbObject::getOwner
   1,46%  openroad  openroad                 [.] odb::dbTable<odb::_dbITerm>::getPtr
   1,18%  openroad  openroad                 [.] odb::dbTable<odb::_dbMaster>::getPtr
   1,18%  openroad  openroad                 [.] odb::dbTable<odb::_dbLib>::getPtr
   1,13%  openroad  openroad                 [.] odb::dbTable<odb::_dbMTerm>::getPtr
   1,12%  openroad  openroad                 [.] odb::dbId<odb::_dbITerm>::operator unsigned int
   1,11%  openroad  openroad                 [.] odb::dbITerm::getMTerm
   1,09%  openroad  openroad                 [.] sta::VertexIterator::findNextPin
   1,02%  openroad  openroad                 [.] odb::dbTable<odb::_dbInstHdr>::getPtr
   0,98%  openroad  openroad                 [.] odb::dbObject::getImpl
   0,94%  openroad  openroad                 [.] sta::NetworkNameAdapter::id
   0,89%  openroad  openroad                 [.] sta::dbNetwork::id
   0,88%  openroad  openroad                 [.] odb::dbId<odb::_dbInst>::operator unsigned int
   0,87%  openroad  openroad                 [.] sta::PinIdLess::operator()
```

oharboe, 2023-03-28:
From a 3hr perf run (stopped with pkill -SIGINT openroad):

```
   8,10%  openroad  openroad                [.] odb::_dbObject::getOID
   7,94%  openroad  openroad                [.] odb::_dbObject::getType
   5,01%  openroad  openroad                [.] sta::dbNetwork::staToDb
   2,06%  openroad  openroad                [.] std::_Rb_tree<sta::Pin const*, sta::Pin const*, std::_Identity<sta::Pin const*>, sta::PinIdLess, std::allocator<sta::Pin const*> >::_M_
   1,91%  openroad  openroad                [.] std::_Identity<sta::Pin const*>::operator()
   1,86%  openroad  openroad                [.] odb::dbSet<odb::dbBTerm>::size
   1,84%  openroad  openroad                [.] sta::VisitPathEnds::visitCheckEnd
   1,81%  openroad  openroad                [.] sta::NetworkNameAdapter::id
   1,74%  openroad  openroad                [.] sta::dbNetwork::id
   1,55%  openroad  openroad                [.] sta::PinIdLess::operator()
   1,54%  openroad  openroad                [.] odb::dbObject::getImpl
   1,28%  openroad  openroad                [.] sta::VertexPathIterator::findNext
   1,27%  openroad  openroad                [.] odb::_dbObject::getOwner
   1,14%  openroad  openroad                [.] odb::dbTable<odb::_dbInst>::getPtr
   0,98%  openroad  openroad                [.] sta::VisitPathEnds::visitClkedPathEnds
   0,94%  openroad  openroad                [.] sta::UnorderedMap<sta::Tag*, int, sta::TagMatchHash, sta::TagMatchEqual>::Iterator::next
   0,77%  openroad  openroad                [.] odb::dbTable<odb::_dbITerm>::getPtr
   0,75%  openroad  openroad                [.] __gnu_cxx::__aligned_membuf<sta::Pin const*>::_M_addr
   0,71%  openroad  openroad                [.] sta::Search::exceptionTo
   0,70%  openroad  openroad                [.] odb::dbTable<odb::_dbMTerm>::getPtr
   0,69%  openroad  openroad                [.] odb::dbObject::getObjectType
   0,68%  openroad  openroad                [.] odb::dbTable<odb::_dbLib>::getPtr
   0,66%  openroad  openroad                [.] odb::dbTable<odb::_dbMaster>::getPtr
   0,65%  openroad  openroad                [.] sta::ObjectTable<sta::Vertex>::pointer
   0,64%  openroad  openroad                [.] odb::dbITerm::getMTerm
   0,63%  openroad  openroad                [.] sta::TimingArcSet::role
   0,63%  openroad  openroad                [.] std::_Rb_tree<sta::Pin const*, sta::Pin const*, std::_Identity<sta::Pin const*>, sta::PinIdLess, std::allocator<sta::Pin const*> >::_S_
   0,61%  openroad  openroad                [.] odb::dbBlock::getBTerms
   0,60%  openroad  openroad                [.] sta::PathVertex::tag
   0,59%  openroad  openroad                [.] sta::VertexInEdgeIterator::next
   0,57%  openroad  openroad                [.] sta::UnorderedMap<sta::Tag*, int, sta::TagMatchHash, sta::TagMatchEqual>::Iterator::hasNext
   0,55%  openroad  openroad                [.] odb::dbTable<odb::_dbInstHdr>::getPtr
   0,54%  openroad  openroad                [.] odb::dbId<odb::_dbITerm>::operator unsigned int
   0,52%  openroad  openroad                [.] sta::RiseFall::index
   0,51%  openroad  openroad                [.] sta::ObjectTable<sta::Edge>::pointer
   0,50%  openroad  openroad                [.] sta::VertexIterator::findNextPin
   0,46%  openroad  openroad                [.] odb::dbId<odb::_dbInst>::operator unsigned int
   0,46%  openroad  openroad                [.] std::__detail::operator!=
   0,43%  openroad  openroad                [.] std::vector<sta::TableBlock<sta::Vertex>*, std::allocator<sta::TableBlock<sta::Vertex>*> >::operator[]
   0,43%  openroad  openroad                [.] sta::GatedClk::isGatedClkEnable
   0,43%  openroad  openroad                [.] sta::VertexInEdgeIterator::hasNext
   0,40%  openroad  openroad                [.] std::__detail::_Node_iterator<std::pair<sta::Tag* const, int>, false, true>::operator->
   0,39%  openroad  openroad                [.] sta::VisitPathEnds::visitDataCheckEnd
   0,38%  openroad  openroad                [.] std::__detail::_Hash_node<std::pair<sta::Tag* const, int>, true>::_M_next
   0,38%  openroad  libstdc++.so.6.0.30     [.] std::_Rb_tree_increment
   0,38%  openroad  openroad                [.] sta::VertexPathIterator::hasNext
   0,37%  openroad  openroad                [.] __gnu_cxx::__aligned_membuf<sta::Pin const*>::_M_ptr
   0,37%  openroad  openroad                [.] std::_Rb_tree<sta::Pin const*, sta::Pin const*, std::_Identity<sta::Pin const*>, sta::PinIdLess, std::allocator<sta::Pin const*> >::end
   0,36%  openroad  openroad                [.] std::_Rb_tree<char const*, std::pair<char const* const, sta::ConcretePort*>, std::_Select1st<std::pair<char const* const, sta::Concrete
   0,35%  openroad  openroad                [.] sta::Sdc::exceptionTo
   0,35%  openroad  openroad                [.] sta::VisitPathEnds::checkEdgeEnabled
   0,34%  openroad  openroad                [.] std::_Rb_tree<sta::Pin const*, sta::Pin const*, std::_Identity<sta::Pin const*>, sta::PinIdLess, std::allocator<sta::Pin const*> >::_S_
   0,34%  openroad  openroad                [.] sta::VisitPathEnds::visitGatedClkEnd
   0,34%  openroad  openroad                [.] std::__detail::_Node_iterator<std::pair<sta::Tag* const, int>, false, true>::_Node_iterator
   0,34%  openroad  openroad                [.] odb::dbTable<odb::_dbBTerm>::size
   0,33%  openroad  openroad                [.] std::__detail::_Hash_node_value_base<std::pair<sta::Tag* const, int> >::_M_valptr
[... cut ...]
   0,32%  openroad  openroad                [.] std::_Rb_tree_const_iterator<sta::Pin const*>::_Rb_tree_const_iterator
   0,32%  openroad  openroad                [.] sta::TimingRole::isTimingCheck
   0,32%  openroad  openroad                [.] odb::_dbObject::getDatabase
   0,32%  openroad  openroad                [.] odb::dbObject::getId
   0,31%  openroad  openroad                [.] sta::TimingArc::fromEdge
   0,31%  openroad  openroad                [.] odb::dbId<odb::_dbITerm>::dbId

```

oharboe, 2023-03-31:
Duplicate of https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/940

closed 2023-03-31 by oharboe

jjcherry56, 2023-04-10:
If the testcase runs quickly it does not reproduce the problem.