The-OpenROAD-Project/OpenROAD #3070: Modify default behavior of etc/DependencyInstaller.sh not to modify /usr/local or ~/.local

oharboe, 2023-03-23:
### Description

Modifying /usr/local and ~/.local makes it convenient to compile OpenROAD, but it can damage the system w.r.t. conflicts with e.g. Ubuntu 22.04 and 22.10 behavior and it is difficult to undo the changes to /usr/local and ~/.local

### Suggested Solution

Modify the default behavior of etc/DependencyInstaller.sh to create a folder e.g. ~/openroad-dev and install tools there. Add a "etc/env.sh" script that sets up env vars to point to ~/openroad-dev

Installing .deb and .rpm packages should be fine. How to uninstall them is well understood.

This behavior is reasonably easy to use and safe.

rovinski, 2023-03-23:
Is there a problem with using `etc/DependencyInstaller.sh -prefix=~/openroad-dev`?

oharboe, 2023-03-23:
> Is there a problem with using `etc/DependencyInstaller.sh -prefix=~/openroad-dev`?

Not as long as "sudo etc/DependencyInstaller.sh" doesnt modify /usr/local and ~/.local

rovinski, 2023-03-23:
It shouldn't - if it does, then that's worthy of a bug report.

oharboe, 2023-03-24:
> It shouldn't - if it does, then that's worthy of a bug report.

That would be this github issue.

rovinski, 2023-03-24:
Can you be more specific about which files are being modified or created in /usr/local or ~/.local?

oharboe, 2023-03-24:
> Can you be more specific about which files are being modified or created in /usr/local or ~/.local?

I was told that it does. I didnt want to run the script to find out, because it messed up my machine. I will follow up next week.

oharboe, 2023-03-26:
With latest OpenROAD:

```
$ which cmake
/usr/bin/cmake
$ sudo etc/DependencyInstaller.sh

```

rovinski, 2023-03-28:
@oharboe the problem is that you are using `sudo` to install. This means you are installing to system directories, and this is expected behavior. If you run with `etc/DependencyInstaller.sh -prefix=DIR` then it will install to `DIR` and not override any of your PATH.

oharboe, 2023-03-28:
> @oharboe the problem is that you are using `sudo` to install. This means you are installing to system directories, and this is expected behavior. If you run with `etc/DependencyInstaller.sh -prefix=DIR` then it will install to `DIR` and not override any of your PATH.

I have done both: if you run DependencyInstaller.sh wo arguments, you must prefix sudo as it installs to /usr/local.

I think the solution to this usability issue is to always add sudo and then drop down to the user when installing user bits. That way the user can never do anything wrong: always add sudo.

rovinski, 2023-03-28:
I disagree, many users don't have sudo access on their machine. I think it's fully reasonable to do:
`sudo DependencyInstaller.sh` --> Install to `/usr/local`
`DependencyInstaller.sh -local` --> Install to `$HOME/.local`
`DependencyInstaller.sh -prefix=DIR` --> Install to `DIR`

I think `DependencyInstaller.sh -help` is reasonably clear about this. Maybe the script could error if not invoked with sudo or another option, I suppose it doesn't do that right now.

oharboe, 2023-04-22:
moot after sudo ./setup.sh was introduced. DependencyInstaller.sh is now an implementation detail used by setup.sh and docker images and the API is not critical as users are not exposed to it.

closed 2023-04-22 by oharboe