The-OpenROAD-Project/OpenROAD #3071: Multi-cut via is not suppurted in detail router yet.

Wen-Tian-Pineapple, 2023-03-23:
### Description

As stated in title, currently OpenROAD doesn't support multi-cut vias which could be a crucial functionality as shown in the figure below.(In FlewPR.cpp) So I believe it's impossible to set constraint in ndr rule for multi-cut vias since it's basically not support in the OpenROAD router.

[image]

### Suggested Solution

Further modification need to be done for the router or create a new router just for multi-cut vias.

rovinski, 2023-03-23:
This is a known limitation to the detailed router and it has been discussed before. It would take some work to implement proper via selection, but we haven't had anyone to commit significant work on this.

I think @antonblanchard may be looking at better via selection such as in #2940 and #3021 but I think it was mostly geared towards single-cut vias. PDN definitely has support for multi-cut vias, but that's completely separate code.

The main issue is that there are some parts of DRT which are hard-coded to assume single-cut vias, and gc is not set up to check for multi-cut rules. dr may need adjustments to the costing to use multi-cut vias as well.

If there is someone to someone to look at these issues, then DRT could have better support for multi-cut vias.

rovinski, 2023-03-23:
I had a conversation with someone on the opensilicon Slack about how to force DRT to use double-cut vias. What they ended up doing was making a single-cut via with the same footprint as the double-cut via, then after DRT ran, they replaced the single-cut via with the double-cut via and checked for DRC errors. That process seemed to work for them, but largely because their PDK had uncomplicated via spacing rules. In more advanced PDKs, this process may not work.

maliberty, 2023-03-23:
For this use model I expect the user will give the vias they want the router to use in the form of an NDR.  It will require enhancing the router to support the rules related to such vias but via selection itself will be a user responsibility (at least for now).   At least gc will have to be smarter if not dr.

Wen-Tian-Pineapple, 2023-03-29:
> 

I think the current NDR rule doesn't really support user to specify custom vias for each layer. I couldn't find any exmaple in OpenROAD and I have tried the via name in tlef file. But it doesn't work. I'm not sure if I'm using it in the right way tho:

create_ndr -name NDR_5W_5S \
           -spacing { *3 } \
           -width { *3 } \
           -via { L1M1_PR_C M1M2_PR M2M3_PR M3M4_PR M3M5_PR }

So in the via input, I just put in all the vias use for connection consecutively.

gkamendje, 2023-10-31:
@maliberty I was wondering if `create_ndr` is documented somewhere. I could not find any mention of it in the docs. Would you please shed some light on how to use it? My goal is to be able to insert at least double cut vias during routing. Any example or pointer to documentation would be welcome. 
  
> For this use model I expect the user will give the vias they want the router to use in the form of an NDR. It will require enhancing the router to support the rules related to such vias but via [...]

maliberty, 2023-10-31:
We only support width and spacing ndrs currently.

gkamendje, 2023-10-31:
I was under the impression when reading #3250, that support for custom vias was added... Is my understanding wrong?
> We only support width and spacing ndrs currently.

maliberty, 2023-10-31:
@eder-matheus please comment on the via support in NDRs.

eder-matheus, 2023-10-31:
@gkamendje The mentioned PR added the ability to specify custom vias in an NDR and assign it to specific nets. You can see an example of how to use it here: https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/drt/test/ndr_vias2.tcl#L9-L19.

However, as @maliberty commented, the detailed router still needs to get official support for multi-cut vias, so you may find issues when using it.

gkamendje, 2023-10-31:
@eder-matheus thanks for the hint. I am using the ORFS. What would be the best place to create the NDRs? Should they be created in `pre_global_route.tcl` file?  
> @gkamendje The mentioned PR added the ability to specify custom vias in an NDR and assign it to specific nets. You can see an example of how to use it here: [...]

eder-matheus, 2023-10-31:
> @eder-matheus thanks for the hint. I am using the ORFS. What would be the best place to create the NDRs? Should they be created in `pre_global_route.tcl` file?

@gkamendje Adding it to the `pre_global_route.tcl` file should be fine. Just remember to make sure that the env var `PRE_GLOBAL_ROUTE` is correctly defined with your script.

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/scripts/global_route.tcl#L5-L7

gkamendje, 2023-10-31:
Yep makes sense. One last question. I have seen the following syntax for `create_ndr` 
```
-spacing { *3}
-width { *3}
```
Does this means that in this case the default width and spacing rules are multiplied by 3 for all layers? 

> > @eder-matheus thanks for the hint. I am using the ORFS. What would be the best place to create the NDRs? Should they be created in `pre_global_route.tcl` file?
> 
> @gkamendje Adding it to the `pre_global_route.tcl` file should be fine. Just remember to make sure that the env var `PRE_GLOBAL_ROUTE` is correctly defined with your script.
> 
> https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/scripts/global_route.tcl#L5-L7

eder-matheus, 2023-10-31:
> Yep makes sense. One last question. I have seen the following syntax for `create_ndr`
> 
> ```
> -spacing { *3}
> -width { *3}
> ```
> 
> Does this means that in this case the default width and spacing rules are multiplied by 3 for all layers?
> 

Yes, though I believe you don't need it to add that into your NDRs. AFAIK, you can just set the vias.

sebinho, 2024-04-22:
Hi guys,

I am following up on the topic regarding multi-cut vias. This is indeed quite typical to use double-cut vias instead of single ones to improve yield.
This is what we have been doing in the past using commercial tools, and when not possible reverting back to single vias.

Is there any progress on this since the last exchange?
I believe this is indeed a much needed feature, but correct me if I am wrong.

Thanks for your help

maliberty, 2024-04-22:
There hasn't been progress on this.  It is an important feature for high volume production where yield is important.  DRC rule completeness has been a higher priority for our limited routing team resources.  We are always open to community contributions.

maliberty, 2025-03-25:
They are no supported so there is no reason to expect it to work like they are supported...

chrisgelinek-rl, 2025-03-25:
Sorry for the confusion, I was referring to [this comment](https://github.com/The-OpenROAD-Project/OpenROAD/issues/3071#issuecomment-1787530860) which I understood to mean that there is in fact some

> ability to specify custom vias in an NDR and assign it to specific nets

and that's why I was assuming (in context) the [testcase](https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/drt/test/ndr_vias2.tcl#L9-L19) creates NDRs with multi-cut vias, but it seems I was wrong. I do see those via names mentioned in the NDR being used in the layout, which is somewhat promising.

I'll see if I can get some time to create a variant of that same testcase where the NDR is actually trying to use double-vias to experiment, or would that be wasted effort?

maliberty, 2025-03-25:
Double cut vias often come with specific DRC rules that OR doesn't support.  If in your technology there is no such issue then theoretically it might work to use an NDR though I don't think we've ever tried it.  If you want to make a simple one-net test case I think we could investigate.