The-OpenROAD-Project/OpenROAD #3077: -group and -order on pin constraints doesn't order the pins

oharboe, 2023-03-24:
### Describe the bug

I see that the pins are not in order when I use -group and -order

### Expected Behavior

Should be grouped and ordered

### Environment

```shell
Git commit: e18c0d570151a95ebc389254d8baa0eb3cb57f00
kernel: Linux 5.19.0-29-generic
os: Ubuntu 22.10 (Kinetic Kudu)
cmake version 3.24.2
```

### To Reproduce

I have seen this in a private repository, but it should be reproducible here: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/905

oharboe, 2023-03-26:
Trying on private project, assert...

```
Using 2 tracks default min distance between IO pins.
/opt/rh/devtoolset-8/root/usr/include/c++/8/bits/stl_vector.h:932: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = std::vector<int>; _Alloc = std::allocator<std::vector<int> >; std::vector<_Tp, _Alloc>::reference = std::vector<int>&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
Signal 6 received
Stack trace:
 0# 0x0000000000CDF647 in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x00007FBB2F1D0400 in /lib64/libc.so.6
 2# gsignal in /lib64/libc.so.6
 3# abort in /lib64/libc.so.6
 4# 0x0000000000CDF722 in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 5# ppl::HungarianMatching::createMatrix() in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 6# ppl::HungarianMatching::findAssignment() in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 7# ppl::IOPlacer::findPinAssignment(std::vector<ppl::Section, std::allocator<ppl::Section> >&) in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 8# ppl::IOPlacer::run(bool) in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 9# 0x0000000000E563BD in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
```

oharboe, 2023-03-26:
The assert above made me file this suggestion: https://github.com/The-OpenROAD-Project/OpenROAD/issues/3085

eder-matheus, 2023-03-27:
@oharboe About the groups not being ordered, how many pins do you have in this group? Are they also constrained to a specific region?

oharboe, 2023-03-27:
> @oharboe About the groups not being ordered, how many pins do you have in this group? Are they also constrained to a specific region?

The setup is very similar to mock-array-big, but with 64 bit datapaths.

The next step is to bottom out on the assert above, requires a full debug setup locally, to see if this is actually working and this is just an error in the .tcl.

oharboe, 2023-03-28:
I tried to get some more debug information: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/discussions/927

oharboe, 2023-03-28:
Segfault fix: https://github.com/The-OpenROAD-Project/OpenROAD/pull/3100

oharboe, 2023-06-04:
@eder-matheus Close?

closed 2023-06-05 by eder-matheus