The-OpenROAD-Project/OpenROAD #3085: Change assert exit to exceptions to get better stack-traces

oharboe, 2023-03-26:
### Description

Exceptions are more informative than asserts. I seem to recall that there's special code to give line numbers for a stacktrace of an exception in OpenROAD?

Consider the example alert, not very informative, I can't look at the calling code location.

Another problem is that if you have an open file, then the partially written file is not flushed to disk, whereas exception cleanup code will close the file so the partially written file can be examined.

```
Using 2 tracks default min distance between IO pins.
/opt/rh/devtoolset-8/root/usr/include/c++/8/bits/stl_vector.h:932: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = std::vector<int>; _Alloc = std::allocator<std::vector<int> >; std::vector<_Tp, _Alloc>::reference = std::vector<int>&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
Signal 6 received
Stack trace:
 0# 0x0000000000CDF647 in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x00007FBB2F1D0400 in /lib64/libc.so.6
 2# gsignal in /lib64/libc.so.6
 3# abort in /lib64/libc.so.6
 4# 0x0000000000CDF722 in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 5# ppl::HungarianMatching::createMatrix() in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 6# ppl::HungarianMatching::findAssignment() in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 7# ppl::IOPlacer::findPinAssignment(std::vector<ppl::Section, std::allocator<ppl::Section> >&) in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 8# ppl::IOPlacer::run(bool) in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 9# 0x0000000000E563BD in /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
```

### Suggested Solution

```C++
extern "C" void ThrowExceptionOnAbort(int signal_number) {
  throw std::runtime_error("abort(" + std::to_string(signal_number) +
                           ") invoked");
}

/*Do this early in your program's initialization */

int main(int argc, char **argv) {
  try {
    signal(SIGABRT, &ThrowExceptionOnAbort);
...
```

maliberty, 2023-03-27:
I think line numbers are about whether you are compiled debug or not.  I don't think it relates to the signal.  We already have signal handlers in Main.cc

oharboe, 2023-03-28:
> I think line numbers are about whether you are compiled debug or not. I don't think it relates to the signal. We already have signal handlers in Main.cc

Makes sense.

Could the docker build be updated to have debug info? 

It would not increase the docker build size much, since it is already gigabytes...

maliberty, 2023-03-28:
Debug with optimized will give line numbers but not always the right ones.  Is that what you mean?

Debug without optimization is a lot slower to run so we wouldn't do that by default.

oharboe, 2023-03-28:
> Debug with optimized will give line numbers but not always the right ones. Is that what you mean?
> 
> Debug without optimization is a lot slower to run so we wouldn't do that by default.

Yes, optimized with line numbers. These line numbers are of course a "best guess".

This would also open up to using "perf" with the docker image.

maliberty, 2023-03-28:
@vvbandeira would you make this change to docker.  Using ```cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo ...``` should do it.

oharboe, 2023-04-22:
> @vvbandeira would you make this change to docker. Using `cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo ...` should do it.

Belay that. RelWithDebugInfo is slower, surprisingly. Use `cmake -DCMAKE_BUILD_TYPE=RELEASE -DCMAKE_CXX_FLAGS=-g`, it doesnt affect runtime peeformance.

maliberty, 2023-04-22:
I found:
```
CMAKE_BUILD_TYPE=Release            # defaults to -O3 -DNDEBUG
CMAKE_BUILD_TYPE=RelWithDebInfo     # defaults to -O2 -g
```
so there is some difference.

maliberty, 2023-04-22:
This idea is moot anyways as you can't throw an exception from a signal handler.

closed 2023-04-22 by maliberty