The-OpenROAD-Project/OpenROAD #3096: etc/DependencyInstaller.sh is using /tmp/installers which is shared between users

oharboe, 2023-03-27:
### Describe the bug

This causes problems because the access permissions will be different and /tmp/installers will fail for mysterious reasons.

The conflict can be between two normal users or between one user and root.

### Expected Behavior

Two independent users on the same machine should be able to install OpenROAD without trouble.

### Environment

```shell
latest master
```

### To Reproduce

Run etc/DepedencyInstaller.sh and see that a folder /tmp/installers is always created

rovinski, 2023-03-28:
Realistically the installer should be using `mktemp` instead of `mkdir` which guarantees that the created directory is unique.

titan73, 2024-08-16:
This as been fixed by https://github.com/The-OpenROAD-Project/OpenROAD/commit/6afff9846eb7b6533465cb5e3e1f0b7551b34be4.
The issue can be closed now.