The-OpenROAD-Project/OpenROAD #3101: No instance is found after Traversed logical hierarchy

LJK314, 2023-03-28:
### Describe the bug

I am running MPL2 and encountered a failure message of "Creating the pin blockage for root cluster". After seeing "Traversed logical hierarchy", all the number of instance show 0. I am unsure if the data was not loaded or if the failure is due to the absence of the lib file in ISPD 2015 which caused MPL2 to fail.

OpenROAD v2.0-7405-gcd9320ef9 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: /home/data/ispd15/mgc_fft_b/tech.lef
[WARNING ODB-0220] WARNING (LEFPARS-2007): NAMESCASESENSITIVE statement is obsolete in version 5.6 and later.
The LEF parser will ignore this statement.
To avoid this warning in the future, remove this statement from the LEF file with version 5.6 or later. See file /home/data/ispd15/mgc_fft_b/tech.lef at line 4.

[INFO ODB-0388] unsupported LEF57_MINSTEP property for layer metal1 :"MINSTEP 0.100 MAXEDGES 1 ;"
[INFO ODB-0388] unsupported LEF57_SPACING property for layer metal1 :"SPACING 0.11 ENDOFLINE 0.12 WITHIN 0.045 PARALLELEDGE 0.11 WITHIN 0.11 ;"
[INFO ODB-0388] unsupported LEF57_SPACING property for layer via1 :"SPACING 0.13 PARALLELOVERLAP ;"
[INFO ODB-0388] unsupported LEF57_MINSTEP property for layer metal2 :"MINSTEP 0.1 MAXEDGES 1 ;"
[INFO ODB-0388] unsupported LEF57_SPACING property for layer metal2 :"SPACING 0.12 ENDOFLINE 0.14 WITHIN 0.045 PARALLELEDGE 0.12 WITHIN 0.12 ;"
[INFO ODB-0388] unsupported LEF57_SPACING property for layer via2 :"SPACING 0.13 PARALLELOVERLAP ;"
[INFO ODB-0388] unsupported LEF57_MINSTEP property for layer metal3 :"MINSTEP 0.1 MAXEDGES 1 ;"
[INFO ODB-0388] unsupported LEF57_SPACING property for layer metal3 :"SPACING 0.12 ENDOFLINE 0.14 WITHIN 0.045 PARALLELEDGE 0.12 WITHIN 0.12 ;"
[INFO ODB-0388] unsupported LEF57_SPACING property for layer via3 :"SPACING 0.13 PARALLELOVERLAP ;"
[INFO ODB-0388] unsupported LEF57_MINSTEP property for layer metal4 :"MINSTEP 0.1 MAXEDGES 1 ;"
[INFO ODB-0388] unsupported LEF57_SPACING property for layer metal4 :"SPACING 0.12 ENDOFLINE 0.14 WITHIN 0.045 PARALLELEDGE 0.12 WITHIN 0.12  ;"
[INFO ODB-0388] unsupported LEF57_SPACING property for layer via4 :"SPACING 0.13 PARALLELOVERLAP ;"
[INFO ODB-0388] unsupported LEF57_MINSTEP property for layer metal5 :"MINSTEP 0.1 MAXEDGES 1 ;"
[INFO ODB-0388] unsupported LEF57_SPACING property for layer metal5 :"SPACING 0.12 ENDOFLINE 0.14 WITHIN 0.045 PARALLELEDGE 0.12 WITHIN 0.12  ;"
[WARNING ODB-0220] WARNING (LEFPARS-2019): TOPOFSTACKONLY statement is obsolete in version 5.6 and later.
The LEF parser will ignore this statement.
To avoid this warning in the future, remove this statement from the LEF file with version 5.6 or later See file /home/data/ispd15/mgc_fft_b/tech.lef at line 472.

[WARNING ODB-0220] WARNING (LEFPARS-2019): TOPOFSTACKONLY statement is obsolete in version 5.6 and later.
The LEF parser will ignore this statement.
To avoid this warning in the future, remove this statement from the LEF file with version 5.6 or later See file /home/data/ispd15/mgc_fft_b/tech.lef at line 482.

[WARNING ODB-0220] WARNING (LEFPARS-2019): TOPOFSTACKONLY statement is obsolete in version 5.6 and later.
The LEF parser will ignore this statement.
To avoid this warning in the future, remove this statement from the LEF file with version 5.6 or later See file /home/data/ispd15/mgc_fft_b/tech.lef at line 638.

[WARNING ODB-0220] WARNING (LEFPARS-2019): TOPOFSTACKONLY statement is obsolete in version 5.6 and later.
The LEF parser will ignore this statement.
To avoid this warning in the future, remove this statement from the LEF file with version 5.6 or later See file /home/data/ispd15/mgc_fft_b/tech.lef at line 648.

[WARNING ODB-0220] WARNING (LEFPARS-2019): TOPOFSTACKONLY statement is obsolete in version 5.6 and later.
The LEF parser will ignore this statement.
To avoid this warning in the future, remove this statement from the LEF file with version 5.6 or later See file /home/data/ispd15/mgc_fft_b/tech.lef at line 804.

[WARNING ODB-0220] WARNING (LEFPARS-2019): TOPOFSTACKONLY statement is obsolete in version 5.6 and later.
The LEF parser will ignore this statement.
To avoid this warning in the future, remove this statement from the LEF file with version 5.6 or later See file /home/data/ispd15/mgc_fft_b/tech.lef at line 814.

[INFO ODB-0223]     Created 9 technology layers
[INFO ODB-0224]     Created 64 technology vias
[INFO ODB-0226] Finished LEF file:  /home/data/ispd15/mgc_fft_b/tech.lef
[INFO ODB-0222] Reading LEF file: /home/data/ispd15/mgc_fft_b/cells.lef
[INFO ODB-0225]     Created 16 library cells
[INFO ODB-0226] Finished LEF file:  /home/data/ispd15/mgc_fft_b/cells.lef
[WARNING ORD-1011] LEF master in01f01 has no liberty cell.
[WARNING ORD-1011] LEF master oa12f01 has no liberty cell.
[WARNING ORD-1011] LEF master oa22f01 has no liberty cell.
[WARNING ORD-1011] LEF master ao12f01 has no liberty cell.
[WARNING ORD-1011] LEF master no02f01 has no liberty cell.
[WARNING ORD-1011] LEF master ao22s01 has no liberty cell.
[WARNING ORD-1011] LEF master na03f01 has no liberty cell.
[WARNING ORD-1011] LEF master na02f01 has no liberty cell.
[WARNING ORD-1011] LEF master no03m01 has no liberty cell.
[WARNING ORD-1011] LEF master h0 has no liberty cell.
[WARNING ORD-1011] LEF master h1 has no liberty cell.
[WARNING ORD-1011] LEF master h2 has no liberty cell.
[WARNING ORD-1011] LEF master h3 has no liberty cell.
[WARNING ORD-1011] LEF master h4 has no liberty cell.
[... 97 lines cut ...]
### Expected Behavior

"Creating the pin blockage for root cluster" succeed

### Environment

```shell
Git commit https://github.com/The-OpenROAD-Project/OpenROAD/commit/cd9320ef9076376a53e4171ccff6d11d6980c346
kernel: Linux 5.4.0-122-generic
os: Ubuntu 18.04.6 LTS
cmake version 3.24.2
gcc version 8.4.0 (Ubuntu 8.4.0-1ubuntu1~18.04)
```

### To Reproduce

[mgc_fft_b.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/11087863/mgc_fft_b.zip)
[mgc_fft_b_tcl.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/11087864/mgc_fft_b_tcl.zip)

### Screenshots

[image]
[image]
[image]
[image]

vijayank88, 2023-08-28:
[image]
@LJK314 
Seems you have pre-placed macros with `floorplan.def`. Is same macros or other macros trying to place with `rtl_macro_placer` again?

maliberty, 2023-08-28:
What is the goal here?  The macros are already placed so there is nothing for macro placement to do.  I'll fix the crash but this is a no-op.

PR #3919 (2023-08-28): mpl2: skip if there are no unplaced macros to work on
Fixes #3101
commit 91264e88 (2023-08-28): mpl2: skip if there are no unplaced macros to work on

Fixes #3101
closed 2023-08-28 by maliberty