The-OpenROAD-Project/OpenROAD #3108: `write_sdc` result is gzipped unprompted

donn, 2023-03-29:
### Describe the bug

When attempting to write_sdc, the result appears to be a gzip file:

[image]

The command is just `write_sdc`. There's no `-gzip` command being passed.

Regardless, `read_sdc` fails on this gzipped file.

### Expected Behavior

The file should be plain text, unless we're missing something?

### Environment

```shell
Git commit: 5d4aeb1c605719de28366908f4fbaae439a57d05
kernel: Linux 6.0.19
os: NixOS 22.11 (Raccoon)
cmake version 3.24.3
-- The CXX compiler identification is Clang 11.1.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /nix/store/naqfglx95i283dwna1zzqfrq3zgyfg89-clang-wrapper-11.1.0/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-7456-gd5e88bef0
-- System name: Linux
-- Compiler: Clang 11.1.0
-- Build type: RELEASE
-- Install prefix: /var/empty/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Could NOT find GTest (missing: GTEST_LIBRARY GTEST_INCLUDE_DIR GTEST_MAIN_LIBRARY) 
-- The C compiler identification is Clang 11.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /nix/store/naqfglx95i283dwna1zzqfrq3zgyfg89-clang-wrapper-11.1.0/bin/clang - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /nix/store/65cp4izx3bllnwqn7c7dhrq9h9gmjkal-python3-3.10.9/bin/python3.10 (found version "3.10.9") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Failed
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Failed
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
[... 84 lines cut ...]
-- STA git sha: 20805c83cf4817f6d2ca7867a52dac226515a21e
-- System name: Linux
-- Compiler: Clang 11.1.0
-- Build type: RELEASE
-- Build CXX_FLAGS: -O3 -DNDEBUG
-- Install prefix: /var/empty/local
-- Found FLEX: /nix/store/5lzf74h8ygk5y5r1cn8a6qc9567cwcw1-flex-2.6.4/bin/flex (found version "2.6.4") 
-- TCL library: /nix/store/378rxqdx6lh4swhw49ifbilc6hbk7h2r-tcl-8.6.11/lib/libtcl.so
-- TCL header: /nix/store/378rxqdx6lh4swhw49ifbilc6hbk7h2r-tcl-8.6.11/include/tcl.h
-- SSTA: 0
-- STA executable: /home/donn/efabless/openroad/src/sta/app/sta
-- GPU is not enabled
-- Found Protobuf: /nix/store/lyklcycj3jdbg6bp80n7pjivizzcfr0c-protobuf-3.21.8/lib/libprotobuf.so (found version "3.21.8") 
-- Found re2: /nix/store/7b2zq7xn1yayhxmrwgsxklfckxyzhfqv-re2-2022-06-01-dev/lib/cmake/re2/re2Config.cmake (found version "9.0.0") 
-- Found GLPK: /nix/store/ycwzcwxr4a3ldpa7b19ypp9zpcxx4gql-glpk-5.0/lib/libglpk.so  
-- Found Eigen3: /nix/store/gzbmbqgjgqm91syx80vn8lzj6gmljw5q-eigen-3.4.0/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0") 
-- GUI is enabled
-- Configuring incomplete, errors occurred!
```

### To Reproduce

`./run.sh`

[repro.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/11105560/repro.tar.gz)

maliberty, 2023-03-29:
Is this really your build configuration: ```Configuring incomplete, errors occurred!```

I ran with the head of OR (d5e88bef0089b8d0525f46ac1e7f74c352b16e26) and ./05-openroad-floorplan/spm.sdc is a plain text file.

donn, 2023-03-29:
I define `USE_SYSTEM_BOOST`, the `./etc/Env.sh` account for that.

I'll try again with the commit you've mentioned.

donn, 2023-03-29:
Ugh, yep, a corner case to end all corner cases.

So `write_sdc` uses `gzopen`:

https://github.com/The-OpenROAD-Project/OpenSTA/blob/882681a674e7089e5f02de4ba2050c6be32b6cd1/sdc/WriteSdc.cc#L330

...which in my compiled OR binary is:

https://github.com/The-OpenROAD-Project/abc/blob/95b3543e928640dfa25f9e882e72a090a8883a9c/src/misc/zlib/gzlib.c#L198

as a shim for:

https://github.com/The-OpenROAD-Project/abc/blob/95b3543e928640dfa25f9e882e72a090a8883a9c/src/misc/zlib/gzlib.c#L92

The OpenSTA invocation uses "T", a mode flag that isn't supported by this implementation which it simply silently ignores. My hypothesis currently is that if ABC is built in an environment without a system `zlib`, it builds the included zlib, which causes this break. Testing further...

donn, 2023-03-29:
Yeah, it's because I compile OpenROAD abc standalone then OpenROAD. I opened a PR for `abc`.

QuantamHD, 2023-03-29:
You'll need to email Alan, the abc author, directly. He doesn't often respond to PRs unless pointed at one though email

donn, 2023-03-29:
@QuantamHD This is OpenROAD's ABC port...

QuantamHD, 2023-03-29:
I'm not sure we're committing patches to those forks.

closed 2025-08-26 by maliberty