The-OpenROAD-Project/OpenROAD #3126: Error when Create dbTechViaLayerRule in odb

Wen-Tian-Pineapple, 2023-04-03:
### Describe the bug

When using odb::dbTechViaLayerRule_create command, I found an error tagged as "signal 6". It doesn't allow me to create new dbTechViaLayerRule. Maybe this is an error.

### Expected Behavior

dbTechViaLayerRule got created successfully

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 3.10.0-1160.88.1.el7.x86_64
os: CentOS Linux 7 (Core)
cmake version 3.24.2
-- The CXX compiler identification is GNU 12.1.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/local/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-7480-g8a4065b09
-- System name: Linux
-- Compiler: GNU 12.1.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Could NOT find GTest (missing: GTEST_LIBRARY GTEST_INCLUDE_DIR GTEST_MAIN_LIBRARY) 
-- The C compiler identification is GNU 12.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/local/bin/gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/local/bin/python3.10 (found version "3.10.2") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
[... 144 lines cut ...]
35# TclEvalObjEx in /lib64/libtcl8.5.so
36# 0x00007FEB1286F1D0 in /lib64/libtcl8.5.so
37# 0x00007FEB12867EB2 in /lib64/libtcl8.5.so
38# 0x00007FEB128AC36C in /lib64/libtcl8.5.so
39# 0x00007FEB128B4647 in /lib64/libtcl8.5.so
40# TclEvalObjEx in /lib64/libtcl8.5.so
41# 0x00007FEB12872F00 in /lib64/libtcl8.5.so
42# 0x00007FEB12867EB2 in /lib64/libtcl8.5.so
43# 0x00007FEB128AC36C in /lib64/libtcl8.5.so
44# TclObjInterpProcCore in /lib64/libtcl8.5.so
45# 0x00007FEB12867EB2 in /lib64/libtcl8.5.so
46# 0x00007FEB128AC36C in /lib64/libtcl8.5.so
47# TclObjInterpProcCore in /lib64/libtcl8.5.so
48# 0x00007FEB12867EB2 in /lib64/libtcl8.5.so
49# 0x00007FEB12868F1E in /lib64/libtcl8.5.so
50# Tcl_EvalEx in /lib64/libtcl8.5.so
51# Tcl_Eval in /lib64/libtcl8.5.so
52# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
53# ord::tclAppInit(Tcl_Interp*) in openroad
54# Tcl_Main in /lib64/libtcl8.5.so
55# main in openroad
56# __libc_start_main in /lib64/libc.so.6
57# 0x0000000000CB5F14 in openroad
Elapsed time: 0:00.49[h:]min:sec. CPU time: user 0.33 sys 0.08 (83%). Peak memory: 83408KB.
```

maliberty, 2023-04-03:
Can you provide a way to reproduce this crash?

maliberty, 2023-04-03:
That doesn't work as dbTechViaLayerRule_create  doesn't take a dbTechNonDefaultRule as an argument
```
Error: foo.tcl, 10 Wrong number or type of arguments for overloaded function 'dbTechViaLayerRule_create'.
  Possible C/C++ prototypes are:
    odb::dbTechViaLayerRule::create(odb::dbTech *,odb::dbTechViaRule *,odb::dbTechLayer *)
    odb::dbTechViaLayerRule::create(odb::dbTech *,odb::dbTechViaGenerateRule *,odb::dbTechLayer *)
```

Wen-Tian-Pineapple, 2023-04-03:
> That doesn't work as dbTechViaLayerRule_create doesn't take a dbTechNonDefaultRule as an argument
>
> ```
> Error: foo.tcl, 10 Wrong number or type of arguments for overloaded function [...]

I see. It works now, I will close the issue

closed 2023-04-03 by Wen-Tian-Pineapple