The-OpenROAD-Project/OpenROAD #3131: PDN spacing violation

kareefardi, 2023-04-05:
### Describe the bug

Wide metal4 spacing violation between PDN strap and inner macro obstruction.

### Expected Behavior

No spacing violation

### OpenROAD Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.1.20-060120-generic
os: Ubuntu 22.04.1 LTS (Jammy Jellyfish)
cmake version 3.25.1
CMake Warning at CMakeLists.txt:96 (message):
  OpenROAD git describe failed, using sha1 instead

-- The CXX compiler identification is GNU 11.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: 580bd35a8b17f0ea7ff8afafb6cf6d23b757063e
-- System name: Linux
-- Compiler: GNU 11.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Found GTest: /home/karim/.local/lib/cmake/GTest/GTestConfig.cmake (found version "1.13.0")  
-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
[... 116 lines cut ...]
fork	git@github.com:kareefardi/OpenLane.git (fetch)
fork	git@github.com:kareefardi/OpenLane.git (push)
hosni	git@github.com:mo-hosni/OpenLane.git (fetch)
hosni	git@github.com:mo-hosni/OpenLane.git (push)
origin	git@github.com:The-OpenROAD-Project/OpenLane.git (fetch)
origin	git@github.com:The-OpenROAD-Project/OpenLane.git (push)
vvbandeira	git@github.com:vvbandeira/OpenLane.git (fetch)
vvbandeira	git@github.com:vvbandeira/OpenLane.git (push)
```

### To Reproduce

[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/11159003/issue_reproducible.tar.gz)

```
bash run.sh
```

### Screenshots

[image]

### Additional Context

If relevant, this issue doesn't happen with the current OpenROAD version in OpenLane -  `6840b7481d49c83870f79646cf979e66f22f6833`

maliberty, 2023-04-09:
@gadfort do you have time for this?

gadfort, 2023-04-10:
@maliberty I'll take a look this week

PR #3187 (2023-04-17): pdn: fix shape cutting when handling obstructions with different spacing requirements
Fixes:
- shapes are cut according to the largest spacing requirement based to ensure shaping is honored correctly.

Closes: #3131
closed 2023-04-17 by maliberty