The-OpenROAD-Project/OpenROAD #3140: 1 drc in mock-array-big

maliberty, 2023-04-06:
### Describe the bug

The head of OR/ORFS is leaving one drc in asap7/mock-array-big.

### Expected Behavior

0 drc

### Environment

```shell
head of OR & ORFS
```

### To Reproduce

Run it yourself or use https://jenkins.openroad.tools/job/OpenROAD-flow-scripts-Nightly-Public/2346/artifact/flow/final_report_mock-array-big_asap7_base_2023-04-06_13-17.tar.gz

osamahammad21, 2023-04-07:
@maliberty this is a result to issue reported in https://github.com/The-OpenROAD-Project/OpenROAD/issues/2827
The solution is to remove this wrong check in lef58_cutspctbl, but It would increase calibre violations for the intel16 designs. The calibre violations are due to unsupported rule by openroad which is lef58_keepoutzone. I have a pr that removes the wrong check and adds the support for the keepout zone rule but it is pending on an update from louii regarding the unreasonable calibre drcs.

yupferris, 2023-04-11:
Does this happen with #955? I'm not seeing it (but I could just be missing it)

maliberty, 2023-04-17:
#2827 has been merged so please try again.