The-OpenROAD-Project/OpenROAD #3149: 2_2_floorplan_io failed with error "Number of pins 2114 exceeds max possible 2084"

DrSpace, 2023-04-10:
### Describe the bug

### I've tried to run a design with very dense IO pins, and 2_2_floorplan_io failed with error "Number of pins 2114 exceeds max possible 2084".

First, when I set the DIE_AREA to`0 0 51 51`, it runs **OK** with `gui_final` below.

[image]

Then, I try to constrait the IO pins to a smaller interval, leaving the DIE_AREA unchanged. It runs **OK** too. You can see below there's some free space at the boundaries.

[image]

In this case, it should be OK if I just set DIE_AREA smaller. So I run again with DIE_AREA set to `0 0 50 50`. It **fails** at the 2_2_floorplan_io stage with error "Number of pins 2114 exceeds max possible 2084".

[image]

I don't understand. If I can constraint pins into a apparently smaller interval and it runs OK, why I get limited when I set DIE_AREA just 1um smaller?

### Expected Behavior

I expect when I set DIE_AREA a little smaller, it should still runs OK, instead of limiting. 

Maybe the pin number limitation should be much loose? Or there is some way for me can change the limitaion?

### Environment

```shell
caitianhao@c111:/OpenROAD-flow-scripts/tools/OpenROAD$ ./etc/Env.sh 
tee: build/openroad-env.log: Permission denied
Unknown git commit, this is not a git repository.

Please make sure that you have the latest code changes and add the commit
hash in the description.

kernel: Linux 5.4.0-146-generic
os: Ubuntu 20.04.5 LTS (Focal Fossa)
cmake version 3.24.2
CMake Warning at CMakeLists.txt:96 (message):
  OpenROAD git describe failed, using sha1 instead

CMake Error: Could not open file for write in copy operation /OpenROAD-flow-scripts/tools/OpenROAD/include/ord/Version.hh.tmp
CMake Error: : System Error: Inappropriate ioctl for device
CMake Error at CMakeLists.txt:133 (configure_file):
  configure_file Problem configuring file

-- The CXX compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: HEAD-HASH-NOTFOUND
-- System name: Linux
-- Compiler: GNU 9.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Found GTest: /usr/lib/x86_64-linux-gnu/libgtest.a  
-- The C compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
[... 98 lines cut ...]
-- Check if compiler accepts -pthread - yes
-- SSTA: 0
-- STA executable: /OpenROAD-flow-scripts/tools/OpenROAD/src/sta/app/sta
-- GPU is not enabled
-- Found re2: /opt/or-tools/lib/cmake/re2/re2Config.cmake (found version "9.0.0") 
-- Found Clp: /opt/or-tools/lib/cmake/Clp/ClpConfig.cmake (found version "1.17.7") 
-- Found Cbc: /opt/or-tools/lib/cmake/Cbc/CbcConfig.cmake (found version "2.10.7") 
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0") 
-- Found SCIP: /opt/or-tools/lib/cmake/scip/scip-config.cmake (found version "8.0.1") 
-- GUI is enabled
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")  
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring incomplete, errors occurred!
```

### To Reproduce

https://drive.google.com/file/d/1b1ZJ-tue9nnEg246WAT4PlvKqhZVN7-z/view?usp=sharing

eder-matheus, 2023-04-11:
@DrSpace Could you try the branch from this PR: https://github.com/The-OpenROAD-Project/OpenROAD/pull/3156?

With the configuration you're using in this design, the number of pins is greater than the number of available positions to place the pins. However, your script manually places pins in pre-determined positions, using multiple layers for vertical and horizontal directions. The previous check wasn't accounting for the pre-placed pins.

But notice that you may have trouble if you don't use the pre-placed pins. You can increase the number of available positions by defining multiple layers for horizontal and vertical pins (in the `place_pins` command, options `-hor_layers` and `-ver_layers`), or reducing the min distance between the pins with the option `-min_distance`.

More details about these options are described here: https://github.com/The-OpenROAD-Project/OpenROAD/tree/master/src/ppl#place-all-pins

closed 2023-04-11 by eder-matheus

hanm2019, 2023-04-12:
It seem that the ORFS use an older version of OpenRoad, Would it be better to update the submodule of ORFS?