The-OpenROAD-Project/OpenROAD #3155: Resizer Crash

donn, 2023-04-11:
### Describe the bug

OpenROAD crashes somewhere when attempting the resizer post-GRT.

### Expected Behavior

OpenROAD should not crash.

### Environment

```shell
Git commit: 76dc134307d935d6516eb6679dff72c6cfb13915
kernel: Linux 6.0.19
os: NixOS 22.11 (Raccoon)
cmake version 3.24.3
-- The CXX compiler identification is Clang 11.1.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /nix/store/naqfglx95i283dwna1zzqfrq3zgyfg89-clang-wrapper-11.1.0/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-7643-g76dc13430
-- System name: Linux
-- Compiler: Clang 11.1.0
-- Build type: RELEASE
-- Install prefix: /var/empty/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Found GTest: /nix/store/4lmdm9a44cwa14jmwqi2wffxkjgl388a-gtest-1.12.1-dev/lib/cmake/GTest/GTestConfig.cmake (found version "1.12.1")  
-- The C compiler identification is Clang 11.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /nix/store/naqfglx95i283dwna1zzqfrq3zgyfg89-clang-wrapper-11.1.0/bin/clang - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Failed
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Failed
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare - Success
[... 111 lines cut ...]
 3# raise in /nix/store/9xfad3b5z4y00mzmk2wnn4900q0qmxns-glibc-2.35-224/lib/libc.so.6
 4# abort in /nix/store/9xfad3b5z4y00mzmk2wnn4900q0qmxns-glibc-2.35-224/lib/libc.so.6
 5# 0x0000000000BEF54F in /nix/store/a61av2sd6wqzqdkf5x2ifbcvjql14wmc-openroad/bin/openroad
 6# grt::FastRouteCore::releaseNetResources(int) in /nix/store/a61av2sd6wqzqdkf5x2ifbcvjql14wmc-openroad/bin/openroad
 7# grt::FastRouteCore::clearNetRoute(int) in /nix/store/a61av2sd6wqzqdkf5x2ifbcvjql14wmc-openroad/bin/openroad
 8# grt::FastRouteCore::addNet(odb::dbNet*, bool, int, int, int, int, float, std::vector<int, std::allocator<int> >*) in /nix/store/a61av2sd6wqzqdkf5x2ifbcvjql14wmc-openroad/bin/openroad
 9# grt::GlobalRouter::makeFastrouteNet(grt::Net*) in /nix/store/a61av2sd6wqzqdkf5x2ifbcvjql14wmc-openroad/bin/openroad
10# grt::GlobalRouter::initNets(std::vector<grt::Net*, std::allocator<grt::Net*> >&) in /nix/store/a61av2sd6wqzqdkf5x2ifbcvjql14wmc-openroad/bin/openroad
11# grt::GlobalRouter::updateDirtyRoutes() in /nix/store/a61av2sd6wqzqdkf5x2ifbcvjql14wmc-openroad/bin/openroad
12# rsz::Resizer::updateParasitics() in /nix/store/a61av2sd6wqzqdkf5x2ifbcvjql14wmc-openroad/bin/openroad
13# rsz::RepairSetup::repairSetup(float, double, int) in /nix/store/a61av2sd6wqzqdkf5x2ifbcvjql14wmc-openroad/bin/openroad
14# 0x0000000000D6BA5A in /nix/store/a61av2sd6wqzqdkf5x2ifbcvjql14wmc-openroad/bin/openroad
15# TclNRRunCallbacks in /nix/store/378rxqdx6lh4swhw49ifbilc6hbk7h2r-tcl-8.6.11/lib/libtcl.so
16# 0x00007F3F4CF6EBC6 in /nix/store/378rxqdx6lh4swhw49ifbilc6hbk7h2r-tcl-8.6.11/lib/libtcl.so
17# Tcl_EvalEx in /nix/store/378rxqdx6lh4swhw49ifbilc6hbk7h2r-tcl-8.6.11/lib/libtcl.so
18# Tcl_Eval in /nix/store/378rxqdx6lh4swhw49ifbilc6hbk7h2r-tcl-8.6.11/lib/libtcl.so
19# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /nix/store/a61av2sd6wqzqdkf5x2ifbcvjql14wmc-openroad/bin/openroad
20# ord::tclAppInit(Tcl_Interp*) in /nix/store/a61av2sd6wqzqdkf5x2ifbcvjql14wmc-openroad/bin/openroad
21# Tcl_MainEx in /nix/store/378rxqdx6lh4swhw49ifbilc6hbk7h2r-tcl-8.6.11/lib/libtcl.so
22# main in /nix/store/a61av2sd6wqzqdkf5x2ifbcvjql14wmc-openroad/bin/openroad
23# 0x00007F3F49C2924E in /nix/store/9xfad3b5z4y00mzmk2wnn4900q0qmxns-glibc-2.35-224/lib/libc.so.6
24# __libc_start_main in /nix/store/9xfad3b5z4y00mzmk2wnn4900q0qmxns-glibc-2.35-224/lib/libc.so.6
25# _start in /nix/store/a61av2sd6wqzqdkf5x2ifbcvjql14wmc-openroad/bin/openroad
./run.sh: line 67: 1263189 Aborted                 (core dumped) openroad -exit -metrics or_metrics_out.json $PACKAGED_SCRIPT_0
```

luis201420, 2023-05-03:
I found the problem, this happens because GRT is not executed before. The GRT information is read from the db file but some data is not being updated. 
I'm going to update that.

maliberty, 2023-05-04:
@donn repair_timing is going to incremental global routing which is not available when reloading from a db today.  When I look at scripts/openroad/resizer_routing_timing.tcl I see you are calling them both in one run.  Is there a need for it to work on reload?

I'm inclined to make this an error unless there is a strong case for it.

donn, 2023-05-05:
The thing is this works for sky130, it has only been when I tried this specific design for GF180 that this issue happened. Do you have more specifics as to when this is needed?

I'm not opposed to just re-running GRT.

maliberty, 2023-05-05:
I don't see why it would be pdk dependent.  I would have thought it is always needed.

I don't see where in OL this actually required though.