The-OpenROAD-Project/OpenROAD #3165: Default colors in ASAP7 are too dark when used on black

oharboe, 2023-04-12:
### Description

The text against black for these pins is illegible.

[image]

### Suggested Solution

Have a default color scheme that's better adapted to be readable against black background.

### Additional Context

Of course it is possible to change the colors in the GUI, but this feature request is about readability in the default case without needing to configure things.

antonblanchard, 2023-04-13:
I've also noticed that the lowest layer color (dark blue / RGB(0, 0, 254)) is hard to see, eg the preferred/non preferred tracks. Perhaps the color palette used is more suited to a white background.

https://github.com/The-OpenROAD-Project/OpenROAD/blob/9f3dc03de34aacd843716e10988c80e56abc448d/src/gui/src/displayControls.cpp#L1726

maliberty, 2023-04-13:
We could turn off the alpha for the text which would make it a bit clearer.  The color of the text matches the layer color which is useful if you have pins on multiple layers.  
[image]
 vs 
[image]
We could just draw it in white as instead.

@gadfort @rovinski any preferences?

maliberty, 2023-04-13:
I see that http://vrl.cs.brown.edu/color has changed some since I generated those colors (more knobs).  I don't get the same set anymore fwiw.  I'm open to suggestions.  The first three layers have traditional values and I wouldn't change them too far.

gadfort, 2023-04-13:
@maliberty I would prefer to preserve the color of the text to match that of the layer, so removing the alpha might be a good solution

maliberty, 2023-04-13:
@antonblanchard you might try setting the alpha=255 for m1 and see if that helps with tracks.

gadfort, 2024-10-18:
@maliberty is this still a thing?

oharboe, 2024-10-18:
Yes, still hard to see.

maliberty, 2024-10-18:
I don't find it hard to see in this case
[image]

do you disagree?

oharboe, 2024-10-18:
> I don't find it hard to see in this case [...]

I find it hard to see, yes. I'll ask a few colleagues.

maliberty, 2024-10-18:
Do you find it hard to see other shapes on that layer or just the text?

oharboe, 2024-10-18:
> Do you find it hard to see other shapes on that layer or just the text?

Both.

maliberty, 2024-10-18:
> The text against black for these pins is illegible.

If this a general color issue and not a text issue that's somewhat different.  You can change the color by double clicking the color chip.

oharboe, 2024-10-18:
> > The text against black for these pins is illegible.
> 
> If this a general color issue and not a text issue that's somewhat different. You can change the color by double clicking the color chip.

The feature request here is that openroad ships with color scheme(s) that don't have to be tweaked to get something readable.

maliberty, 2024-10-18:
I tried http://vrl.cs.brown.edu/color with 
[image]
[image]
 
and get
[image]

How does that seem?  Perhaps you can play around and see if you find a palette that works for you.

maliberty, 2024-10-18:
Readability is subjective - I find it does that already.  You need to be more specific/helpful to get anything to happen.

oharboe, 2024-10-18:
Pointer to OpenROAD code where this is stored?

oharboe, 2024-10-18:
> Readability is subjective - I find it does that already. You need to be more specific/helpful to get anything to happen.

I'll try to make it quantitative: running a poll now.

maliberty, 2024-10-18:
https://github.com/The-OpenROAD-Project/OpenROAD/blob/b972b34737c9382bbcf0ca492ecec74d3e958622/src/gui/src/displayControls.cpp#L1912

oharboe, 2024-10-18:
@maliberty Quick poll by 3 colleagues no-one thought it was easy, 2x hard to read, 1x readable but could be brighted.

maliberty, 2024-10-18:
I already gave a suggestion above but got no response.

oharboe, 2024-10-18:
I made a PR... https://github.com/The-OpenROAD-Project/OpenROAD/pull/5982

commit 1f00f4c3 (2024-10-21): colors: tweak default colors to work better on black background

fixes #3165
commit c08f6d18 (2024-10-21): colors: use bright orange for metal 5 to make it easier to read

fixes #3165
commit c727317d (2024-10-21): colors: tweak default colors to work better on black background

fixes #3165
closed 2024-10-21 by maliberty