The-OpenROAD-Project/OpenROAD #3172: SEGFAULT

oharboe, 2023-04-13:
### Describe the bug

This only happens with release build, if I try with debug build, it doesn't happen.

I can't share the data with which this is happening.

I tried to make an make global_place_issue, but with the "run*.sh" file it doesn't happen.

### Expected Behavior

Should not crash.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.19.0-38-generic
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
cmake version 3.24.2
```

### To Reproduce

[repeats until SEGGFAULT]
rsz::RepairDesign::repairNet(std::shared_ptr<rsz::BufferedNet>, int, int&, sta::Vector<sta::Pin const*>&) (@rsz::RepairDesign::repairNet(std::shared_ptr<rsz::BufferedNet>, int, int&, sta::Vector<sta::Pin const*>&):82)
rsz::RepairDesign::repairNetWire(std::shared_ptr<rsz::BufferedNet>, int, int&, sta::Vector<sta::Pin const*>&) (@rsz::RepairDesign::repairNetWire(std::shared_ptr<rsz::BufferedNet>, int, int&, sta::Vector<sta::Pin const*>&):114)
rsz::RepairDesign::repairNet(std::shared_ptr<rsz::BufferedNet>, int, int&, sta::Vector<sta::Pin const*>&) (@rsz::RepairDesign::repairNet(std::shared_ptr<rsz::BufferedNet>, int, int&, sta::Vector<sta::Pin const*>&):31)
rsz::RepairDesign::repairNet(std::shared_ptr<rsz::BufferedNet>, sta::Pin const*, float, int, sta::Corner const*) (@rsz::RepairDesign::repairNet(std::shared_ptr<rsz::BufferedNet>, sta::Pin const*, float, int, sta::Corner const*):29)
rsz::RepairDesign::repairNet(sta::Net*, sta::Pin const*, sta::Vertex*, bool, bool, bool, int, bool, int&, int&, int&, int&, int&) (@rsz::RepairDesign::repairNet(sta::Net*, sta::Pin const*, sta::Vertex*, bool, bool, bool, int, bool, int&, int&, int&, int&, int&):356)
rsz::RepairDesign::repairDesign(double, double, double, int&, int&, int&, int&, int&) (@rsz::RepairDesign::repairDesign(double, double, double, int&, int&, int&, int&, int&):195)
rsz::Resizer::findResizeSlacks() (@rsz::Resizer::findResizeSlacks():24)
gpl::TimingBase::updateGNetWeights(float) (@gpl::TimingBase::updateGNetWeights(float):18)
gpl::NesterovPlace::doNesterovPlace(int) (@gpl::NesterovPlace::doNesterovPlace(int):1262)
_wrap_replace_nesterov_place_cmd (@_wrap_replace_nesterov_place_cmd:20)
TclNRRunCallbacks (@TclNRRunCallbacks:37)
___lldb_unnamed_symbol1504 (@___lldb_unnamed_symbol1504:315)
Tcl_EvalEx (@Tcl_EvalEx:9)
Tcl_Eval (@Tcl_Eval:11)
sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) (@sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*):31)
ord::tclAppInit(Tcl_Interp*) (@ord::tclAppInit(Tcl_Interp*):128)
Tcl_MainEx (@Tcl_MainEx:111)
main (@main:742)
__libc_start_call_main (@__libc_start_call_main:29)
__libc_start_main_impl (@__libc_start_main@@GLIBC_2.34:43)
_start (@_start:15)

maliberty, 2023-04-13:
@eder-matheus @luis201420 any ideas?

oharboe, 2023-04-13:
Tried valgrind:

```
==392841== Stack overflow in thread #1: can't grow stack to 0x1ffe801000
==392841== Can't extend stack to 0x1ffe800c88 during signal delivery for thread 1:
==392841==   no stack segment
==392841== 
==392841== Process terminating with default action of signal 11 (SIGSEGV)
==392841==  Access not within mapped region at address 0x1FFE800C88
==392841== Stack overflow in thread #1: can't grow stack to 0x1ffe801000
==392841==    at 0xD160DB: rsz::RepairDesign::repairNetLoad(std::shared_ptr<rsz::BufferedNet>, int, int&, sta::Vector<sta::Pin const*>&) (RepairDesign.cc:931)
==392841==  If you believe this happened as a result of a stack
==392841==  overflow in your program's main thread (unlikely but
==392841==  possible), you can try to increase the size of the
==392841==  main thread stack using the --main-stacksize= flag.
==392841==  The main thread stack size used in this run was 8388608.
==392841== 
==392841== HEAP SUMMARY:
==392841==     in use at exit: 261,241,250 bytes in 2,487,907 blocks
==392841==   total heap usage: 8,329,656 allocs, 5,841,749 frees, 773,890,807 bytes allocated
==392841== 
==392841== LEAK SUMMARY:
==392841==    definitely lost: 488,728 bytes in 4,232 blocks
==392841==    indirectly lost: 4,886,401 bytes in 33,780 blocks
==392841==      possibly lost: 6,508,551 bytes in 6,031 blocks
==392841==    still reachable: 249,357,570 bytes in 2,443,864 blocks
==392841==                       of which reachable via heuristic:
==392841==                         multipleinheritance: 354,480 bytes in 8,747 blocks
==392841==         suppressed: 0 bytes in 0 blocks
==392841== Rerun with --leak-check=full to see details of leaked memory
==392841== 
==392841== For lists of detected and suppressed errors, rerun with: -s
==392841== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
Command terminated by signal 11
```

QuantamHD, 2023-04-13:
Looks like you're blowing the stack. I'm assuming you're on Linux. If you are can run `ulimit -s unlimited` and see if it still fails?

QuantamHD, 2023-04-13:
If that does work we need to change this routine from recursion based to an explicit stack

oharboe, 2023-04-13:
> Looks like you're blowing the stack. I'm assuming you're on Linux. If you are can run `ulimit -s unlimited` and see if it still fails?

Yes, it still crashes.

However, why would it not run out of stack with a debug build?

antonblanchard, 2023-04-24:
```
[INFO GPL-0100] worst slack 1e+30
```

That slack seems unlikely, and since there is a difference between building with debug on and off, I wonder if there is some undefined behaviour going on. Building with `-fsanitize=undefined` might show something.

oharboe, 2023-04-24:
I will file a new issue when it is more actionable.

closed 2023-04-24 by oharboe