The-OpenROAD-Project/OpenROAD #3183: I don't see new BTerms added to an existing net in verilog

fgaray, 2023-04-14:
### Describe the bug

For the DFT Module, I am trying to add a scan_out port (module level) to an existing scan cell:

Original:

```
scan cell -> scan_out_pin -> net -> output1
```

Expected:

```
scan cell -> scan_out_pin -> net -> (output1 + scan_out)
```

But the verilog at the end only shows the original connection: net -> output1.

Also, my new port scan_out is being show in the verilog as "input" instead of "output"

### Expected Behavior

1) I expect the verilog to contain a connection from cells's scan_out to my new created scan_out port in the verilog module.
2) I expect that my new created port in the module, scan_out, will have type output instead of input, as I am setting the IO type with:

```
scan_out->setIoType(odb::dbIoType::OUTPUT);
```

### Environment

```shell
Unknown git commit, this is not a git repository.

Please make sure that you have the latest code changes and add the commit
hash in the description.

kernel: Linux 6.1.15-1rodete3-amd64
os: Debian GNU/Linux 11 (bullseye)
cmake version 3.24.2
CMake Warning at CMakeLists.txt:96 (message):
  OpenROAD git describe failed, using sha1 instead

-- The CXX compiler identification is GNU 10.2.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: 39688105d56aec37e654938ed8df618a6f718f50
-- System name: Linux
-- Compiler: GNU 10.2.1
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Could NOT find GTest (missing: GTEST_LIBRARY GTEST_INCLUDE_DIR GTEST_MAIN_LIBRARY) 
-- The C compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.9 (found version "3.9.2") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
[... 128 lines cut ...]

```
git apply PATCH
```

And run:

```
cd src/dft/test
openroad one_cell_sky130.tcl
```

I expect the file in results/one_cell_sky130-tcl.v to have Q connected to output1 and scan_out. Also, looks like scan_out is configured as input, but I am setting it as output with setIoType.

I am printing the BTerms too and I see scan_out and output1 there but not in the verilog.

### Screenshots

Unexpected result:

[image]

### Additional Context

This is for the DFT module, where I need to connect the output of scan_cells's to scan_out port or next's cells scan_in while preserving the functional data paths of the design.

maliberty, 2023-04-17:
I'll take a look at this.  It will have to be represented by a Verilog assign statement.

I am curious why you are adding an extra output port though.  What is the use model for this?

maliberty, 2023-04-17:
Fix coming - I now get:
[image]

rovinski, 2023-04-17:
> I am curious why you are adding an extra output port though. What is the use model for this?

I think that if you are working at a block level and you want to add a scan chain, the scan in / scan out should usually be ports that are added.

maliberty, 2023-04-17:
It has physical routing implications to split inside the block versus at the next higher level.  Is this desirable?

rovinski, 2023-04-17:
Not sure I follow the question - I'm assuming these are physically hardened blocks, i.e. IP. If signals are added, there has to be physically placed pins for the BTerms that are added.

I don't know exactly how @fgaray has the DFT flow set up, but I would imagine that the scan chains are added before floorplanning so that the user can assign the scan in / scan out pin locations.

maliberty, 2023-04-17:
@rovinski something like
[image]

vs
[image]

maliberty, 2023-04-17:
Punching the extra port forces the top view where the bottom view might be better.

closed 2023-04-17 by maliberty

rovinski, 2023-04-17:
But every block needs both a scan in and scan out. Each block is basically a chain that can be connected to other chains
[image]

maliberty, 2023-04-17:
@rovinski these are two dbBTerms on the same dbNet - your diagram shows two different nets.

rovinski, 2023-04-18:
No, there's logic that I abstracted out of the drawing. The squiggles are arbitrarily long chains of scan flops connected from one to another, so they would be different nets. Each scan flop's output is another one's input.

maliberty, 2023-04-18:
That's not what is happening in the code here.  The bug presented was two bterms on one net.  See 
[image]

rovinski, 2023-04-19:
Ok, I think I may have misinterpreted the original question of "why would you want to add an output port" for "why would you want to add an output port _in this case_" 🙃 my bad

commit 950d9302 (2023-04-17): dbSta: correctly set the port direction on the top cell in sta

Listen to the odb callback and update the top_cell's Port dir

Fixes #3183