The-OpenROAD-Project/OpenROAD #3184: make gui_target is not able to find .odb file

ripudamank2, 2023-04-15:
### Describe the bug

[INFO][FLOW] Using platform directory ./platforms/nangate45
make: *** No rule to make target 'gui_6_final.odb', needed by 'gui_final'.  Stop.

### Expected Behavior

it should open gui

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.19.0-38-generic
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
cmake version 3.24.2
CMake Error: The source directory "/home/embedded/Documents/OpenROAD-flow-scripts" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
```

### To Reproduce

make DESIGN_CONFIG=./designs/sky130hd/ibex/config.mk

maliberty, 2023-04-16:
Does results/sky130hd/ibex/6_final.odb exist?  The gui target only loads an existing file.

ripudamank2, 2023-04-16:
[image]

ripudamank2, 2023-04-16:
I found the issue i was using "make gui_final" but to open the design I need to do `make gui_final DESIGN_CONFIG=./designs/sky130hd/ibex/config.m`

closed 2023-04-16 by ripudamank2