The-OpenROAD-Project/OpenROAD #3188: Names on floorplan of macros is illegible with out of the box colors (light yellow on light grey)

oharboe, 2023-04-16:
### Description

[image]

### Suggested Solution

Better default color scheme without need for configuration.

AcKoucher, 2023-06-19:
@oharboe Would it be possible to get a larger print screen of the problem?

oharboe, 2023-06-20:
@AcKoucher I wonder if the color scheme changed. 

Here is a screenshot from https://github.com/The-OpenROAD-Project/OpenROAD/issues/3484

Quite readable out of the box...

[image]

maliberty, 2023-06-20:
I don't think anything has changed.  It might depend on what layers you have showing as your first image was against a more gray background.

oharboe, 2023-06-20:
I have forgotten how I got the grey background and I don't have that design any longer(it was confidential, so I couldn't share it). I don't have a reproduction case from mock-array-big.

Close?

closed 2023-06-20 by maliberty