The-OpenROAD-Project/OpenROAD #3192: Add filtering option when loading large .odb files

oharboe, 2023-04-17:
### Description

Large .odb files, 10s of gigabytes, are slow to load into the OpenROAD GUI.

### Suggested Solution

When large files are being loaded in the GUI(>500mByte?), pop up a requester to filter out numerous, but not particularly interesting bits. 

Fill cells?

oharboe, 2023-04-20:
16x16 8 bit to start with, faster...

```
MOCK_ARRAY_WIDTH=16 MOCK_ARRAY_HEIGHT=16 MOCK_ARRAY_DATAWIDTH=8 MOCK_ARRAY_PITCH_SCALE=2 designs/asap7/mock-array-big/configure.sh
```

```
MOCK_ARRAY_WIDTH=16 MOCK_ARRAY_HEIGHT=16 MOCK_ARRAY_DATAWIDTH=8 MOCK_ARRAY_PITCH_SCALE=2 make DESIGN_CONFIG=designs/asap7/mock-array-big/config.mk
```

maliberty, 2023-04-20:
You can just use -DCMAKE_BUILD_TYPE=RelWithDebInfo

antonblanchard, 2023-04-24:
> MOCK_ARRAY_PITCH_SCALE is a multiple to scale the size of the Element macros. If you increase the datawidth, you'll need to adjust this. Run a floorplan to see that it looks OK before proceeding

I could make your smaller example almost work (fails in drt because of a pin access issue), but the bigger one (16x16x64) seems to fail regardless of what I set for `MOCK_ARRAY_PITCH_SCALE`:

```
MOCK_ARRAY_WIDTH=16 MOCK_ARRAY_HEIGHT=16 MOCK_ARRAY_DATAWIDTH=64 MOCK_ARRAY_PITCH_SCALE=20 make DESIGN_CONFIG=designs/asap7/mock-array-big/config.mk

[ERROR PPL-0089] Could not create matrix for groups. Not available slots inside section.
```

oharboe, 2023-04-24:
@maliberty If the error message above can be reproduced, then I suppose this problem is actionable? https://github.com/The-OpenROAD-Project/OpenROAD/issues/3192#issuecomment-1519208449

oharboe, 2024-03-28:
@maliberty Close? Nothing specifically actionable here. Improving performance is always welcome anywhere, but unless there is something specific suggested or there are resources dedicated, the github issue is just spam.

closed 2024-03-28 by maliberty