The-OpenROAD-Project/OpenROAD #3197: Not quite straight wires

oharboe, 2023-04-18:
### Description

The router doesn't quite route in straight wires.

[image]

Test-case:

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/994

### Suggested Solution

Route in straight wires when possible.

maliberty, 2023-04-18:
@eder-matheus I wonder if this is related to the problem @luis201420 saw with locales (mentioned in today's mtg notes)

eder-matheus, 2023-04-18:
> @eder-matheus I wonder if this is related to the problem @luis201420 saw with locales (mentioned in today's mtg notes)

I think it's slightly different. I've found some odd characters in the "designs/sky130hs/mock-array-big/macro-placement.tcl" file. Interestingly, it isn't present in the GitHub PR, but the file had these characters when I checked out this branch.

eder-matheus, 2023-04-18:
I've removed the characters from the script, and now the design fails in gpl:
`[ERROR GPL-0305] RePlAce diverged at newStepLength.`

maliberty, 2023-04-18:
The underscore in "{config.cols}_{config.pitch_and_margin}" appears to be unusual.  If you delete it and replace it with a normal underscore the problem goes away (though a new one appears later).

maliberty, 2023-04-18:
Actually I think it is supposed to be a space not an underscore.  Once I fix that it gets through this step.

eder-matheus, 2023-04-18:
> The underscore in "{config.cols}_{config.pitch_and_margin}" appears to be unusual. If you delete it and replace it with a normal underscore the problem goes away (though a new one appears later).

Yes, this is exactly the place where I see the odd characters:
[image]

I was able to pass through this step removing it, but it fails in gpl now.

maliberty, 2023-04-18:
Same replace failure for me as well.

maliberty, 2023-04-18:
Things look odd - there are macros stacked on each other in the corner:
[image]

maliberty, 2023-04-18:
@oharboe I think you need to sort this out before we go further.

oharboe, 2023-04-19:
@maliberty @eder-matheus force pushed fix. 

It fails *after* the 6_final.odb is made, but that failure isn't what is of interest here.

```
make DESIGN_CONFIG=designs/sky130hs/mock-array-big/config.mk
[deleted]
[WARNING PSM-0038] Unconnected PDN node on net VDD at location (1711.940um, 753.860um), layer: 5.
[WARNING PSM-0039] Unconnected instance ces_2_6 at location (1711.940um, 753.860um) layer: 5.
[WARNING PSM-0038] Unconnected PDN node on net VDD at location (1711.940um, 753.860um), layer: 6.
[WARNING PSM-0039] Unconnected instance ces_2_6 at location (1711.940um, 753.860um) layer: 6.
[WARNING PSM-0038] message limit reached, this message will no longer print
[WARNING PSM-0039] message limit reached, this message will no longer print
[ERROR PSM-0078] IR drop setup failed.  Analysis can't proceed.
Error: final_report.tcl, 58 PSM-0078
Command exited with non-zero status 1
Elapsed time: 0:14.40[h:]min:sec. CPU time: user 14.18 sys 0.21 (99%). Peak memory: 621644KB.
make: *** [Makefile:676: logs/sky130hs/mock-array-big/base/6_report.log] Error 1
```

maliberty, 2023-04-19:
That doesn't match the title or the description.  Are you reporting a new issue?

oharboe, 2023-04-19:
> That doesn't match the title or the description. Are you reporting a new issue?

No.

The build now runs far enough that the 6_final.odb is being produced and that the problem of the title can be seen.

The flow fails *later*, but that is not a related problem to the title.

vijayank88, 2023-04-19:
@oharboe Can you check the `make DESIGN_CONFIG=designs/sky130hs/mock-array-big/config.mk gui_fianl` is macro placed properly and also check `2_6_pdn.log` is macros power grid generated or not?

oharboe, 2023-04-19:
> @oharboe Can you check the `make DESIGN_CONFIG=designs/sky130hs/mock-array-big/config.mk gui_fianl` is macro placed properly and also check `2_6_pdn.log` is macros power grid generated or not?

Everything is placed fine:

[image]

Power is not the concern in this PR, this PR is only to study routing...

There needs to be a powerstrategy for this PR, which is still being worked out for mock-array-big in asap7. Once a power strategy is sorted out for asap7, then perhaps a similar solution can be transplated into the sky130hs case.

From `2_6_pdn.log`:

```
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - ces_7_4
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - ces_7_5
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - ces_7_6
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - ces_7_7
[WARNING PDN-0110] No via inserted between met4 and met5 at (716.2800, 476.0100) - (716.4000, 477.6100) on VDD
[WARNING PDN-0110] No via inserted between met4 and met5 at (716.2800, 1917.6100) - (716.4000, 1917.7000) on VDD
[WARNING PDN-0110] No via inserted between met4 and met5 at (1083.6000, 244.8100) - (1084.2700, 246.4100) on VSS
[WARNING PDN-0110] No via inserted between met4 and met5 at (1083.6000, 272.0100) - (1084.2700, 273.6100) on VSS
[WARNING PDN-0110] No via inserted between met4 and met5 at (1083.6000, 299.2100) - (1084.2700, 300.8100) on VSS
[WARNING PDN-0110] No via inserted between met4 and met5 at (1083.6000, 326.4100) - (1084.2700, 328.0100) on VSS
[WARNING PDN-0110] No via inserted between met4 and met5 at (1083.6000, 353.6100) - (1084.2700, 355.2100) on VSS
```

vijayank88, 2023-04-19:
> @maliberty @eder-matheus force pushed fix.
>
> It fails _after_ the 6_final.odb is made, but that failure isn't what is of interest here.
>
> ```
> make [...]

If this issue focus on routing and to avoid above failure add the following in `config.mk`
``
export PWR_NETS_VOLTAGES  = ""
export GND_NETS_VOLTAGES  = ""
```
Above configuration skip IR drop report check.

oharboe, 2023-04-19:
Fixed by adding the above, so now this runs to the end:

```
make DESIGN_CONFIG=designs/sky130hs/mock-array-big/config.mk
```

eder-matheus, 2023-04-19:
@oharboe The wires are not straight because the design has P/G routing in these areas. The P/G routing has vias from met1 up to met5, and the detailed router avoids the met3 via shapes with the detours. Check the figure below:
[image]

The same happens in the asap7 version of this design.

closed 2023-04-19 by oharboe