The-OpenROAD-Project/OpenROAD #3205: local OpenROAD GUI crashes when loading an .odb file made by OpenROAD docker image

oharboe, 2023-04-19:
### Describe the bug

Run a flow from Docker to create a .odb file, load with local OpenROAD GUI in Ubuntu 22, crash:

```
Signal 11 received
Stack trace:
 0# 0x000055F5B1B2BAD5 in openroad
 1# 0x00007FF80DA3BCF0 in /lib/x86_64-linux-gnu/libc.so.6
 2# 0x00007FF80DB71DB7 in /lib/x86_64-linux-gnu/libc.so.6
 3# 0x00007FF80DA86DAE in /lib/x86_64-linux-gnu/libc.so.6
 4# fread in /lib/x86_64-linux-gnu/libc.so.6
 5# 0x000055F5B3D643EA in openroad
 6# odb::operator>>(odb::dbIStream&, odb::_dbTechLayer&) in openroad
 7# odb::operator>>(odb::dbIStream&, odb::_dbTech&) in openroad
 8# odb::operator>>(odb::dbIStream&, odb::_dbDatabase&) in openroad
 9# odb::dbDatabase::read(_IO_FILE*) in openroad
10# ord::OpenRoad::readDb(char const*) in openroad
11# gui::MainWindow::openDesign() in openroad
12# 0x00007FF80EAF372C in /lib/x86_64-linux-gnu/libQt5Core.so.5
13# QAction::triggered(bool) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
14# QAction::activate(QAction::ActionEvent) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
15# 0x00007FF80F8F0FE2 in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
16# 0x00007FF80F8F8EEC in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
17# QWidget::event(QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
18# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
19# QApplication::notify(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
20# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
21# QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
22# 0x00007FF80F7C8535 in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
23# 0x00007FF80F7CB4A0 in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
24# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
25# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
26# QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) in /lib/x86_64-linux-gnu/libQt5Gui.so.5
27# QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Gui.so.5
28# 0x00007FF80A70278E in /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
29# g_main_context_dispatch in /lib/x86_64-linux-gnu/libglib-2.0.so.0
30# 0x00007FF80D5713C8 in /lib/x86_64-linux-gnu/libglib-2.0.so.0
31# g_main_context_iteration in /lib/x86_64-linux-gnu/libglib-2.0.so.0
32# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Core.so.5
33# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Core.so.5
34# QCoreApplication::exec() in /lib/x86_64-linux-gnu/libQt5Core.so.5
35# gui::startGui(int&, char**, Tcl_Interp*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) in openroad
36# ord::tclAppInit(Tcl_Interp*) in openroad
37# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
38# main in openroad
39# 0x00007FF80DA23510 in /lib/x86_64-linux-gnu/libc.so.6
40# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
41# _start in openroad
Segmentation fault (core dumped)

```

### Expected Behavior

This used to work. 

### Environment

```shell
master
```

### To Reproduce

As above

oharboe, 2023-04-19:
@maliberty Strange. After this fix, which I wouldn't expect to fix the error in parsing the file, I no longer get an error, nor a crash: https://github.com/The-OpenROAD-Project/OpenROAD/pull/3206

oharboe, 2023-04-19:
@maliberty Does .odb files have version numbers? I rebased this fix on master and then tried against the .odb file from the docker image. So if .odb files are not compatible, but do not have a version number check, that would explain the segfault.

oharboe, 2023-04-19:
https://github.com/The-OpenROAD-Project/OpenROAD/pull/3207

maliberty, 2023-04-19:
Coverity just flagged an issue related to dbIStream this morning that is likely to be the cause:
```
*** CID 1508956:  Memory - illegal accesses  (STRING_NULL)
/src/odb/src/db/dbProperty.cpp: 187 in odb::operator >>(odb::dbIStream &, odb::_dbProperty &)()
181         }
182         case DB_STRING_PROP: {
183           char* char_string;
184           stream >> char_string;
185           prop._value = "";
186           if (char_string != nullptr) {
>>>     CID 1508956:  Memory - illegal accesses  (STRING_NULL)
>>>     Passing unterminated string "char_string" to "basic_string", which expects a null-terminated string. [Note: The source code implementation of the function has been overridden by a builtin model.]
187             prop._value = std::string(char_string);
188             free(char_string);
189           }
190           break;
191         }
192         case DB_DOUBLE_PROP: {
```

I've asked @QuantamHD to fix it as it comes from his recent change.

QuantamHD, 2023-04-20:
Looks like this isn't related to the coverity as all the strings are read and serialized with a null termination.

maliberty, 2023-04-20:
@oharboe can you provide a specific set of steps to reproduce this issue.

oharboe, 2023-04-23:
> @oharboe can you provide a specific set of steps to reproduce this issue.

@maliberty I provided a confidential example, not being able to reproduce with public information. Hopefully that will be enough to move this forward.

oharboe, 2023-04-26:
@maliberty I'll reopen a github issue if this problem persists after the sanatizer cleanups.

closed 2023-04-26 by oharboe