The-OpenROAD-Project/OpenROAD #3211: Preserving Hierarchy During Synthesis Disrupts Delay Modeling

petergrossmann21, 2023-04-19:
### Describe the bug

When running OpenROAD in a flow where hierarchy is preserved during synthesis (i.e. no flatten command during yosys execution), but OpenROAD flattens the circuit during place and route, some portions of the resulting netlist exhibit gate delays reported as zero.  This can be observed in timing reports but is most easily inspected by writing an SDF file so that each gate path delay can be inspected manually.

The symptom is that various paths in various logic gates have IOPATH specifications in the SDF file of zero.  For example:

 (CELL
  (CELLTYPE "sky130_fd_sc_hd__or2_4")
  (INSTANCE soc\/uart\/u_reg\/u_wdata\/wr_en_data_arb\/_10_)
  (DELAY
   (ABSOLUTE
    (IOPATH A X (0.621::0.623) (1.282::1.295))
    (IOPATH B X (0.000::0.000))
   )
  )
 )

Note that not all paths on a given gate are necessarily affected.

### Expected Behavior

The expected behavior would be that all IOPATH entries in the SDF file are nonzero.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.14.0-1059-oem
os: Ubuntu 20.04.5 LTS (Focal Fossa)
cmake version 3.24.2
-- The CXX compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-7537-g47991be87
-- System name: Linux
-- Compiler: GNU 9.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Could NOT find GTest (missing: GTEST_LIBRARY GTEST_INCLUDE_DIR GTEST_MAIN_LIBRARY) 
-- The C compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.9 (found version "3.9.5") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
[... 103 lines cut ...]
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.06kfY8NZvJ
```

### To Reproduce

Download the following:

[test case with hierarchy preserved by yosys](https://drive.google.com/file/d/1aguWG_oJnqVYF23ikackS4KX41xcyPKh/view?usp=sharing)
[test case with flattening by yosys](https://drive.google.com/file/d/1UyRHQVbNjWU8TiGFMPC8RoAdnqXZBT8t/view?usp=sharing)

untar both bundles to directory of your choosing; contents include an open-source design OpenROAD database and related data files to enable post-route timing analysis.

Run the test case with flattening
`cd testcase_works`
`openroad analyze_timing_core.tcl`

Run the test case with yosys hierarchy preserved
`cd testcase`
`openroad analyze_timing_core.tcl`

For the case with hierarchy preserved, see line 61639 of asic_core.sdf for example zero delay annotation, or search on the string "B X (0.000" in your favorite text editor.

maliberty, 2023-04-19:
Did you notice the thousands of warning messages in your non-working case?
```
...
[WARNING STA-0172] ./asic_core.maximum.spef line 298936, instance soc/u_rv_core_ibex/u_core/genblk1.genblk1.gen_regfile_latch.register_file_i/wdata_a_q_$_DFFE_PN0P__Q_12 not found.
[WARNING STA-0172] message limit reached, this message will no longer print
```

The spef files are not equivalent either:
```
*D_NET *14 0.0776701
*CONN
*P ea_dout[0] O
```
vs
```
*D_NET *10652 0.0739977
*CONN
*P ea_dout[0] O
```
Given these differences there is no reason to expect the SDFs to match.  You should start by looking at RCX and figuring out what is going on.  If you have issues there please file an issue with reproducers for that.

maliberty, 2023-04-19:
The zero delays are almost certainly because you have no annotated parasitics.

closed 2023-04-19 by maliberty

petergrossmann21, 2023-04-20:
For a bit of clarification, I did not expect the SDFs to match since the synthesis solution was intentionally different between the two implementations.  That said, if the right way to chase the issue is to look into the extraction issues (I did notice these, wasn't sure whether they were a symptom of something else or the root cause) I'm happy to put a bit of effort into that.