The-OpenROAD-Project/OpenROAD #3219: Rows are not read from settings

maliberty, 2023-04-20:
### Describe the bug

Since #3204 the rows are always on when you start the OR gui.  I believe we don't handle the settings now that it is per site.

### Expected Behavior

I think we should store the overall row settings (on/off) even if we don't store the per site values which would be PDK dependent.

### Environment

```shell
head of master
```

### To Reproduce

any design

gadfort, 2023-04-20:
@maliberty if there are two sites and only one is on, should that count as both on when the GUI restarts or as both off?

maliberty, 2023-04-20:
Its a bit arbitrary but I would count it as on since they are at least interested in something to do with rows.

maliberty, 2023-04-20:
We might change the default to Unchecked as well.

PR #3221 (2023-04-21): gui: save row settings save and restore for group
Closes #3219
closed 2023-04-21 by maliberty