The-OpenROAD-Project/OpenROAD #3225: Problems with 2 gByte limit in file handling

oharboe, 2023-04-21:
### Describe the bug

Do a pass of the source code to hunt down 32 bit int problems with file pointers.

With .odb files, this can reach the gigabyte size and overflow the 32 bit size of int.

Example of a problem:

https://github.com/The-OpenROAD-Project/OpenROAD/blob/0b74722ee863e76ed6a648c33927b91cf1d9e6a9/src/odb/include/odb/dbStream.h#L235

### Expected Behavior

Handle >2gByte files

### Environment

```shell
master
```

### To Reproduce

search as above to find the spots to be fixed

maliberty, 2023-04-24:
That example is dead code.  The problem description itself is too vague to be actionable.  I haven't observed any issues with files > 2gb (including those from you).

closed 2023-04-24 by maliberty

oharboe, 2023-04-24:
> That example is dead code. The problem description itself is too vague to be actionable. I haven't observed any issues with files > 2gb (including those from you).

@maliberty Search for tellg(), you will find lots of cases which store the result in an "int", which is 4 bytes.

Example, this will overflow with >2gByte files:

[image]

Certainly .odb files can be > 2gByte, I don't know about which file types that can realistically be >2gByte.