The-OpenROAD-Project/OpenROAD #3228: Static assertion compiler error with latest spdlog library

andreili, 2023-04-23:
### Describe the bug

While I try to compile OpenROAD binary with spdlog library version 1.11.0, I get a wall of static asser errors about invalid formats.

### Expected Behavior

Normal build, without errors.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.90.1-microsoft-standard-WSL2
os: Gentoo
cmake version 3.25.3
CMake Warning at CMakeLists.txt:96 (message):
  OpenROAD git describe failed, using sha1 instead

-- The CXX compiler identification is GNU 12.2.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: 82fa75d549bf017ff819496d799cabbd39403dd3
-- System name: Linux
-- Compiler: GNU 12.2.1
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Found GTest: /usr/lib64/cmake/GTest/GTestConfig.cmake (found version "1.11.0")
-- The C compiler identification is GNU 12.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
[... 110 lines cut ...]
 1757 |       formattable,
      |       ^~~~~~~~~~~
/usr/include/fmt/core.h:1757:7: note: ‘formattable’ evaluates to false
gmake[1]: *** [src/dst/CMakeFiles/dst.dir/build.make:145: src/dst/CMakeFiles/dst.dir/src/LoadBalancer.cc.o] Error 1

cd /home/andreil/ASIC/OpenROAD_clean/build/src/odb/test/cpp && /usr/bin/g++ -DFMT_SHARED -DSPDLOG_COMPILED_LIB -DSPDLOG_FMT_EXTERNAL -DSPDLOG_SHARED_LIB -I/home/andreil/ASIC/OpenROAD_clean/src/odb/include -I/home/andreil/ASIC/OpenROAD_clean/src/utl/include -I/home/andreil/ASIC/OpenROAD_clean/src/utl/src -I/home/andreil/ASIC/OpenROAD_clean/src/odb/include/odb -I/home/andreil/ASIC/OpenROAD_clean/src/odb/src/def/def -I/home/andreil/ASIC/OpenROAD_clean/src/odb/src/def/defzlib -I/home/andreil/ASIC/OpenROAD_clean/src/odb/src/lef/lef -I/home/andreil/ASIC/OpenROAD_clean/src/odb/src/lef/lefzlib -O3 -DNDEBUG -Wall -Wredundant-decls -Wformat-security -Wno-sign-compare -Wp,-D_GLIBCXX_ASSERTIONS -std=c++17 -MD -MT src/odb/test/cpp/CMakeFiles/TestGuide.dir/TestGuide.cpp.o -MF CMakeFiles/TestGuide.dir/TestGuide.cpp.o.d -o CMakeFiles/TestGuide.dir/TestGuide.cpp.o -c /home/andreil/ASIC/OpenROAD_clean/src/odb/test/cpp/TestGuide.cpp
In file included from /usr/include/spdlog/fmt/fmt.h:31,
                 from /home/andreil/ASIC/OpenROAD_clean/src/utl/include/utl/Logger.h:52,
                 from /home/andreil/ASIC/OpenROAD_clean/src/dst/src/BalancerConnection.cc:45:
/usr/include/fmt/core.h: In instantiation of ‘constexpr fmt::v9::detail::value<Context> fmt::v9::detail::make_value(T&&) [with Context = fmt::v9::basic_format_context<fmt::v9::appender, char>; T = const boost::asio::ip::address&]’:
/usr/include/fmt/core.h:1777:29:   required from ‘constexpr fmt::v9::detail::value<Context> fmt::v9::detail::make_arg(T&&) [with bool IS_PACKED = true; Context = fmt::v9::basic_format_context<fmt::v9::appender, char>; type <anonymous> = fmt::v9::detail::type::custom_type; T = const boost::asio::ip::address&; typename std::enable_if<IS_PACKED, int>::type <anonymous> = 0]’
/usr/include/fmt/core.h:1901:77:   required from ‘constexpr fmt::v9::format_arg_store<Context, Args>::format_arg_store(T&& ...) [with T = {const char* const&, const char* const&, int&, const char* const&, const boost::asio::ip::address&, const short unsigned int&}; Context = fmt::v9::basic_format_context<fmt::v9::appender, char>; Args = {const char*, const char*, int, const char*, boost::asio::ip::address, short unsigned int}]’
/usr/include/fmt/core.h:1918:31:   required from ‘constexpr fmt::v9::format_arg_store<Context, typename std::remove_cv<typename std::remove_reference<Args>::type>::type ...> fmt::v9::make_format_args(Args&& ...) [with Context = basic_format_context<appender, char>; Args = {const char* const&, const char* const&, int&, const char* const&, const boost::asio::ip::address&, const short unsigned int&}]’
/usr/include/spdlog/logger.h:374:75:   required from ‘void spdlog::logger::log_(spdlog::source_loc, spdlog::level::level_enum, spdlog::string_view_t, Args&& ...) [with Args = {const char* const&, const char* const&, int&, const char* const&, const boost::asio::ip::address&, const short unsigned int&}; spdlog::string_view_t = fmt::v9::basic_string_view<char>]’
/usr/include/spdlog/logger.h:90:13:   required from ‘void spdlog::logger::log(spdlog::source_loc, spdlog::level::level_enum, fmt::v9::format_string<T ...>, Args&& ...) [with Args = {const char* const&, const char* const&, int&, const char* const&, const boost::asio::ip::address&, const short unsigned int&}; fmt::v9::format_string<T ...> = fmt::v9::basic_format_string<char, const char* const&, const char* const&, int&, const char* const&, const boost::asio::ip::address&, const short unsigned int&>]’
/usr/include/spdlog/logger.h:96:12:   required from ‘void spdlog::logger::log(spdlog::level::level_enum, fmt::v9::format_string<T ...>, Args&& ...) [with Args = {const char* const&, const char* const&, int&, const char* const&, const boost::asio::ip::address&, const short unsigned int&}; fmt::v9::format_string<T ...> = fmt::v9::basic_format_string<char, const char* const&, const char* const&, int&, const char* const&, const boost::asio::ip::address&, const short unsigned int&>]’
/home/andreil/ASIC/OpenROAD_clean/src/utl/include/utl/Logger.h:252:19:   required from ‘void utl::Logger::log(utl::ToolId, spdlog::level::level_enum, int, const std::string&, const Args& ...) [with Args = {const char*, boost::asio::ip::address, short unsigned int}; std::string = std::__cxx11::basic_string<char>]’
/home/andreil/ASIC/OpenROAD_clean/src/utl/include/utl/Logger.h:157:8:   required from ‘void utl::Logger::warn(utl::ToolId, int, const std::string&, const Args& ...) [with Args = {const char*, boost::asio::ip::address, short unsigned int}; std::string = std::__cxx11::basic_string<char>]’
/home/andreil/ASIC/OpenROAD_clean/src/dst/src/BalancerConnection.cc:136:30:   required from here
/usr/include/fmt/core.h:1757:7: error: static assertion failed: Cannot format an argument. To make type T formattable provide a formatter<T> specialization: https://fmt.dev/latest/api.html#udt
 1757 |       formattable,
      |       ^~~~~~~~~~~
/usr/include/fmt/core.h:1757:7: note: ‘formattable’ evaluates to false
gmake[1]: *** [src/dst/CMakeFiles/dst.dir/build.make:173: src/dst/CMakeFiles/dst.dir/src/BalancerConnection.cc.o] Error 1
```

QuantamHD, 2023-04-23:
OpenROAD isn't compatible with fmt v9 due to a change in default behavior around ostreams. You will have to downgrade to fmt v8.

antonblanchard, 2023-04-24:
No idea if it works on the very latest versions, but I've been working around it with `-DCMAKE_CXX_FLAGS="-DFMT_DEPRECATED_OSTREAM=1"`

andreili, 2023-04-24:
> No idea if it works on the very latest versions, but I've been working around it with `-DCMAKE_CXX_FLAGS="-DFMT_DEPRECATED_OSTREAM=1"`

Oh, thanks, this parameter are fixed build.

closed 2023-04-24 by andreili

maliberty, 2023-04-24:
FMT_DEPRECATED_OSTREAM is my suggestion for working with later versions.  If you want to test it out and report back we can incorporate that into OR.

antonblanchard, 2023-04-24:
> FMT_DEPRECATED_OSTREAM is my suggestion for working with later versions. If you want to test it out and report back we can incorporate that into OR.

I created https://github.com/The-OpenROAD-Project/OpenROAD/pull/3238