The-OpenROAD-Project/OpenROAD #3230: routing stops after warning signal 11 recieved

rahulearn2019, 2023-04-23:
### Describe the bug

I used the make command set in the makefile to build my design from RTL to GDS, but the flow stops during routing stage - after the warning signal 11 received.

### Expected Behavior

I expected the routing to take place and final GDS to be generated for my design.

### Environment

```shell
rahul@rahul-VirtualBox:~/OpenROAD-flow-scripts/tools/OpenROAD$ ./etc/Env.sh
tee: build/openroad-env.log: Permission denied
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.0-69-generic
os: Ubuntu 20.04.6 LTS (Focal Fossa)
cmake version 3.24.2
CMake Error at CMakeLists.txt:142 (configure_file):
  Operation not permitted

-- The CXX compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-6895-g5c85b36f8
-- System name: Linux
-- Compiler: GNU 9.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
[... 88 lines cut ...]
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")  
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring incomplete, errors occurred!
```

### To Reproduce

not clear what is required

### Relevant log output

```shell
unable to drag build_openroad.log
```

### Screenshots

[image]

antonblanchard, 2023-04-24:
> To Reproduce 
> not clear what is required

It looks like you are using ORFS, and the failure is at global routing. If so `make global_route_issue` should create a tarball that you can upload.

rahulearn2019, 2023-04-24:
Thanks, here's the tarball file generated from ```make gobal_route_issue``` in flow directory
[global_route_imc_sky130hd__2023-04-24_14-17.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/11308480/global_route_imc_sky130hd__2023-04-24_14-17.tar.gz)

antonblanchard, 2023-04-24:
I was unable to reproduce, although I did have to make a change because `blocks/sky130hd/VSS_connection.txt` and `blocks/sky130hd/VDD_connection.txt` are not in the tarball.

Your version of OpenROAD is very old, could you retest with the latest upstream?

vijayank88, 2023-06-19:
@rahulearn2019 
Have you updated the ORFS and tried?

maliberty, 2025-08-28:
no updates

closed 2025-08-28 by maliberty