The-OpenROAD-Project/OpenROAD #3236: Separate H&V for set_wire_rc

maliberty, 2023-04-24:
### Description

Current set_wire_rc takes a single -layer argument (or -res/cap).  When we use Steiner tree based estimates we don't know layer but we do know the H&V components of each edge.  We should have separate layers H&V.  

In sky130hd this is particularly important.  Metal2 & metal3 have different thickness and therefore R&C values.

### Suggested Solution

Have separate layers (or res/cap) values per direction.

gadfort, 2023-04-25:
Wouldn't it be better to change RSZ to hold the resistance per square and cap per um2 or something like that so they can be used with the widths of the lines in those directions instead of adding additional parameters which ultimately are just a scalar of eachother?

openroadie, 2023-04-25:
@gadfort Looks like  "RESISTANCE RPERSQ" values are different for M2 and M3 (at least in sky130hd)

M2:   RESISTANCE RPERSQ 0.125 ;
M3:   RESISTANCE RPERSQ 0.047 ;

I suspect the thickness of deposited metal is different and that what @maliberty is referring to (not the width of the wire).

gadfort, 2023-04-25:
Ahh, I see, I think I misread the prompt, and thought it referred to the `set_layer_rc`.

openroadie, 2023-04-25:
@maliberty / @gadfort Should we stick to the convention that the first layer is "X" and the second layer is "Y"? The command can then change from:

`set_wire_rc -signal -layer met2`

to:

`set_wire_rc -signal -layer [met2, met3]
`
Or do we want to do:

`set_wire_rc -signal -x_layer met2 -y_layer met3
`
We can keep the current -layer option (which would control both x and y like it does now).

antonblanchard, 2023-04-26:
```
set_layer_rc [-layer layer]
             [-via via_layer]
             [-capacitance cap]
             [-resistance res]
             [-corner corner]
```

We might want to specify resistance/capacitance values separately in the X and Y directions, so it would be good to handle that too.

gadfort, 2023-04-26:
@antonblanchard so this is what I didn't want to do for `set_layer_rc`. We could change `set_layer_rc` to take in a single value (for resistance ohms/sq) which could be used to generate the X and Y resistances if the layer has different min-widths. This would also let the estimations better honor metals of non-min-width.

It doesn't make physical sense to have resistances/capacitances that are directionally dependent.

`set_wire_rc` is a special case since it's just announcing which layers we want to be using for a given estimate

gadfort, 2023-04-26:
@openroadie something like `set_wire_rc -signal -layer [met2, met3]` would be better I think.
If we can figure out the weights you could use a weighted average for the resistances and capacitances. That could always be done later since it might require some study of metal usage for a large number of designs

antonblanchard, 2023-04-27:
> It doesn't make physical sense to have resistances/capacitances that are directionally dependent.

But aren't we doing that indirectly when specifying two layers?

In ORFS, correlateRC.py uses linear regression to take the results of a design and produce better layer resistance/capacitance estimates, but also better estimates for placement STA (`set_wire_rc -resistance X -capacitance Y`). I'm just wondering if we are giving something up with those estimates (since they can't replicate -layer [met2, met3])

maliberty, 2023-04-27:
I don't think we need to touch set_layer_rc as those values are for a specific layer.  I'm only propose set_wire_rc which is about layer assumptions for unrouted nets.

antonblanchard, 2023-04-27:
> I don't think we need to touch set_layer_rc as those values are for a specific layer. I'm only propose set_wire_rc which is about layer assumptions for unrouted nets.

The discussion was not about the layer estimates I confused things by mentioning it), but whether `set_wire_rc -capacitance nnn -resistance nnn` should be able to take separate values for the X and Y directions.