The-OpenROAD-Project/OpenROAD #3247: Clean up and remove true and false positives with -fsanatize=undefined errors in mock-array-big design

oharboe, 2023-04-25:
### Description

Build with:

```
./build_openroad.sh --local --clean-force --openroad-args -DCMAKE_CXX_FLAGS=-fsanitize=undefined
```

Then run:

```
make DESIGN_CONFIG=designs/asap7/mock-array-big/config.mk
```

Hits:

```
/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/gpl/src/nesterovBase.cpp:344:36: runtime error: signed integer overflow: -2147483648 - 2147483647 cannot be represented in type 'int'
/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/gpl/src/nesterovBase.cpp:344:50: runtime error: signed integer overflow: -2147483648 - 2147483647 cannot be represented in type 'int'
/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/grt/src/fastroute/include/DataType.h:175:8: runtime error: load of value 85, which is not a valid value for type 'bool'
/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/drt/src/gc/FlexGC_init.cpp:88:20: runtime error: downcast of address 0x7f0258010f90 which does not point to an object of type 'drShape'
0x7f0258010f90: note: object is of type 'fr::drVia'
 00 00 00 00  d0 10 33 51 a2 55 00 00  ff ff ff ff 3d 0e 00 00  25 26 00 00 02 7f 00 00  80 a7 5e 61
              ^~~~~~~~~~~~~~~~~~~~~~~
              vptr for 'fr::drVia'
/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/gui/src/gui.cpp:255:29: runtime error: member call on address 0x565503bbd800 which does not point to an object of type 'MainWindow'
0x565503bbd800: note: object is of type 'QWidget'
 00 00 00 00  48 31 e9 ef 6b 7f 00 00  20 ec b0 03 55 56 00 00  f8 32 e9 ef 6b 7f 00 00  00 00 00 00
              ^~~~~~~~~~~~~~~~~~~~~~~
              vptr for 'QWidget'
/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/gui/src/mainWindow.h:94:49: runtime error: member access within address 0x565503bbd800 which does not point to an object of type 'MainWindow'
0x565503bbd800: note: object is of type 'QWidget'
 00 00 00 00  48 31 e9 ef 6b 7f 00 00  20 ec b0 03 55 56 00 00  f8 32 e9 ef 6b 7f 00 00  00 00 00 00
              ^~~~~~~~~~~~~~~~~~~~~~~
              vptr for 'QWidget'
/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/gui/src/gui.cpp:267:22: runtime error: member call on address 0x565503bbd800 which does not point to an object of type 'MainWindow'
0x565503bbd800: note: object is of type 'QWidget'
 00 00 00 00  48 31 e9 ef 6b 7f 00 00  20 ec b0 03 55 56 00 00  f8 32 e9 ef 6b 7f 00 00  00 00 00 00
              ^~~~~~~~~~~~~~~~~~~~~~~
              vptr for 'QWidget'
[etc]
```

maliberty, 2023-04-25:
@vvbandeira please setup a sanitizer build pipeline

maliberty, 2023-04-25:
@gadfort would you look at the gui issues and @eder-matheus can look at the others.

oharboe, 2023-04-25:
> @vvbandeira please setup a sanitizer build pipeline

I think there is a lot to be said for automatic failure of PRs upon sanatizer failures, but that does mean that false positives have to be dealt with.

oharboe, 2023-04-25:
FYI, if you see this one, it is a false positive:

```
/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/gpl/src/nesterovBase.cpp:344:36: runtime error: signed integer overflow: -2147483648 - 2147483647 cannot be represented in type 'int'
/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/gpl/src/nesterovBase.cpp:344:50: runtime error: signed integer overflow: -2147483648 - 2147483647 cannot be represented in type 'int'
[NesterovSolve] Iter: 1 overflow: 0.995667 HPWL: 7577714
[NesterovSolve] Iter: 10 overflow: 0.995671 HPWL: 2144047
```

The code can easily be rewritten not to trigger a false positive by using INT_MAX - 1 and INT_MIN + 1:

[image]

oharboe, 2023-04-25:
See also https://github.com/The-OpenROAD-Project/OpenROAD/issues/3244

gadfort, 2023-04-25:
@maliberty I'm not sure what I'm looking for here. I'm happy to fix anything, but I'm not 100% sure that the issue is

maliberty, 2023-04-26:
To be honest I don't see the problem either now that I look at it.  I wonder if it is confused by multiple inheritance.

antonblanchard, 2023-04-26:
A possible for fix the gui error: https://github.com/The-OpenROAD-Project/OpenROAD/pull/3251

oharboe, 2024-07-14:
@maliberty I think we're agreed that removing sanatizer issues and checking for them in CI would be great.

If so, then this issue can be closed. It doesn't contain anything specifically actionable.

gadfort, 2024-10-18:
@maliberty is this still a thing?

closed 2024-10-18 by maliberty