The-OpenROAD-Project/OpenROAD #3253: Support -region -group -order  all at the same time to allow putting pins on an edge and match up with neighbouring macro

oharboe, 2023-04-26:
### Description

If I have two macros, not of the same Verilog module, and I want to place them close together, then I need the pins to match up at least in the same order.

Ideally the pins should match up so that vertical and horizontal wires can be used, but it would improve the current situation enormously if the pins at least was in the same order for the two macros.

In the image below, there should only be straight vertical wires and a very narrow gap:

[image]

[image]

closed 2023-04-26 by eder-matheus

oharboe, 2023-04-26:
@eder-matheus  I tried with your fix:

```
Elapsed time: 0:04.11[h:]min:sec. CPU time: user 4.04 sys 0.06 (99%). Peak memory: 210556KB.
(/usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' /home/oyvind/2/tools/install/OpenROAD/bin/openroad -exit -no_init  ./scripts/io_placement_random.tcl -metrics ./logs/asap7/mock-array-big/base/2_2_floorplan_io.json) 2>&1 | tee ./logs/asap7/mock-array-big/base/2_2_floorplan_io.log
OpenROAD v2.0-7480-g8a4065b09 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[WARNING STA-0164] ./platforms/asap7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13178, timing group from output port.
[WARNING STA-0164] ./platforms/asap7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13211, timing group from output port.
[WARNING STA-0164] ./platforms/asap7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13244, timing group from output port.
[WARNING STA-0164] ./platforms/asap7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13277, timing group from output port.
[WARNING STA-0164] ./platforms/asap7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13310, timing group from output port.
[WARNING STA-0164] ./platforms/asap7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13343, timing group from output port.
[WARNING STA-0164] ./platforms/asap7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 13376, timing group from output port.
[WARNING STA-0164] ./platforms/asap7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 14772, timing group from output port.
[WARNING STA-0164] ./platforms/asap7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 14805, timing group from output port.
[WARNING STA-0164] ./platforms/asap7/lib/asap7sc7p5t_SIMPLE_RVT_FF_nldm_211120.lib.gz line 14838, timing group from output port.
[ERROR PPL-0086] Both -region and -group constraints not allowed.
Error: io.tcl, 49 PPL-0086
Command exited with non-zero status 1
Elapsed time: 0:03.48[h:]min:sec. CPU time: user 3.40 sys 0.06 (99%). Peak memory: 204336KB.
make: *** [Makefile:441: results/asap7/mock-array-big/base/2_2_floorplan_io.odb] Error 1
```

eder-matheus, 2023-04-26:
@oharboe Could you double-check the branch you're using? From the commit ID in the log you attached (8a4065b09), this is a very old commit.

oharboe, 2023-04-26:
> @oharboe Could you double-check the branch you're using? From the commit ID in the log you attached ([8a4065b](https://github.com/The-OpenROAD-Project/OpenROAD/commit/8a4065b09f5caf8998af059660c63201792a7ea5)), this is a very old commit.

1. I created https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1009
2. ran: `./build_openroad.sh --local`

Could there be a problem with `./build_openroad.sh --local` not rebuilding everything it needs to rebuild?

If `./build_openroad.sh --local` can not be trusted to rebuild, then that would be nice to get fixed, because it could otherwise lead to a ton of confusion.

Trying again with: `./build_openroad.sh --local --clean-force`

eder-matheus, 2023-04-26:
> > @oharboe Could you double-check the branch you're using? From the commit ID in the log you attached ([8a4065b](https://github.com/The-OpenROAD-Project/OpenROAD/commit/8a4065b09f5caf8998af059660c63201792a7ea5)), this is a very old commit.
> 
> 1. I created [Mock array big group order OpenROAD-flow-scripts#1009](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1009)
> 2. ran: `./build_openroad.sh --local`
> 
> Could there be a problem with `./build_openroad.sh --local` not rebuilding everything it needs to rebuild?
> 
> If `./build_openroad.sh --local` can not be trusted to rebuild, then that would be nice to get fixed, because it could otherwise lead to a ton of confusion.
> 
> Trying again with: `./build_openroad.sh --local --clean-force`

To get the latest updates from master branch, you can use:
`./build_openroad.sh --local --latest`

oharboe, 2023-04-26:
> To get the latest updates from master branch, you can use: `./build_openroad.sh --local --latest`

Thanks!

I needed to create the pull request though, so I had to add the commit.

Cheers,

oharboe, 2023-04-26:
@eder-matheus Never mind. mea culpa, pebkac. Continuing testing now with OpenROAD v2.0-7935-g59e34f303.