The-OpenROAD-Project/OpenROAD #3278: Mouse wheel zooming renderes all intermediate steps

oharboe, 2023-05-02:
### Description

If zooming is slow and I use the mouse wheel to zoom in/out, then all the zoom steps are rendered, which makes the GUI sluggish.

### Suggested Solution

Skip interveening zoom levels that have queued up while zooming.

maliberty, 2023-05-02:
@gadfort maybe a small timer on the mouse wheel event?

gadfort, 2023-05-02:
A timer might work, but when looking at a small design the wheel zoom works great and I wonder if the time would result in more jumpy zooming there.
We could try it and see how it handles.

maliberty, 2023-05-02:
Something small like 10ms might be a good tradeoff for continuous scrolling where events should be quite close.

oharboe, 2023-05-02:
> Something small like 10ms might be a good tradeoff for continuous scrolling where events should be quite close.

Why is that needed? the rendering is much slower than 10ms, so by the time rendering finishes, there are lots of events queued up.

If the GUI sets a flag to rerender(dirty) when processing the quueue and then clears the dirty flag and render every time there are no events in the queue and the dirty flag is set, I think the problem would be solved?

maliberty, 2023-05-02:
Qt already does event compression.  When we get the first event we will start drawing at that scale and when we finish that render we will have to start another at where ever you are now zoomed to.  It is better to delay any rendering until things are static.

oharboe, 2023-05-02:
> Qt already does event compression. When we get the first event we will start drawing at that scale and when we finish that render we will have to start another at where ever you are now zoomed to. It is better to delay any rendering until things are static.

I see. I guess what I am observing is that event compression is not working then.

oharboe, 2023-05-03:
I think what is needed is a general strategy and implementaiton to collapse events as Qt doesn't do this out of box in every case. For instance, mouse wheel events are not collapsed.

Also, ref #3279 would be largely addressed if events that change the heatmap configuration was collapsed.

A strawman strategy: push out rendering by 4x the rendering time until no new events(keyboard, mouse, etc.) come in, if rendering time is >200ms would work.

oharboe, 2023-05-05:
@maliberty An idea to make the GUI more interactive. Add a bitmap cache and scale images from the cache before a proper version of can be rendered. Eventually the cache might be populated in the background while the user is staring at the screen pondering the design.

oharboe, 2023-05-10:
@maliberty Interesting: the docker image doesn't queue up zooming like the local install does. So the local install is in fact not behaving as well as the docker version, since the local version will render all intermediate steps, whereas docker version doesn't.

rovinski, 2023-05-19:
By "all intermediate steps" how many zoom events are queued up? If you scroll X times while the first zoom is rendering, does it finish rendering the first zoom, then render each of the X zooms? So the more you zoom the more behind it gets, or do some of the zoom events get collapsed?

oharboe, 2023-05-19:
> By "all intermediate steps" how many zoom events are queued up?

Easily 10s of steps.

> If you scroll X times while the first zoom is rendering, does it finish rendering the first zoom, then render each of the X zooms?

Yes.

> So the more you zoom the more behind it gets,

Yes

> or do some of the zoom events get collapsed?

Apparently not.