The-OpenROAD-Project/OpenROAD #3279: Render in background

oharboe, 2023-05-02:
### Description

When I click to enable heatmap, the GUI becomes unresponsive for a long time.

### Suggested Solution

Do heavy calculations in the background and leave GUI responsive.

maliberty, 2023-05-02:
The database is not thread safe and collisions could happen (say you started the global router while the congestion map was being calculated).  It would require some special limited operation mode and it would be tricky to get right.  It has been discussed before but hasn't risen to the top because of the effort involved relative to the gain.

oharboe, 2023-05-02:
> The database is not thread safe and collisions could happen (say you started the global router while the congestion map was being calculated). It would require some special limited operation mode [...]

I see.

As a first step, it would be nice to push out rendering so that it isnt done 4x when changing heatmap settings.

maliberty, 2023-05-02:
@gadfort thoughts?

gadfort, 2023-05-02:
Once the heatmap is computed it's just drawing it thats the issue (which shoudn't take too long).
It's not until the heatmap is needed that it is populated to avoid slowing down actually useful work in OpenROAD. There might be other inefficiencies that could be addressed, but as long as the database doesn't allow for threaded the options are somewhat limited.

@oharboe where does this 4x come from?

maliberty, 2023-05-02:
@gadfort what makes the computation of the heatmap slow?

gadfort, 2023-05-02:
There are some r-tree lookups (which are not that slow) and some add and multiply, but thats really it.
When it comes to the congestion map, the real issue it probably when zoomed out on a big design there can be a lot of gcells to draw. If the bins for congestion maps were larger it would be faster.

oharboe, 2023-05-02:
> Once the heatmap is computed it's just drawing it thats the issue (which shoudn't take too long). It's not until the heatmap is needed that it is populated to avoid slowing down actually useful work [...]

Any time you click in the heatmap GUI, the screen is rerendered; I need 4 clicks. Every time I change the min max number by a single digit, e.g. from 0 to 80, the screen is rerendered.

maliberty, 2023-05-02:
I could see 2X
- Click to turn off one heat map
- Click to turn on another heat map

Where do the other two clicks go?

oharboe, 2023-05-02:
> I could see 2X
> 
> * Click to turn off one heat map
> * Click to turn on another heat map
> 
> Where do the other two clicks go?

Configure, double click, change min max, enable legend, enable heat map.

oharboe, 2023-05-03:
A generalized solution to #3278 would be a good first step to make working with bulky .odb files tolerable.

closed 2023-05-03 by oharboe