The-OpenROAD-Project/OpenROAD #3321: sky130hd/coyote_tc : [ERROR PAD-9256] Padcell ring cannot be filled Error: floorplan.tcl, 54 PAD-9256

mbautista-lab, 2023-05-11:
### Describe the bug

When I try to run make DESIGN_CONFIG=./designs/sky130hd/coyote_tc/config.mk, I am getting an error. What steps can I take to resolve this issue? 

[WARNING ORD-1011] LEF master sky130_ef_io__gpiov2_pad_wrapped has no liberty cell.
[WARNING STA-0337] port ‘en_i’ not found.
[WARNING STA-0337] port ‘en_i’ not found.
[WARNING STA-0337] port ‘en_i’ not found.
==========================================================================
Floorplan check_setup
--------------------------------------------------------------------------
Warning: There is 1 input port missing set_input_delay.
Warning: There are 5 unclocked register/latch pins.
Warning: There are 255 unconstrained endpoints.
number instances in verilog is 167002
Loading library IO cell meta-data
Loaded library IO cell data
[WARNING IFP-0028] Core area lower left (210.000, 210.000) snapped to (210.220, 212.160).
[INFO IFP-0001] Added 1535 rows of 10390 site unithd with height 1.
[WARNING PDN-9035] No track information found for layer li1.
[WARNING PDN-9035] No track information found for layer met1.
[WARNING PDN-9035] No track information found for layer met2.
[WARNING PDN-9035] No track information found for layer met3.
[WARNING PDN-9035] No track information found for layer met4.
[WARNING PDN-9035] No track information found for layer met5.
[WARNING PAD-9050] Multiple nets found on AMUXBUS_A in padring.
[WARNING PAD-9050] Multiple nets found on AMUXBUS_B in padring.
[INFO PAD-9053] Creating padring nets: VSSIO, VSSA, VDDIO_Q, VCCD, VSSD, VSSIO_Q, AMUXBUS_A, AMUXBUS_B, VDDIO, VDDA, VCCHIB, VSWITCH.
[WARNING PAD-9254] Unfilled gaps in the padring on right side
[WARNING PAD-9255]     4395.0 -> 4395.14
[WARNING PAD-9254] Unfilled gaps in the padring on left side
[WARNING PAD-9255]     4395.0 -> 4395.14
[ERROR PAD-9256] Padcell ring cannot be filled
Error: floorplan.tcl, 54 PAD-9256
Command exited with non-zero status 1
Elapsed time: 0:12.13[h:]min:sec. CPU time: user 11.80 sys 0.32 (99%). Peak memory: 1027056KB.
make: *** [Makefile:437: results/sky130hd/coyote_tc/base/2_1_floorplan.odb] Error 1

### Expected Behavior

I'm expecting a clean run  RTL to GDS using the flow script 

### Environment

```shell
n/a
```

### To Reproduce

n/a

vijayank88, 2023-05-11:
@mbautista-lab 
Are you using latest OpenROAD commit?

vijayank88, 2023-05-11:
@mbautista-lab 
Can you try this [solution](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/89#issuecomment-1004547916)?

vijayank88, 2023-05-11:
Issue reproduced with latest commit as well.

maliberty, 2023-05-11:
This test case has a checkered history.  It was started to demonstrate coyote in a pad ring (tc==test chip).  The person who started it has moved on and it never got to be fully working and isn't in the CI.  We should either finish it or remove it.

Is there a particular reason you need to run this design?

maliberty, 2023-05-11:
@vijayank88 do you want to look at this further?  It needs to be converted from the old ICeWall style to the new pad style for a start (you could look at the pad unit tests for examples).

mbautista-lab, 2023-05-11:
Thanks for looking into the issue I submitted. 
Yes, I believe I have installed the latest commit for OpenRoad. Currently, I am familiarizing myself with the design flow by running example circuits provided in the makefile through the tutorial. However, I have encountered an error and would appreciate your thoughts on what might have caused it and possible solutions to fix it.

maliberty, 2023-05-11:
I suggest you stick to designs in the CI which can be found at https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/jenkins/public_nightly.Jenkinsfile#L54

mbautista-lab, 2023-05-11:
I see, ok, thanks for your reply.  I'll check out the link you just give me.

vijayank88, 2023-05-12:
@maliberty 
If I use `filler_overlap_allowed 1` it is passing pad placement stage. Is that right?
I've discussed same [here](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/89).

vijayank88, 2023-05-12:
@mbautista-lab 
Based on the platform choose `ibex/aes` for flat design (i.e, without macros).
Refer `chameleon/microwatt` for macro based designs.
Implement similar method for your own design for GDSII generation.
If you've any issues feel free to open a new issue.

Similar issue reported here: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/89

maliberty, 2023-05-12:
You would 

> @maliberty If I use `filler_overlap_allowed 1` it is passing pad placement stage. Is that right? I've discussed same [here](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/89).

You would have to look at the gds of the cells involved to see if it is legal or not.  My guess is that the configuration is just wrong and this allows it to go through but will result in drc/lvs errors.

vijayank88, 2023-05-13:
With overlap_allowed got struck at global routing with congestion issue. Let me explore further to fix it.

closed 2025-08-26 by maliberty