The-OpenROAD-Project/OpenROAD #3322: Error: global_route.tcl, 21 GRT-0119

mbautista-lab, 2023-05-11:
### Describe the bug

When I try to run make DESIGN_CONFIG=./designs/sky130hd/chameleon_hier/config.mk, I am getting an error. 

[INFO GRT-0101] Running extra iterations to remove overflow.
[WARNING GRT-0230] Congestion iterations cannot increase overflow, reached the maximum number of times the total overflow can be increased.
[INFO GRT-0197] Via related to pin nodes: 14773
[INFO GRT-0198] Via related Steiner nodes: 555
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 22592
[INFO GRT-0112] Final usage 3D: 321877
[ERROR GRT-0119] Routing congestion too high. Check the congestion heatmap in the GUI and load ./reports/sky130hd/chameleon_hier/base/congestion.rpt in the DRC viewer.
Error: global_route.tcl, 21 GRT-0119
Command exited with non-zero status 1
Elapsed time: 0:33.32[h:]min:sec. CPU time: user 32.20 sys 1.10 (99%). Peak memory: 2623396KB.
make: *** [Makefile:598: results/sky130hd/chameleon_hier/base/5_1_grt.odb] Error 1

### Expected Behavior

A clean RTL-to-GDS flow

### Environment

```shell
n/a
```

### To Reproduce

n/a

maliberty, 2023-05-11:
Have you tried "Check the congestion heatmap in the GUI and load ./reports/sky130hd/chameleon_hier/base/congestion.rpt in the DRC viewer."?

maliberty, 2023-05-11:
Similarly to https://github.com/The-OpenROAD-Project/OpenROAD/issues/3321 not a CI'ed design

maliberty, 2023-05-11:
@vijayank88 do you want to pursue this?

closed 2023-05-18 by maliberty