The-OpenROAD-Project/OpenROAD #3328: write_verilog produce incorrect nets

kareefardi, 2023-05-12:
### Describe the bug

The attached odb produces a verilog with the following nets:
```
 wire [4:0] \clknet_11111111;
```
And sometimes:
```
 wire [4:0] \WF U ;
```
Also other cases reported here https://github.com/The-OpenROAD-Project/OpenLane/pull/1774#issuecomment-1545691791. 

### Expected Behavior

No such nets.

### Environment

```shell
Git commit: aa1f8060d856733b82678802809c57bb260b3aac
kernel: Linux 6.1.20-060120-generic
os: Ubuntu 22.04.1 LTS (Jammy Jellyfish)
cmake version 3.25.1
-- The CXX compiler identification is GNU 11.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-8114-gaa1f8060d
-- System name: Linux
-- Compiler: GNU 11.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Found GTest: /home/karim/.local/lib/cmake/GTest/GTestConfig.cmake (found version "1.13.0")  
-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-sign-compare
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-sign-compare - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-deprecated
-- Performing Test C_COMPILER_SUPPORTS__-Wno-deprecated - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-deprecated
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-deprecated - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-c++11-narrowing
-- Performing Test C_COMPILER_SUPPORTS__-Wno-c++11-narrowing - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-c++11-narrowing
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-c++11-narrowing - Failed
-- Performing Test C_COMPILER_SUPPORTS__-Wno-register
-- Performing Test C_COMPILER_SUPPORTS__-Wno-register - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-register
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-register - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format - SCMake Warning at src/CMakeLists.txt:226 (message):
  spdlog: SPDLOG_FMT_EXTERNAL=ON

uccess
-- Performing Test C_COMPILER_SUPPORTS__-Wno-reserved-user-defined-literal
-- Performing Test C_COMPILER_SUPPORTS__-Wno-reserved-user-defined-literal - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-reserved-user-defined-literal
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-reserved-user-defined-literal - Failed
-- Performing Test C_COMPILER_SUPPORTS__-fpermissive
-- Performing Test C_COMPILER_SUPPORTS__-fpermissive - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-fpermissive
-- Performing Test CXX_COMPILER_SUPPORTS__-fpermissive - Success
-- Performing Test C_COMPILER_SUPPORTS__-x
-- Performing Test C_COMPILER_SUPPORTS__-x - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-x
-- Performing Test CXX_COMPILER_SUPPORTS__-x - Failed
-- Performing Test C_COMPILER_SUPPORTS__c++
-- Performing Test C_COMPILER_SUPPORTS__c++ - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__c++
-- Performing Test CXX_COMPILER_SUPPORTS__c++ - Failed
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-but-set-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-but-set-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-but-set-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-but-set-variable - Success
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- TCL readline library: /usr/lib/x86_64-linux-gnu/libtclreadline.so
-- TCL readline header: /usr/include/x86_64-linux-gnu
-- Found SWIG: /usr/bin/swig4.0 (found suitable version "4.0.2", minimum required is "3.0")  
-- Found Boost: /usr/local/lib/cmake/Boost-1.81.0/BoostConfig.cmake (found version "1.81.0")  
-- boost: 1.81.0
-- Found Python3: /usr/include/python3.10 (found version "3.10.6") found components: Development Development.Module Development.Embed 
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.11") 
-- Found Threads: TRUE  
-- spdlog: 1.9.2
-- Found BISON: /usr/bin/bison (found version "3.8.2") 
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- STA version: 2.4.0
-- STA git sha: 555493cba6e476bd2ff0b9a543de7a781276c2b3
-- System name: Linux
-- Compiler: GNU 11.3.0
-- Build type: RELEASE
-- Build CXX_FLAGS: -O3 -DNDEBUG
-- Install prefix: /usr/local
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- SSTA: 0
-- STA executable: /home/karim/work/OpenROAD/src/sta/app/sta
-- GPU is not enabled
-- Found re2: /home/karim/Downloads/or-tools_x86_64_Ubuntu-22.10_cpp_v9.5.2237/lib/cmake/re2/re2Config.cmake (found version "9.0.0") 
-- Found Clp: /home/karim/Downloads/or-tools_x86_64_Ubuntu-22.10_cpp_v9.5.2237/lib/cmake/Clp/ClpConfig.cmake (found version "1.17.7") 
-- Found Cbc: /home/karim/Downloads/or-tools_x86_64_Ubuntu-22.10_cpp_v9.5.2237/lib/cmake/Cbc/CbcConfig.cmake (found version "2.10.7") 
-- Found Eigen3: /usr/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0") 
-- Found SCIP: /home/karim/Downloads/or-tools_x86_64_Ubuntu-22.10_cpp_v9.5.2237/lib/cmake/scip/scip-config.cmake (found version "8.0.1") 
-- GUI is enabled
-- Found Boost: /usr/local/lib/cmake/Boost-1.81.0/BoostConfig.cmake (found version "1.81.0") found components: serialization 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/local/lib/cmake/Boost-1.81.0/BoostConfig.cmake (found suitable version "1.81.0", minimum required is "1.78")  
-- Found Boost: /usr/local/lib/cmake/Boost-1.81.0/BoostConfig.cmake (found version "1.81.0") found components: serialization system thread 
-- Found Boost: /usr/local/lib/cmake/Boost-1.81.0/BoostConfig.cmake (found version "1.81.0")  
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.XKWMA8f2EV
```

### To Reproduce

[reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/11465455/reproducible.tar.gz)

### Additional Context

The odb was generated in OpenLane flow using the latest OpenROAD version. In the flow, the generated bad wire is `clknet_11111111`. Off the flow, each time a random bunch of characters are generated.

dlmiles, 2023-05-13:
My attempt at a **bisect**:

```
  test8-871faff35-good              f99158b7 test8-871faff35
  test9-cd03c5cf8-bad               9584a78e test9-cd03c5cf8
  test10-84ae3cb8-good              c6f21ab6 test10-84ae3cb8
  test11-59cdc1a5e-good             6a920389 test11-59cdc1a5e
  test12-8a4065b09-bad              bdfe278d test12-8a4065b09
  test13-5d4aeb1c6-good             8f3499e1 test13-5d4aeb1c6
  test14-298fbe4af-bad              8a23d0a6 test14-298fbe4af
  test15-5c1e261be-good             2a4f9a57 test15-5c1e261be
  test16-d5e88bef0-bad              303dae3d test16-d5e88bef0
* test17-e905416f4-buildfailure     bcfa433c test17-e905416f4
```

The commit ID after the testXY- is a n OpenROAD commit ID

```
test12-8a4065b09-bad

test14-298fbe4af-bad           298fbe4af Thu Mar 30 14:51:53 2023 -0700 Merge pull request #3114 from Pinata-Consulting/deltadebug-fix-false-negatives
                               7ae04fcb3 Thu Mar 30 22:24:51 2023 +0200 deltaDebug: add --exit_early_on_error option to speed up bisection
                               ed2d04e36 Wed Mar 29 23:33:36 2023 -0700 Merge pull request #3111 from The-OpenROAD-Project-staging/tidy
                               1f5007a6f Wed Mar 29 22:00:26 2023 -0700 ord: clang-tidy
                               71a707440 Wed Mar 29 15:32:41 2023 -0700 Merge pull request #3107 from donn/allow_disabling_tests
                               d534d3fcb Wed Mar 29 22:58:03 2023 +0200 Allow Disabling Tests
test16-d5e88bef0-bad           d5e88bef0 Wed Mar 29 13:56:37 2023 -0700 Merge pull request #3078 from openroadie/master
                               948ee813e Wed Mar 29 10:36:18 2023 -0700 fix: Fix power unit suffix nw -> nW
                               6698a5592 Wed Mar 29 10:33:15 2023 -0700 fix: Fix power unit suffix nw -> nW
                               450801082 Wed Mar 29 09:58:46 2023 -0700 fix: Updated OpenSTA to fix test crashes in the last update
                               21d16e0c3 Tue Mar 28 22:24:16 2023 -0700 fix: Add missing dependency needed to link with OpenSTA
                               e0674f0ca Tue Mar 28 22:06:11 2023 -0700 fix: Make clang-tidy happy
                               188ddecfa Tue Mar 28 21:55:12 2023 -0700 fix: Add missing dependency needed to link with OpenSTA
                               b9f9168b5 Tue Mar 28 19:02:49 2023 -0700 fix: OpenSTA API return type changes
test17-e905416f4-buildfailure  e905416f4 Tue Mar 28 18:33:01 2023 -0700 fix: Latest OpenSTA
test15-5c1e261be-good          5c1e261be Tue Mar 28 16:46:33 2023 -0700 fix: Updated to latest OpenSTA to fix compile warning treated as error in OR
                               39a9c9628 Tue Mar 28 10:38:58 2023 -0700 fix: update golden due to STA update
                               2e7a9eee0 Tue Mar 28 00:56:52 2023 -0700 Merge branch 'master' into master
                               3c9d9265c Sat Mar 25 23:38:37 2023 -0700 fix: update to the latest OpenROAD/OpenSTA
                               ec43ea082 Sat Mar 25 23:16:52 2023 -0700 fix: update to the latest OpenROAD/OpenSTA
                               d9a93706a Fri Mar 24 12:36:33 2023 -0700 fix: update golden due to STA changes

test13-5d4aeb1c6-good

```

maliberty, 2023-05-13:
@openroadie the code in VerilogWriter::writeWireDcls is clearly wrong:
```
        string bus_name;
        int index;
        parseBusName(net_name, '[', ']', escape, is_bus, bus_name, index);
        BusIndexRange &range = bus_ranges[bus_name.c_str()];
```
bus_name will go out of scope and be deleted but bus_ranges will still be holding a pointer to it.  Please have this fixed in sta.

closed 2023-05-23 by openroadie

kareefardi, 2023-05-25:
I know that this is fixed, but OpenSTA in OR is upstreamed yet. So, I think this should remain open until then.