The-OpenROAD-Project/OpenROAD #3329: check_antennas double free segfault

gadfort, 2023-05-12:
### Describe the bug

When running `check_antennas` it results in a double free or corruption segfault.
In this case, the net was `rst_ni`

Here is the backtrace: 
```
[INFO GRT-0043] No OR_DEFAULT vias defined.
double free or corruption (out)

Thread 1 "openroad" received signal SIGABRT, Aborted.
__pthread_kill_implementation (no_tid=0, signo=6, threadid=140737242349696) at ./nptl/pthread_kill.c:44
44      ./nptl/pthread_kill.c: No such file or directory.
(gdb) bt
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=140737242349696) at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=140737242349696) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=140737242349696, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007ffff4021476 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007ffff40077f3 in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007ffff40686f6 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff41bab8c "%s\n") at ../sysdeps/posix/libc_fatal.c:155
#6  0x00007ffff407fd7c in malloc_printerr (str=str@entry=0x7ffff41bd7b0 "double free or corruption (out)") at ./malloc/malloc.c:5664
#7  0x00007ffff4081ef0 in _int_free (av=0x7ffff41f8c80 <main_arena>, p=0x55556d37c650, have_lock=<optimized out>) at ./malloc/malloc.c:4588
#8  0x00007ffff40844d3 in __GI___libc_free (mem=<optimized out>) at ./malloc/malloc.c:3391
#9  0x00005555591e7497 in __gnu_cxx::new_allocator<std::__detail::_Hash_node_base*>::deallocate (this=0x7fffffffd1ef, __p=0x55556d37c660, __t=5087) at /usr/include/c++/11/ext/new_allocator.h:145
#10 0x00005555591e5458 in std::allocator_traits<std::allocator<std::__detail::_Hash_node_base*> >::deallocate (__a=..., __p=0x55556d37c660, __n=5087) at /usr/include/c++/11/bits/alloc_traits.h:496
#11 0x0000555559bffa47 in std::__detail::_Hashtable_alloc<std::allocator<std::__detail::_Hash_node<grt::GSegment, true> > >::_M_deallocate_buckets (this=0x7fffffffd380, __bkts=0x55556d37c660, __bkt_count=5087)
    at /usr/include/c++/11/bits/hashtable_policy.h:1942
#12 0x0000555559bfe32e in std::_Hashtable<grt::GSegment, grt::GSegment, std::allocator<grt::GSegment>, std::__detail::_Identity, std::equal_to<grt::GSegment>, grt::GSegmentHash, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, true, true> >::_M_deallocate_buckets (this=0x7fffffffd380, __bkts=0x55556d37c660, __bkt_count=5087)
    at /usr/include/c++/11/bits/hashtable.h:449
#13 0x0000555559bfca44 in std::_Hashtable<grt::GSegment, grt::GSegment, std::allocator<grt::GSegment>, std::__detail::_Identity, std::equal_to<grt::GSegment>, grt::GSegmentHash, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, true, true> >::_M_deallocate_buckets (this=0x7fffffffd380) at /usr/include/c++/11/bits/hashtable.h:454
#14 0x0000555559bfb464 in std::_Hashtable<grt::GSegment, grt::GSegment, std::allocator<grt::GSegment>, std::__detail::_Identity, std::equal_to<grt::GSegment>, grt::GSegmentHash, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, true, true> >::~_Hashtable (this=0x7fffffffd380, __in_chrg=<optimized out>) at /usr/include/c++/11/bits/hashtable.h:1533
#15 0x0000555559bfabc6 in std::unordered_set<grt::GSegment, grt::GSegmentHash, std::equal_to<grt::GSegment>, std::allocator<grt::GSegment> >::~unordered_set (this=0x7fffffffd380, __in_chrg=<optimized out>)
    at /usr/include/c++/11/bits/unordered_set.h:97
#16 0x0000555559bf82c8 in grt::RepairAntennas::makeNetWire (this=0x55556b842790, db_net=0x5555638afc08, route=std::vector of length 4194, capacity 8192 = {...}, default_vias=std::map with 5 elements = {...})
    at /home/pgadfort/OpenROAD/src/grt/src/RepairAntennas.cpp:185
#17 0x0000555559bf7cad in grt::RepairAntennas::makeNetWires (this=0x55556b842790, routing=std::map with 9488 elements = {...}, max_routing_layer=6) at /home/pgadfort/OpenROAD/src/grt/src/RepairAntennas.cpp:105
#18 0x0000555559b9f97e in grt::GlobalRouter::makeNetWires (this=0x55555f647800) at /home/pgadfort/OpenROAD/src/grt/src/GlobalRouter.cpp:383
#19 0x000055555a60b0b6 in ant::AntennaChecker::checkAntennas (this=0x55555f661df0, net=0x0, verbose=false) at /home/pgadfort/OpenROAD/src/ant/src/AntennaChecker.cc:1737
#20 0x000055555a61baaf in ant::check_antennas (net_name=0x7ffff7fa9038 "", verbose=false) at /home/pgadfort/OpenROAD/build/src/ant/src/CMakeFiles/ant.dir/AntennaCheckerTCL_wrap.cxx:1735
#21 0x000055555a61beda in _wrap_check_antennas (clientData=0x0, interp=0x55555f694590, objc=3, objv=0x55555f6a2c40) at /home/pgadfort/OpenROAD/build/src/ant/src/CMakeFiles/ant.dir/AntennaCheckerTCL_wrap.cxx:1854
#22 0x00007ffff7e36d32 in TclNRRunCallbacks () from /lib/x86_64-linux-gnu/libtcl8.6.so
#23 0x00007ffff7e38028 in ?? () from /lib/x86_64-linux-gnu/libtcl8.6.so
#24 0x00007ffff7e37aa7 in Tcl_EvalEx () from /lib/x86_64-linux-gnu/libtcl8.6.so
#25 0x00007ffff7e38c9a in Tcl_Eval () from /lib/x86_64-linux-gnu/libtcl8.6.so
#26 0x00005555595261de in sta::sourceTclFile (filename=0x7fffffffe313 "../../sc_collected_files/openroad_db7b5b88649b2143f9a805a3e7197c48e1ea0a60/sc_apr.tcl", echo=false, verbose=false, interp=0x55555f694590)
    at /home/pgadfort/OpenROAD/src/sta/app/StaMain.cc:95
#27 0x000055555910d28e in tclAppInit (argc=@0x55555f363498: 2, argv=0x7fffffffe048, init_filename=0x55555c4fe9b7 ".openroad", interp=0x55555f694590) at /home/pgadfort/OpenROAD/src/Main.cc:417
#28 0x000055555910d543 in ord::tclAppInit (interp=0x55555f694590) at /home/pgadfort/OpenROAD/src/Main.cc:445
#29 0x00007ffff7eeab19 in Tcl_MainEx () from /lib/x86_64-linux-gnu/libtcl8.6.so
#30 0x000055555910cbf3 in main (argc=3, argv=0x7fffffffe048) at /home/pgadfort/OpenROAD/src/Main.cc:296
```

### Expected Behavior

Not segfaulting

### Environment

```shell
Git commit: aa1f8060d856733b82678802809c57bb260b3aac
kernel: Linux 5.15.90.1-microsoft-standard-WSL2
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
cmake version 3.25.0
-- The CXX compiler identification is GNU 11.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-8114-gaa1f8060d
[... 263 lines cut ...]
[INFO GRT-0018] Total wirelength: 793941 um
[INFO GRT-0014] Routed nets: 9488
######################
# Report and Repair Antennas
######################
estimate_parasitics -global_routing
if {[check_antennas -report_file "reports/${sc_design}_antenna.rpt"] != 0} {
  if {[llength [dict get $sc_cfg library $sc_mainlib asic cells antenna]] != 0} {
    set sc_antenna [lindex [dict get $sc_cfg library $sc_mainlib asic cells antenna] 0]
    # Remove filler cells before attempting to repair antennas
    remove_fillers
    repair_antenna $sc_antenna \
      -iterations $openroad_ant_iterations \
      -ratio_margin $openroad_ant_margin
    # Add filler cells back
    insert_fillers
    # Check antennas again to get final report
    check_antennas -report_file "reports/${sc_design}_antenna_post_repair.rpt"
  }
}
[INFO GRT-0043] No OR_DEFAULT vias defined.
double free or corruption (out)
Signal 6 received
Stack trace:
```

luis201420, 2023-05-29:
Hi @gadfort,
I've been looking into your bug for the last week but it seems to be fixed in the latest versions of OR (commit [grt: do not create virtual wires on first routing layer](https://github.com/The-OpenROAD-Project-private/OpenROAD/commit/31baf6442f3443a7a25050743d3386e374c2403d)).
Could you run the test again with the most recent version of OR and confirm that the bug is gone?
I will be waiting for your confirmation

gadfort, 2023-05-29:
@luis201420 I don't see the issue anymore. Thanks for checking. I'll close the issue.

closed 2023-05-29 by gadfort