The-OpenROAD-Project/OpenROAD #3366: Unable to build OpenROAD locally

Jainil25, 2023-05-22:
### Describe the bug

Hey,
Greeting of the day!

I'm using Ubuntu 22.04.2 LTS on my Windows 11(WSL).

I typed the following commands to build OpenROAD locally. But there is some error which I'm unable to solve.
Please help. Here's the code:

**~/OpenROAD-flow-scripts$** git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
sudo ./setup.sh
**~/OpenROAD-flow-scripts/tools/OpenROAD$** sudo etc/DependencyInstaller.sh -dev
**~/OpenROAD-flow-scripts$** ./build_openroad.sh --local

Here's the screenshot of the error:

[image]

### Expected Behavior

Expected Behavior:  OpenROAD build locally successfully.

### Environment

```shell
**jainil25@Jainil:~/OpenROAD-flow-scripts/tools/OpenROAD$** ./etc/Env.sh
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.90.1-microsoft-standard-WSL2
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-8118-gb95ce9901
-- System name: Linux
-- Compiler: GNU 11.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Found GTest: /usr/lib/x86_64-linux-gnu/cmake/GTest/GTestConfig.cmake (found version "1.11.0")
-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
[... 86 lines cut ...]
-- Found Cbc: /opt/or-tools/lib/cmake/Cbc/CbcConfig.cmake (found version "2.10.7")
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0")
-- Found SCIP: /opt/or-tools/lib/cmake/scip/scip-config.cmake (found version "8.0.1")
-- GUI is enabled
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization
-- Found OpenMP_CXX: -fopenmp (found version "4.5")
-- Found OpenMP: TRUE (found version "4.5")
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS)
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.ZVsjXNUfkH
```

### To Reproduce

Please help me solve this.

Thanks, 
Jainil.

maliberty, 2023-05-22:
That snippet doesn't contain the error either.  Once it fails please include the first line with "error:"

maliberty, 2023-05-22:
c++: fatal error: Killed signal terminated program cc1plus

This is not a normal compiler error.  I suggests you are running out of system resources or something is killing the job.  Perhaps try fewer threads and ensure you have adequate disk space.

Jainil25, 2023-05-22:
Thanks for your suggestion but I don't know how to implement those.
My laptop has 8gb Ram and more than 100gb available on each C and D drive. 
I have a question, do I need to install Microsoft Visual C++ for building OpenROAD locally?

maliberty, 2023-05-22:
No, we don't support Microsoft Visual C++.   Try ./build_openroad.sh --local --threads 1

Jainil25, 2023-05-22:
Thank you, It has been successfully built locally.
I do have one more question, would I need to require to do the same while installing Docker or while using OpenROAD tools?

maliberty, 2023-05-22:
In docker it would likely be the same as it is using the same machine.  When running OR it will depend on your design size.  In general 8gb is pretty small for EDA tool usage.

vijayank88, 2023-06-07:
Seems it's resolved, so closing it.

closed 2023-06-07 by vijayank88

QuantamHD, 2026-03-12:
Try using the new bazel build instead. Install bazelisk https://github.com/bazelbuild/bazelisk

Then run 

`bazelisk build //:openroad`