The-OpenROAD-Project/OpenROAD #3369: Error, while trying to add a new platform

maliberty, 2023-05-22:
### Discussed in https://github.com/The-OpenROAD-Project/OpenROAD/discussions/3368

<div type='discussions-op-text'>

<sup>Originally posted by **lokki11199** May 22, 2023</sup>
Hello,
I am trying to integrate a FreePDK45 platform and get an error on global route stage:
[image]

A FreePDK45 platform and Nangate45 are similar to each other, excepting tha geometry of standard cells (in FreePDK45 they are polygon). I am making config.mk based on Nangate45 config.mk.
OpenROAD Flow version: v2.0-7988-g20b907452 

Help me please to solve this problem.

[global_route_PWM_freepdk45nm_base_2023-05-23_00-37.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/11537324/global_route_PWM_freepdk45nm_base_2023-05-23_00-37.tar.gz)
</div>

maliberty, 2023-05-22:
@lokki11199 FYI

eder-matheus, 2023-05-23:
> > @maliberty @luis201420 AFAIK, Nangate45 doesn't have antenna rules. If the same happens in FreePDK45, the antenna checker should detect when the platform doesn't have antenna rules and skip this stage.
> > The crash seems related to the ant not finding the antenna rules values.
> 
> Yes, FreePDK45 doesn't have antenna rules too. Are there ways to skip this stage by hand?

You could play with your scripts to check if the platform has antenna rules and skip check_antennas. Are you using ORFS or some other flow?

But in the end, we will update our tools to avoid such errors and properly handle platforms without antenna rules.

lokki11199, 2023-05-23:
> You could play with your scripts to check if the platform has antenna rules and skip check_antennas. Are you using ORFS or some other flow?
>
> But in the end, we will update our tools to avoid such [...]

I am using ORFS.

eder-matheus, 2023-05-24:
@lokki11199 Could you try this: https://github.com/The-OpenROAD-Project/OpenROAD/pull/3383?

I've tried with your test case, and my PR fixes it. Can you check if it works in your end too?

lokki11199, 2023-05-25:
> @lokki11199 Could you try this: #3383?
> 
> I've tried with your test case, and my PR fixes it. Can you check if it works in your end too?

Thanks! It's work.