The-OpenROAD-Project/OpenROAD #3381: CTS is failing with "Command terminated by signal 11"

sakundu, 2023-05-24:
### Describe the bug

Here is the error message I received when I am running CTS using ORFS flow scripts. 
```
[INFO CTS-0034]     Segment length (rounded): 3.
Command terminated by signal 11
Elapsed time: 2:06.21[h:]min:sec. CPU time: user 111.93 sys 12.07 (98%). Peak memory: 10346536KB.
```

### Expected Behavior

It should go through the CTS flow. 

I have tried to run CTS for multiple floorplan utilizations of MemPool Group design and it failed for all the cases. 

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 4.18.0-408.el8.x86_64
os: CentOS Stream 8
cmake version 3.16.2
-- The CXX compiler identification is GNU 9.3.0
-- Check for working CXX compiler: /home/tool/gcc/gcc-9.3.0/bin/c++
-- Check for working CXX compiler: /home/tool/gcc/gcc-9.3.0/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-8120-g8859e39e9
-- System name: Linux
-- Compiler: GNU 9.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Found GTest: /home/tool/GTest/gtest/1.13.0/lib64/libgtest.so  
-- The C compiler identification is GNU 9.3.0
-- Check for working C compiler: /home/tool/gcc/gcc-9.3.0/bin/gcc
-- Check for working C compiler: /home/tool/gcc/gcc-9.3.0/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
[... 82 lines cut ...]
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Found VTune: /home/tool/intel/vtune_amplifier/lib64/libittnotify.a  
-- Found Boost: /home/tool/boost/boost_1_78_0/lib/cmake/Boost-1.78.0/BoostConfig.cmake (found suitable version "1.78.0", minimum required is "1.78")  
-- Found Boost: /home/tool/boost/boost_1_78_0/lib/cmake/Boost-1.78.0/BoostConfig.cmake (found version "1.78.0") found components: serialization system thread 
-- Found Boost: /home/tool/boost/boost_1_78_0/lib/cmake/Boost-1.78.0/BoostConfig.cmake (found version "1.78.0")  
-- TCL readline disabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.tebLV0iAF6
```

### To Reproduce

I have ran make cts_issue command and generated the tarfile. Here is the links:
- [Link](https://drive.google.com/file/d/1SND13OF-wKd73xk9RC4ix_kYa33twfOR/view?usp=share_link): 0.25 Floorplan Utilization
- [Link](https://drive.google.com/file/d/1qC9hLbjScBVT5ZIzxwWTI9O_ysDRp6eS/view?usp=share_link): 0.20 Floorplan Utilization 
- [Link](https://drive.google.com/file/d/1YQPjEI-z7CR67fvEioBgmhMN9baC_xXO/view?usp=share_link): 0.15 Floorplan Utilization with manual macro placement.

Please let me know if I am supposed to provide different inputs. 

Thanks,
Sayak

maliberty, 2023-05-24:
I have a similar crash from UMich that I'm looking at.  Hopefully it is the same.

maliberty, 2023-05-25:
Please try master again and see if https://github.com/The-OpenROAD-Project/OpenROAD/pull/3387 fixed this

maliberty, 2023-10-17:
no response

closed 2023-10-17 by maliberty