The-OpenROAD-Project/OpenROAD #3389: fail early in drt with off-grid pins

maliberty, 2023-05-25:
### Describe the bug

When a macro's pins are off the manufacturing grid there is no hope for drt to succeed.  Please detect this and error early.  We can check for this in odb import (in io).

### Expected Behavior

Error with informative message in such cases.

### Environment

```shell
head of master
```

### To Reproduce

modify the macro placement of any design to trigger this

maliberty, 2023-05-25:
@oharboe FYI

oharboe, 2023-05-25:
Add a default off option that has to be enabled to allow offgrid placement, or simply error?

rovinski, 2023-05-25:
DRT won't be able to find a path that doesn't cause a violation, so it will keep spinning until it runs out of iterations. It is better to error.

oharboe, 2023-05-26:
Here is an example where I'm struggling. I can't find the single source of truth of what grid to place the macro on...

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1097/commits/581aae81523d50991e47059e7af7bf7a07042fc6#r1206153460

rovinski, 2023-05-26:
There are three main types of grids: manufacturing, cell (placement), and routing.
The manufacturing grid represents the finest granularity available in the process. _All_ geometries in the layout file must be a multiple of the manufacturing grid.

For example, the manufacturing grid in ASAP7 is 0.001 um.
https://github.com/The-OpenROAD-Project/OpenROAD/blob/575222da2e7b669c11899253b61a897df46a4f20/test/asap7/asap7_tech_1x_201209.lef#L41

The cell grid represents a quantized grid of locations where cells can be placed. The grid is composed of cell sites and cell rows. Multiple cell grids can exist in a design (for example, one for standard cells and one for I/O cells). The cell grid must be a multiple of the manufacturing grid.

For example, a cell site in ASAP7 is 0.054 um x 0.270 um.
https://github.com/The-OpenROAD-Project/OpenROAD/blob/575222da2e7b669c11899253b61a897df46a4f20/test/asap7/asap7sc7p5t_28_R_1x_220121a.lef#L36-L40

The routing grid represents the quantized grid that the router uses to route nets. The grid for each metal layer can be different. The pitch and offset of the grid is usually inferred from the minimum allowable distances (but doesn't always have to). The routing grid must be a multiple of the manufacturing grid.

In ASAP7 for example, the M3 offset is 0.0 um and the pitch is 0.036 um.
https://github.com/The-OpenROAD-Project/OpenROAD/blob/575222da2e7b669c11899253b61a897df46a4f20/test/asap7/asap7_tech_1x_201209.lef#L174-L180

For macro generation, the only hard rule is that the internal shapes (metals, pins, transistors, vias, etc.) have to be on the manufacturing grid when they are instantiated at the top-level.

_But_, the best practice is for the total macro area to be a multiple of the cell sites (for ASAP7, x is multiple of 0.054, y is multiple of 0.270), and the pins to be on the routing grid (for ASAP7 M3, pins should be a multiple of 0.036 um apart).

One problem you can run into is that the routing grid inside the macro and the routing grid outside the macro don't line up. This makes it a bit more difficult for the router if the pins are congested, but usually the router can handle it. If the pins are off the _manufacturing_ grid, then your macro was either not generated correctly or not placed correctly at the top-level.

TL;DR: Make sure that your pins are on the routing grid both inside the macro and when the macro is instantiated. You macro size doesn't necessarily need to be a multiple of the cell grid, but it's usually cleaner if it is.

maliberty, 2023-05-26:
Note that in asap7 the track pattern is more complex and is defined in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/platforms/asap7/openRoad/make_tracks.tcl rather than taken from the LEF.  The M2 pattern is the complex bit.

maliberty, 2023-05-26:
Also asap7 doesn't allow wrong way routing so pins off the routing tracks are more difficult to deal with.

oharboe, 2023-05-26:
@rovinski Thank you for this explanation! So what position should I align macros to? I use 0.054 and 0.270 in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1097

Still... the pins dont line up...

rovinski, 2023-05-26:
Honestly, I would just turn on the routing grid in your top level and get the center of your pins to line up with it.

Mathematically, it probably needs to be a common multiple of both the routing pitch and the placement pitch. It's actually more important for the pins to line up with the routing grid than for the macro to line up with the placement grid.

oharboe, 2023-05-26:
> Honestly, I would just turn on the routing grid in your top level and get the center of your pins to line up with it.
>
> Mathematically, it probably needs to be a common multiple of both the routing [...]

Quite, but I need to write tcl to place macros :-)

oharboe, 2023-05-26:
Least common denominator of 0.036 and 0.054 is 0.108?

rovinski, 2023-05-26:
You also need to account for the routing grid offset in both the top-level and the macro. Keep in mind that the routing grid offset is from the _die_ boundary not the _core_ boundary. The math is annoying which is why I am hesitant to work it out late at night :)

oharboe, 2023-05-26:
> You also need to account for the routing grid offset in both the top-level and the macro. Keep in mind that the routing grid offset is from the _die_ boundary not the _core_ boundary. The math is annoying which is why I am hesitant to work it out late at night :)

I am relieved to know it is a hard problem best attempted at the beginning of the day :-)

maliberty, 2023-05-26:
Looking at io_outs_down[0] on M5.  In the block it lies on track with the pin center at 5.58 = (track=116 * x_pitch=0.048  + x_offset=0.012)
[From ```make_tracks M5  -x_offset 0.012 -x_pitch 0.048 -y_offset 0.012 -y_pitch 0.048```]
```
  PIN io_outs_down[0]
    DIRECTION OUTPUT ;
    USE SIGNAL ;
    PORT
      LAYER M5 ;
        RECT  5.568 0 5.592 0.084 ;
```

Looking at ces_0_0/io_outs_down[0] in the top level using your branch from #1097:
[image]

The location comes from :
26.208 = instance_x=20.628 + location_above=5.58

Not that this is not a track multiple.  You can see the offset is built into the pin location so the macro needs to be on a multiple of the track pattern.  For example

26.172 = instance_x=20.592 + location_above=5.58 = track=545 * x_pitch=.048 + x_offset=0.012
[instance_x = track=429 * x_pitch=0.048]

oharboe, 2023-05-26:
After update(stopped it after 5 itreations so it still has some DRC errors, but seems to be converging).

The pins seem to be landing nicely into the macro:

[image]

[image]

osamahammad21, 2023-06-01:
I think it can find a route but always with DRVs, shouldn't we log a warning instead of  an error?

osamahammad21, 2023-06-01:
@maliberty , Ok. The term has multiple ports, should I fire an error if any of them is off grid, or if none of them is on grid?

maliberty, 2023-06-01:
> @maliberty , Ok. The term has multiple ports, should I fire an error if any of them is off grid, or if none of them is on grid?

Anything off the manufacturing grid is a drc error so any

maliberty, 2023-06-01:
@osamahammad21 PA can take a while so it would be better to check it earlier.

osamahammad21, 2023-06-01:
I think I understand how it could not be detected. The Master pin itself is on grid but the macro placement put it off grid. so MPin is on grid but InstPin is not.

maliberty, 2023-06-01:
That's correct - it was a bad macro placement.