The-OpenROAD-Project/OpenROAD #3391: antenna inserted with the wrong orientation

gadfort, 2023-05-26:
### Describe the bug

Antenna failed placement check and looks to be inserted with a rotation.
[image]

### Expected Behavior

Antenna inserted into legal site

### Environment

```shell
Git commit: 575222da2e7b669c11899253b61a897df46a4f20
kernel: Linux 5.15.90.1-microsoft-standard-WSL2
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
cmake version 3.25.0
-- The CXX compiler identification is GNU 11.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-8233-g575222da2
-- System name: Linux
-- Compiler: GNU 11.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Found GTest: /home/pgadfort/.local/lib/cmake/GTest/GTestConfig.cmake (found version "1.13.0")
-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare
[... 118 lines cut ...]
  Pin: _16967_/A1 (sky130_fd_sc_hd__a21oi_1)
    Layer: met2
      Partial area ratio: 1698.28
      Required ratio:  400.00 (Side area) (VIOLATED)

[INFO ANT-0002] Found 35 net violations.
[INFO ANT-0001] Found 48 pin violations.
[INFO GRT-0006] Repairing antennas, iteration 1.
[INFO GRT-0043] No OR_DEFAULT vias defined.
[WARNING ANT-0009] Net core.clk requires more than 10 diodes per gate to repair violations.
[INFO GRT-0012] Found 36 antenna violations.
[INFO GRT-0015] Inserted 101 diodes.
[INFO GRT-0054] Using detailed placer to place 32 diodes.
[INFO GRT-0009] rerouting 43 nets.
[INFO GRT-0001] Minimum degree: 2
[INFO GRT-0002] Maximum degree: 239
[INFO GRT-0006] Repairing antennas, iteration 2.
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0012] Found 0 antenna violations.
[INFO DPL-0001] Placed 255954 filler instances.
[WARNING DPL-0006] Site aligned check failed (1).
 ANTENNA_55
[ERROR DPL-0033] detailed placement checks failed.
Error: sc_route.tcl, 96 DPL-0033
```

eder-matheus, 2023-05-30:
@gadfort Could you try this: https://github.com/The-OpenROAD-Project/OpenROAD/pull/3402 ?

gadfort, 2023-05-30:
> @gadfort Could you try this: #3402 ?

@eder-matheus it appears to be working. Thanks.

closed 2023-05-30 by maliberty