The-OpenROAD-Project/OpenROAD #3393: Crash in global placement

oharboe, 2023-05-27:
### Describe the bug

Create settings.mk locally in the flow folder:

```
export MOCK_ARRAY_WIDTH=8
export MOCK_ARRAY_HEIGHT=8
export MOCK_ARRAY_DATAWIDTH=8
export MOCK_ARRAY_PITCH_SCALE=20
export DESIGN_CONFIG=designs/asap7/mock-array-big/config.mk
export FLOW_VARIANT=8x8x8x20 
```

Run:

```
make
```

Output:

```
[INFO-FLOW] ASU ASAP7 - version 2
Default PVT selection: BC
[INFO][FLOW] Invoked hierarchical flow.
Block Element needs to be hardened.
(/usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -exit -no_init  ./scripts/global_place_skip_io.tcl -metrics ./logs/asap7/mock-array-big/8x8x8x20/3_1_place_gp_skip_io.json) 2>&1 | tee ./logs/asap7/mock-array-big/8x8x8x20/3_1_place_gp_skip_io.log
OpenROAD v2.0-8226-gf1293afb7 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO GPL-0002] DBU: 1000
[deleted]
[NesterovSolve] Iter: 1330 overflow: 0.100011 HPWL: 55837063
[NesterovSolve] Finished with Overflow: 0.099936
double free or corruption (!prev)
Signal 6 received
Stack trace:
```

### Expected Behavior

no crash

### Environment

```shell
Git commit: 9a713f0e8a51769207918da28adfd3c02b854375
kernel: Linux 5.15.0-72-generic
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
cmake version 3.22.1
```

### To Reproduce

[crash.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/11580666/crash.zip)

oharboe, 2023-05-27:
Rebuilt with latest master, got slightly different output:

```
[NesterovSolve] Iter: 1320 overflow: 0.100842 HPWL: 55901981
[NesterovSolve] Iter: 1330 overflow: 0.100083 HPWL: 56045124
[NesterovSolve] Finished with Overflow: 0.099933
double free or corruption (!prev)
Signal 6 received
Stack trace:
```

oharboe, 2023-05-27:
Stack trace:

```
futex_wait (@__lll_lock_wait_private:17)
__GI___lll_lock_wait_private (@__lll_lock_wait_private:12)
__GI___libc_malloc (@__libc_malloc:190)
operator new(unsigned long) (@operator new(unsigned long):11)
__gnu_cxx::new_allocator<char>::allocate(unsigned long, void const*) (/usr/include/c++/11/ext/new_allocator.h:127)
std::allocator_traits<std::allocator<char>>::allocate(std::allocator<char>&, unsigned long) (/usr/include/c++/11/bits/alloc_traits.h:464)
std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>>::_M_create(unsigned long&, unsigned long) (/usr/include/c++/11/bits/basic_string.tcc:153)
std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>>::reserve(unsigned long) (/usr/include/c++/11/bits/basic_string.tcc:291)
handler(int) (/home/oyvind/OpenROAD-flow-scripts/dependencies/include/boost/stacktrace/detail/frame_unwind.ipp:65)
__restore_rt (@__restore_rt:3)
__pthread_kill_implementation (@pthread_kill@@GLIBC_2.34:81)
__pthread_kill_internal (@pthread_kill@@GLIBC_2.34:59)
__GI___pthread_kill (@pthread_kill@@GLIBC_2.34:59)
__GI_raise (@raise:10)
__GI_abort (@abort:46)
__libc_message (@__libc_message:179)
malloc_printerr (@7f6c2f0ecd7c..7f6c2f0ecde5:3)
_int_free (@_int_free:476)
__GI___libc_free (@free:31)
gpl::FFT::~FFT() (/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/gpl/src/fft.cpp:70)
std::default_delete<gpl::FFT>::operator()(gpl::FFT*) const (/usr/include/c++/11/bits/unique_ptr.h:85)
std::unique_ptr<gpl::FFT, std::default_delete<gpl::FFT>>::~unique_ptr() (/usr/include/c++/11/bits/unique_ptr.h:361)
gpl::NesterovBase::~NesterovBase() (/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/gpl/src/nesterovBase.cpp:1016)
std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() (/usr/include/c++/11/bits/shared_ptr_base.h:168)
std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() (/usr/include/c++/11/bits/shared_ptr_base.h:161)
std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count() (/usr/include/c++/11/bits/shared_ptr_base.h:705)
std::__shared_ptr<gpl::NesterovBase, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr() (/usr/include/c++/11/bits/shared_ptr_base.h:1154)
std::shared_ptr<gpl::NesterovBase>::~shared_ptr() (/usr/include/c++/11/bits/shared_ptr.h:122)
gpl::RouteBase::~RouteBase() (/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/gpl/src/routeBase.cpp:329)
std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() (/usr/include/c++/11/bits/shared_ptr_base.h:168)
std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() (/usr/include/c++/11/bits/shared_ptr_base.h:161)
std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count() (/usr/include/c++/11/bits/shared_ptr_base.h:705)
std::__shared_ptr<gpl::RouteBase, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr() (/usr/include/c++/11/bits/shared_ptr_base.h:1154)
std::__shared_ptr<gpl::RouteBase, (__gnu_cxx::_Lock_policy)2>::reset() (/usr/include/c++/11/bits/shared_ptr_base.h:1272)
gpl::Replace::reset() (/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/gpl/src/replace.cpp:128)
_wrap_replace_reset_cmd (/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/build/src/gpl/CMakeFiles/gpl.dir/replaceTCL_wrap.cxx:1780)
TclNRRunCallbacks (@TclNRRunCallbacks:37)
___lldb_unnamed_symbol1504 (@___lldb_unnamed_symbol1504:315)
Tcl_EvalEx (@Tcl_EvalEx:9)
Tcl_Eval (@Tcl_Eval:11)
sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) (/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/sta/app/StaMain.cc:95)
ord::tclAppInit(Tcl_Interp*) (/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/Main.cc:418)
Tcl_MainEx (@Tcl_MainEx:111)
main (/home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/Main.cc:297)
__libc_start_call_main (@__libc_start_call_main:29)
__libc_start_main_impl (@__libc_start_main@@GLIBC_2.34:43)
_start (@_start:15)
```

oharboe, 2023-05-28:
With sanitizer:

```
$ git diff
diff --git a/build_openroad.sh b/build_openroad.sh
index d2af3b56..c84d5090 100755
--- a/build_openroad.sh
+++ b/build_openroad.sh
@@ -275,7 +275,8 @@ __local_build()
         ${NICE} make install -C tools/yosys -j "${PROC}" ${YOSYS_ARGS}
 
         echo "[INFO FLW-0018] Compiling OpenROAD."
-        ${NICE} cmake tools/OpenROAD -B tools/OpenROAD/build ${OPENROAD_APP_ARGS}
+        ${NICE} cmake tools/OpenROAD -B tools/OpenROAD/build ${OPENROAD_APP_ARGS} -DCMAKE_BUILD_TYPE=RELEASE -DCMAKE_CXX_FLAGS="-g -fsanitize=address -fsanitize=undefined"
+
         ${NICE} cmake --build tools/OpenROAD/build --target install -j "${PROC}"
 
         if [ ! -z "${LSORACLE_ENABLE+x}" ]; then
```

```
[INFO GPL-0030] NumBins: 4194304
=================================================================
==34796==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x6250000db900 at pc 0x5641b6c73c95 bp 0x7ffc200e8220 sp 0x7ffc200e8210
WRITE of size 4 at 0x6250000db900 thread T0
    #0 0x5641b6c73c94 in gpl::FFT::updateDensity(int, int, float) /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/gpl/src/fft.cpp:133
    #1 0x5641b6ba7069 in gpl::NesterovBase::updateDensityForceBin() /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/gpl/src/nesterovBase.cpp:1868
    #2 0x5641b6a53ac1 in gpl::NesterovPlace::init() /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/gpl/src/nesterovPlace.cpp:178
    #3 0x5641b6a6396f in gpl::NesterovPlace::NesterovPlace(gpl::NesterovPlaceVars const&, std::shared_ptr<gpl::PlacerBase>, std::shared_ptr<gpl::NesterovBase>, std::shared_ptr<gpl::RouteBase>, std::shared_ptr<gpl::TimingBase>, utl::Logger*) /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/gpl/src/nesterovPlace.cpp:132
    #4 0x5641b69b1e32 in gpl::Replace::initNesterovPlace() /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/gpl/src/replace.cpp:330
    #5 0x5641b69be198 in gpl::Replace::doNesterovPlace(int) /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/gpl/src/replace.cpp:338
    #6 0x5641b6dc0026 in replace_nesterov_place_cmd() /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/build/src/gpl/CMakeFiles/gpl.dir/replaceTCL_wrap.cxx:1794
    #7 0x5641b6dc0026 in _wrap_replace_nesterov_place_cmd /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/build/src/gpl/CMakeFiles/gpl.dir/replaceTCL_wrap.cxx:2240
    #8 0x7f9b1be9dd31 in TclNRRunCallbacks (/lib/x86_64-linux-gnu/libtcl8.6.so+0x3cd31)
    #9 0x7f9b1be9f027  (/lib/x86_64-linux-gnu/libtcl8.6.so+0x3e027)
    #10 0x7f9b1be9eaa6 in Tcl_EvalEx (/lib/x86_64-linux-gnu/libtcl8.6.so+0x3daa6)
    #11 0x7f9b1be9fc99 in Tcl_Eval (/lib/x86_64-linux-gnu/libtcl8.6.so+0x3ec99)
    #12 0x5641b7e8e1d9 in sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/sta/app/StaMain.cc:95
    #13 0x5641b6596dee in tclAppInit /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/Main.cc:418
    #14 0x7f9b1bf51b18 in Tcl_MainEx (/lib/x86_64-linux-gnu/libtcl8.6.so+0xf0b18)
    #15 0x5641b6369397 in main /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/Main.cc:297
    #16 0x7f9b179b8d8f in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58
    #17 0x7f9b179b8e3f in __libc_start_main_impl ../csu/libc-start.c:392
    #18 0x5641b6555e64 in _start (/home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad+0x2ee83e64)

0x6250000db900 is located 0 bytes to the right of 8192-byte region [0x6250000d9900,0x6250000db900)
allocated by thread T0 here:
    #0 0x7f9b1c0c7337 in operator new[](unsigned long) ../../../../src/libsanitizer/asan/asan_new_delete.cpp:102
    #1 0x5641b6c77c7c in gpl::FFT::init() /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/gpl/src/fft.cpp:97

SUMMARY: AddressSanitizer: heap-buffer-overflow /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/gpl/src/fft.cpp:133 in gpl::FFT::updateDensity(int, int, float)
Shadow bytes around the buggy address:
  0x0c4a800136d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c4a800136e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c4a800136f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c4a80013700: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c4a80013710: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x0c4a80013720:[fa]fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c4a80013730: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c4a80013740: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c4a80013750: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c4a80013760: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c4a80013770: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==34796==ABORTING
Command exited with non-zero status 1
Elapsed time: 0:10.62[h:]min:sec. CPU time: user 9.35 sys 1.22 (99%). Peak memory: 3551376KB.
make: *** [Makefile:513: results/asap7/mock-array-big/8x8x8x20/3_1_place_gp_skip_io.odb] Error 1
```

oharboe, 2023-05-28:
Debug session:

```
[INFO][FLOW] Using platform directory ./platforms/asap7
[INFO-FLOW] ASU ASAP7 - version 2
Default PVT selection: BC
[INFO][FLOW] Invoked hierarchical flow.
Block Element needs to be hardened.
(/usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -exit -no_init  ./scripts/global_place_skip_io.tcl -metrics ./logs/asap7/mock-array-big/8x8x8x20/3_1_place_gp_skip_io.json) 2>&1 | tee ./logs/asap7/mock-array-big/8x8x8x20/3_1_place_gp_skip_io.log
OpenROAD v2.0-8236-g9a713f0e8 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO GPL-0002] DBU: 1000
[INFO GPL-0003] SiteSize: 54 270
[INFO GPL-0004] CoreAreaLxLy: 2160 2160
[INFO GPL-0005] CoreAreaUxUy: 1360530 1360530
[INFO GPL-0006] NumInstances: 66369
[INFO GPL-0007] NumPlaceInstances: 1
[INFO GPL-0008] NumFixedInstances: 64
[INFO GPL-0009] NumDummyInstances: 66304
[INFO GPL-0010] NumNets: 2819
[INFO GPL-0011] NumPins: 5121
[INFO GPL-0012] DieAreaLxLy: 0 0
[INFO GPL-0013] DieAreaUxUy: 1362720 1362720
[INFO GPL-0014] CoreAreaLxLy: 2160 2160
[INFO GPL-0015] CoreAreaUxUy: 1360530 1360530
[INFO GPL-0016] CoreArea: 1845169056900
[INFO GPL-0017] NonPlaceInstsArea: 1435567372380
[INFO GPL-0018] PlaceInstsArea: 102060
[INFO GPL-0019] Util(%): 0.00
[INFO GPL-0020] StdInstsArea: 102060
[INFO GPL-0021] MacroInstsArea: 0
[INFO GPL-0031] FillerInit: NumGCells: 1204002
[INFO GPL-0032] FillerInit: NumGNets: 2819
[INFO GPL-0033] FillerInit: NumGPins: 5121
[INFO GPL-0023] TargetDensity: 0.30
[INFO GPL-0024] AveragePlaceInstArea: 102060
[INFO GPL-0025] IdealBinArea: 340200
[INFO GPL-0026] IdealBinCnt: 5423777
[INFO GPL-0027] TotalBinArea: 1845169056900
[INFO GPL-0028] BinCnt: 2048 2048
[INFO GPL-0029] BinSize: 664 664
[INFO GPL-0030] NumBins: 4194304
=================================================================
==46137==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x6250000db900 at pc 0x55eff3ff2b65 bp 0x7ffd0e9e0270 sp 0x7ffd0e9e0260
WRITE of size 4 at 0x6250000db900 thread T0
    #0 0x55eff3ff2b64 in gpl::FFT::updateDensity(int, int, float) /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/gpl/src/fft.cpp:133
    #1 0x55eff3f7e0b2 in gpl::NesterovBase::updateDensityForceBin() /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/gpl/src/nesterovBase.cpp:1868
    #2 0x55eff3ea0210 in gpl::NesterovPlace::init() /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/gpl/src/nesterovPlace.cpp:178
    #3 0x55eff3e9cc7a in gpl::NesterovPlace::NesterovPlace(gpl::NesterovPlaceVars const&, std::shared_ptr<gpl::PlacerBase>, std::shared_ptr<gpl::NesterovBase>, std::shared_ptr<gpl::RouteBase>, std::shared_ptr<gpl::TimingBase>, utl::Logger*) /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/gpl/src/nesterovPlace.cpp:132
    #4 0x55eff3e4ff06 in gpl::Replace::initNesterovPlace() /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/gpl/src/replace.cpp:330
    #5 0x55eff3e5033b in gpl::Replace::doNesterovPlace(int) /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/gpl/src/replace.cpp:338
    #6 0x55eff40b7845 in replace_nesterov_place_cmd() /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/build/src/gpl/CMakeFiles/gpl.dir/replaceTCL_wrap.cxx:1794
    #7 0x55eff40b985c in _wrap_replace_nesterov_place_cmd /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/build/src/gpl/CMakeFiles/gpl.dir/replaceTCL_wrap.cxx:2240
    #8 0x7f1897caed31 in TclNRRunCallbacks (/lib/x86_64-linux-gnu/libtcl8.6.so+0x3cd31)
    #9 0x7f1897cb0027  (/lib/x86_64-linux-gnu/libtcl8.6.so+0x3e027)
    #10 0x7f1897cafaa6 in Tcl_EvalEx (/lib/x86_64-linux-gnu/libtcl8.6.so+0x3daa6)
    #11 0x7f1897cb0c99 in Tcl_Eval (/lib/x86_64-linux-gnu/libtcl8.6.so+0x3ec99)
    #12 0x55eff4a3d2e4 in sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/sta/app/StaMain.cc:95
    #13 0x55eff3b8e1dd in tclAppInit /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/Main.cc:418
    #14 0x55eff3b8e9ad in ord::tclAppInit(Tcl_Interp*) /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/Main.cc:446
    #15 0x7f1897d62b18 in Tcl_MainEx (/lib/x86_64-linux-gnu/libtcl8.6.so+0xf0b18)
    #16 0x55eff3b8ce60 in main /home/oyvind/OpenROAD-flow-scripts/tools/OpenROAD/src/Main.cc:297
    #17 0x7f18937c9d8f in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58
    #18 0x7f18937c9e3f in __libc_start_main_impl ../csu/libc-start.c:392
    #19 0x55eff3b58ff4 in _start (/home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad+0x21a27ff4)

0x6250000db900 is located 0 bytes to the right of 8192-byte region [0x6250000d9900,0x6250000db900)
allocated by thread T0 here:
    #0 0x7f1897ed8337 in operator new[](unsigned long) ../../../../src/libsanitizer/asan/asan_new_delete.cpp:102
[... 30 lines cut ...]
  0x0c4a80013770: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==46137==ABORTING
```

[image]

oharboe, 2023-05-28:
@maliberty I hope some of what I found above makes this a bit more actionable... I don't know this code, so I'm running out of ideas....

maliberty, 2023-05-28:
I tried crash.zip but I get:
```
Error: read_liberty.tcl, 25 cannot read file ./results/asap7/mock-array-big_Element/8x8x8x20/Element.lib.
```

oharboe, 2023-05-28:
> I tried crash.zip but I get:
> 
> ```
> Error: read_liberty.tcl, 25 cannot read file ./results/asap7/mock-array-big_Element/8x8x8x20/Element.lib.
> ```

Yes, sorry. You have to build locally per instructions

PR #3403 (2023-05-30): gpl: simply and correct BinGrid::initBins
Fixes #3393
commit 2d0d049c (2023-05-30): gpl: simply and correct BinGrid::initBins

Fixes #3393
closed 2023-05-30 by maliberty