The-OpenROAD-Project/OpenROAD #3394: PDN insertion takes hours for 1300x1300um^2

oharboe, 2023-05-29:
### Description

To reproduce, create settings.mk:

``` 
export MOCK_ARRAY_WIDTH=8
export MOCK_ARRAY_HEIGHT=8
export MOCK_ARRAY_DATAWIDTH=64
export MOCK_ARRAY_PITCH_SCALE=20
export DESIGN_CONFIG=designs/asap7/mock-array-big/config.mk
```

Then run `make floorplan_info` to see floorplan size:

```
[INFO][FLOW] Using platform directory ./platforms/asap7
[INFO-FLOW] ASU ASAP7 - version 2
Default PVT selection: BC
[INFO][FLOW] Invoked hierarchical flow.
Block Element needs to be hardened.
DIE_AREA=0 0 1362.72 1362.72
CORE_AREA=2.16 2.16 1360.5600000000002 1360.5600000000002
```

Run `make floorplan` and `make elapsed`

```
Log                       Elapsed seconds
[deleted]
2_6_pdn                        10983
[deleted]
```

### Suggested Solution

I think there is an algorithm that does not scale, identify and make it scale a bit better.

oharboe, 2023-05-29:
Profiling:

```
OPENROAD_EXE="perf record -o perf.data ../tools/install/OpenROAD/bin/openroad" make
```

oharboe, 2023-05-29:
I will do a profile run (3 hrs) with https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1097 and post the `perf report` here:

oharboe, 2023-05-29:
Need to verify this first, I could be confusing myself with the sanatize options...

QuantamHD, 2023-05-29:
I would suggest you not run it for three hours. That's going to create a huge sample file. It's often better to just run it for a few minutes the exit the program to see where the bottle neck is

oharboe, 2023-05-29:
> I would suggest you not run it for three hours. That's going to create a huge sample file. It's often better to just run it for a few minutes the exit the program to see where the bottle neck is

When I press ctrl-c, I often get a corrupt file, so I will let it simmer and see if I cant have some data...

I have been debugging a crash with sanitizer and it is entirely possible I confused myself and that I was running w sanitizer when it took hours, so closed Issue while I check.

closed 2023-05-30 by oharboe