The-OpenROAD-Project/OpenROAD #3410: Add a GUI option to draw iterm labels

maliberty, 2023-05-31:
### Description

This would be similar to the option to draw instance name labels:
[image]

Instance names:
[image]

### Suggested Solution

Add the option to the display controls and implement the drawing in the layout viewer.

Hypothetical output:
[image]

Iyury1, 2023-06-08:
I would like to work on this.

maliberty, 2023-06-08:
Great!  You can ask questions here as you need.

Iyury1, 2023-06-26:
I added a function to renderThread.cpp called drawItermLabels(), which gets called in exactly the same way as drawInstanceNames().

I replaced the variable ``Rect instance_box = inst->getBBox()->getBox();`` with ``Rect iterm_box = inst_iterm->getBBox();``. I also obtain the name of an iterm through its corresponding mterm, since dbMTerm has a getName() funciton but dbITerm does not.

I left in the same checks as drawInstanceNames() to prevent printing the names when text_height_check > ..bbox_in_px.height(). This caused the iterm labels to not be drawn at any zoom level.

I then commented out those checks, so I expected the labels to always be drawn. The labels are then drawn even when fully zoomed out, and in the correct locations it seems. However as I zoom in, the labels become truncated with ellipses and then disappear entirely. Whereas the instance names are still visible when fully zoomed in.

Any ideas what could cause this problem? Besides the text_height_check I mentioned, what could be preventing the names from being drawn when zoomed in?

https://github.com/Iyury1/OpenROAD/tree/gui-add-iterm-labels

I will keep reading the code to figure this out.

rovinski, 2023-06-26:
@Iyury1 you could probably just open a draft pull request and ask for feedback there.

PR #3528 (2023-06-29): gui: Add iterm labels to layout viewer
Fixes #3410 

I added a function to renderThread.cpp called drawItermLabels(), which gets called in exactly the same way as drawInstanceNames().

I replaced the variable Rect instance_box = inst->getBBox()->getBox(); with Rect iterm_box = inst_iterm->getBBox();. I also obtain the name of an iterm through its corresponding mterm, since dbMTerm has a getName() funciton but dbITerm does not.

I left in the same checks as drawInstanceNames() to prevent printing the names when text_height_check > ..bbox_in_px.height(). This caused the iterm labels to not be dr [... cut]
closed 2023-06-29 by maliberty