The-OpenROAD-Project/OpenROAD #3422: Click on routing congestion and get some information on the cause

oharboe, 2023-06-02:
### Description

We sometimes have problems with routing congestion on top of a macro, from `make gui_cts` below. It is difficult to figure out where this routing congestion comes from. It is usually something simple to fix, like a pin being constrained to the wrong edge.

[image]

### Suggested Solution

In this case, it would be useful to click on the routing congestion to get more information on the matter(buffer tree? net?):

[image]

oharboe, 2023-06-02:
@maliberty @eder-matheus @luis201420 FYI

eder-matheus, 2023-06-02:
@oharboe I've added the nets crossing the congested areas into the congestion report. Could you check it here: https://github.com/The-OpenROAD-Project/OpenROAD/pull/3424?

oharboe, 2023-06-02:
> @oharboe I've added the nets crossing the congested areas into the congestion report. Could you check it here: #3424?

I wont be able to test this before you merge it with master, I think, but I will definitely use this feature!

eder-matheus, 2023-06-02:
> > @oharboe I've added the nets crossing the congested areas into the congestion report. Could you check it here: #3424?
> 
> I wont be able to test this before you merge it with master, I think, but I will definitely use this feature!

Got it. Here's a screenshot of how it looks:
[image]

eder-matheus, 2023-06-02:
> Sweet! And from this I can navigate to the buffertree to find the terminals, which I can then relate to Verilog?

Yes, I think you can. From the nets, you can go to all the terms connected to it, so I think you can get the information you want.

closed 2023-06-13 by maliberty