The-OpenROAD-Project/OpenROAD #3425: DRT: Rewriting the dbNet in io

osamahammad21, 2023-06-04:
### Describe the bug

When logger debug drt snapshot is on, we update the db Nets after each iteration. But each time, we append to the existing odb net which leads to rewriting the new net segments over the old ones. I ended up with an odb file of 25+GB because of this. 

### Expected Behavior

Every iteration, we should delete the old routed wire of the net and create a new one.

### Environment

```shell
Master branch OR
```

### To Reproduce

Add this to any DRT test case
`set_debug_level DRT snapshot 1`

osamahammad21, 2023-06-04:
@eder-matheus I remember that you had a PR recently on this part but I can't remember what was it. I could fix it myself by just destroying the dbWire each time and creating a new one but I am not sure what implications does this have on your recent PR on this.

closed 2023-06-14 by maliberty

eder-matheus, 2023-06-14:
@osamahammad21 I've made my tests with the regression tests of drt and I see that the final odb is smaller than before my change (actually, it is the same as when I run drt without the `set_debug_level DRT snapshot 1`). Could you check with your testcase?