The-OpenROAD-Project/OpenROAD #3444: DFT: crash when running jpeg_sky130hd.tcl with preview_dft and insert_dft

fgaray, 2023-06-09:
### Describe the bug

When running preview_dft and insert_dft in the design test/jpeg_sky130hd.tcl, I get a crash in Sta:

[image]

### Expected Behavior

Not crashing.

### Environment

```shell
Unknown git commit, this is not a git repository.                                                                                                                                                                                                    [102/4951]

Please make sure that you have the latest code changes and add the commit
hash in the description.

kernel: Linux 6.1.25-1rodete1-amd64
os: Debian GNU/Linux 11 (bullseye)
cmake version 3.24.2
CMake Warning at CMakeLists.txt:96 (message):
  OpenROAD git describe failed, using sha1 instead

-- The CXX compiler identification is GNU 10.2.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: 93fdf28ec5359714c80e27b59c376cea446d64ed
-- System name: Linux
-- Compiler: GNU 10.2.1
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Could NOT find GTest (missing: GTEST_LIBRARY GTEST_INCLUDE_DIR GTEST_MAIN_LIBRARY)
-- The C compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.9 (found version "3.9.2") found components: Interpreter
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
[... 140 lines cut ...]
### Additional Context

I think what is happening is that we need to create a new vertice for the new port in STA's graph.

Usually, makePinVertices is called for every pin that is created in Sta::makeInstanceAfter, but for new created ports at the top level block, we are not calling makePinVertices.

I think we can add the following code in dbStaCbk::inDbBTermCreate:

```
diff --git a/src/dbSta/src/dbSta.cc b/src/dbSta/src/dbSta.cc
index 7f0da42f0..c3f665152 100644
--- a/src/dbSta/src/dbSta.cc
+++ b/src/dbSta/src/dbSta.cc
@@ -563,6 +563,8 @@ void dbStaCbk::inDbBTermPreDisconnect(dbBTerm* bterm)
 void dbStaCbk::inDbBTermCreate(dbBTerm* bterm)
 {
   sta_->getDbNetwork()->makeTopPort(bterm);
+  Pin* pin = network_->dbToSta(bterm);
+  sta_->graph()->makePinVertices(pin);
 }

 void dbStaCbk::inDbBTermDestroy(dbBTerm* bterm)
```

However, I get a crash in read_verilog9.tcl because looks like network_ is not initialized. Are we missing some previous setup?

fgaray, 2023-06-09:
So I gave this a try and now all regressions and the jpeg design pass.

```
diff --git a/src/dbSta/src/dbSta.cc b/src/dbSta/src/dbSta.cc
index 7f0da42f0..fce07d992 100644
--- a/src/dbSta/src/dbSta.cc
+++ b/src/dbSta/src/dbSta.cc
@@ -563,6 +563,10 @@ void dbStaCbk::inDbBTermPreDisconnect(dbBTerm* bterm)
 void dbStaCbk::inDbBTermCreate(dbBTerm* bterm)
 {
   sta_->getDbNetwork()->makeTopPort(bterm);
+  if (sta_->graph()) {
+    Pin* pin = network_->dbToSta(bterm);
+    sta_->graph()->makePinVertices(pin);
+  }
 }

 void dbStaCbk::inDbBTermDestroy(dbBTerm* bterm)
```

I think this code is fine because if there is no graph available, once it is created it will assign a vertice to all the pins in the design.

If @maliberty is fine with this, I can send a PR with the fix.

maliberty, 2023-06-10:
I tried your instructions with OR commit ca2e30f462bc0274bedf0643c1ee8dbc21e2723a but I get no crash.

maliberty, 2023-06-10:
Hmm.  I tried again and it did crash.  I'm not sure what was different...

fgaray, 2023-06-10:
It is crashing with the "fix"?

maliberty, 2023-06-11:
I haven't tried it.  I don't think we should need to create the vertices so I'd like to see what is going on.

maliberty, 2023-06-19:
@fgaray are you blocked by this or can you apply your workaround locally?

fgaray, 2023-06-19:
I can apply the WA locally so no really blocked, but users will encounter this if they try to add DFT to a flow.

openroadie, 2023-06-26:
@fgaray The second fix is legit. Would you like to turn in a PR into OpenROAD?

maliberty, 2023-06-26:
@openroadie why is it necessary for the client to do this?  There is no other circumstance in which the client manipulates the graph.  It seems like a workaround to an sta bug.

openroadie, 2023-06-26:
The client is not doing this. It is the callback and this is a top level terminal. 

The client is not "manipulating" the graph ... it is merely calling the correct API in the graph to add the new terminal. The reason there is no other case where we touch the graph is maybe because we have not tested those scenarios till now? There are plenty of other examples in the dbSta file where the call backs have some code. Is it our position that the callbacks should be all one liners?

maliberty, 2023-06-26:
We have clearly tested creating instances in rsz.  Compare to 
```
void dbStaCbk::inDbInstCreate(dbInst* inst)
{
  sta_->makeInstanceAfter(network_->dbToSta(inst));
}
```
no graph manipulation code exists.  It has nothing to do with function length and is about separation of concerns.  The graph is sta's job to maintain not the clients.  The client should only notify sta of netlist manipulations.

tspyrou, 2023-06-26:
@maliberty @openroadie I think this is an sta bug. I have a 1-1 with Cherry today. Let me discuss with him. The network adapter should keep the network sync'd. I don't think it should touch the graph. It would be OK, I think, if this is urgent to put the fix into the adapter with a comment saying fixme this is a workaround to an sta bug. I will let you know what I find out.

closed 2023-06-29 by maliberty

fgaray, 2023-06-29:
Hi @maliberty @openroadie  , so the fix will not be implemented in sta? or we plan in the future to remove this WA/fix once the one in sta is ready?

maliberty, 2023-06-29:
The fix is in STA - I'm not sure I understand your question.

fgaray, 2023-06-29:
Ah, you are right. Sorry, I didn't realize the PR was merging the fix from STA.

Thanks everyone for working on this!

PR #3548 (2023-06-29): Openroad dft 3444
fixes #3444