The-OpenROAD-Project/OpenROAD #3450: Build error with GCC v13

andreili, 2023-06-11:
### Describe the bug

After lasest updates, I get a compilation error on my system, similar to below:
```
FAILED: src/grt/src/fastroute/CMakeFiles/FastRoute4.1.dir/src/FastRoute.cpp.o 
/usr/bin/x86_64-pc-linux-gnu-g++ -DFMT_SHARED -DSPDLOG_COMPILED_LIB -DSPDLOG_FMT_EXTERNAL -DSPDLOG_SHARED_LIB -I/var/tmp/portage/sci-electronics/openroad-9999/work/openroad-9999/src/grt/src/fastroute/include -I/var/tmp/portage/sci-electronics/openroad-9999/work/openroad-9999/src/grt/src/fastroute/../../include -I/var/tmp/portage/sci-electronics/openroad-9999/work/openroad-9999/src/utl/include -I/var/tmp/portage/sci-electronics/openroad-9999/work/openroad-9999/src/utl/src -I/var/tmp/portage/sci-electronics/openroad-9999/work/openroad-9999/src/stt/include -I/var/tmp/portage/sci-electronics/openroad-9999/work/openroad-9999/src/sta/include -I/var/tmp/portage/sci-electronics/openroad-9999/work/openroad-9999/src/odb/include -I/var/tmp/portage/sci-electronics/openroad-9999/work/openroad-9999/src/odb/include/odb -I/var/tmp/portage/sci-electronics/openroad-9999/work/openroad-9999/src/odb/src/def/def -I/var/tmp/portage/sci-electronics/openroad-9999/work/openroad-9999/src/odb/src/def/defzlib -I/var/tmp/portage/sci-electronics/openroad-9999/work/openroad-9999/src/odb/src/lef/lef -I/var/tmp/portage/sci-electronics/openroad-9999/work/openroad-9999/src/odb/src/lef/lefzlib  -fPIC -DFMT_DEPRECATED_OSTREAM=1 -std=c++17 -Wall -Wredundant-decls -Wformat-security -Wno-sign-compare -Wp,-D_GLIBCXX_ASSERTIONS -MD -MT src/grt/src/fastroute/CMakeFiles/FastRoute4.1.dir/src/FastRoute.cpp.o -MF src/grt/src/fastroute/CMakeFiles/FastRoute4.1.dir/src/FastRoute.cpp.o.d -o src/grt/src/fastroute/CMakeFiles/FastRoute4.1.dir/src/FastRoute.cpp.o -c /var/tmp/portage/sci-electronics/openroad-9999/work/openroad-9999/src/grt/src/fastroute/src/FastRoute.cpp
In file included from /var/tmp/portage/sci-electronics/openroad-9999/work/openroad-9999/src/grt/src/fastroute/include/FastRoute.h:46,
                 from /var/tmp/portage/sci-electronics/openroad-9999/work/openroad-9999/src/grt/src/fastroute/src/FastRoute.cpp:37:
/var/tmp/portage/sci-electronics/openroad-9999/work/openroad-9999/src/grt/src/fastroute/include/DataType.h:43:20: error: ‘int64_t’ in namespace ‘std’ does not name a type
   43 | using int64 = std::int64_t;
      |                    ^~~~~~~
```
To fix it, need to add next line into file "src/grt/src/fastroute/include/DataType.h":
```
#include <cstdint>
```

### Expected Behavior

Successfully build.

### Environment

```shell
Unknown git commit, this is not a git repository.

Please make sure that you have the latest code changes and add the commit
hash in the description.

kernel: Linux 6.1.31-gentoo-x86_64
os: Gentoo 
cmake version 3.26.3
CMake Warning at CMakeLists.txt:99 (message):
  OpenROAD git describe failed, using sha1 instead

-- The CXX compiler identification is GNU 13.1.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: ca2e30f462bc0274bedf0643c1ee8dbc21e2723a
-- System name: Linux
-- Compiler: GNU 13.1.1
-- Build type: 
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Found GTest: /usr/lib64/cmake/GTest/GTestConfig.cmake (found version "1.13.0")  
-- The C compiler identification is GNU 13.1.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
[... 93 lines cut ...]
-- STA version: 2.4.0
-- STA git sha: 555493cba6e476bd2ff0b9a543de7a781276c2b3
-- System name: Linux
-- Compiler: GNU 13.1.1
-- Build type: 
-- Build CXX_FLAGS: 
-- Install prefix: /usr/local
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- TCL library: /usr/lib64/libtcl.so
-- TCL header: /usr/include/tcl.h
-- SSTA: 0
-- STA executable: /var/tmp/portage/sci-electronics/openroad-9999/work/openroad-9999/src/sta/app/sta
-- GPU is not enabled
-- TCL library: /usr/lib64/libtcl.so
-- TCL header: /usr/include/tcl.h
-- Found Protobuf: /usr/lib64/libprotobuf.so (found version "3.21.9") 
-- Found Eigen3: /usr/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0") 
-- GUI is enabled
-- Charts widget is not enabled
-- Configuring incomplete, errors occurred!
```

### To Reproduce

Build latest GIT sources with GCC v13.

QuantamHD, 2023-06-11:
Would you mind sending a pull request with the change?

andreili, 2023-06-12:
> Would you mind sending a pull request with the change?
@QuantamHD Done.

umarcor, 2023-06-30:
Since #3451 was merged, should this be closed as fixed/completed?

QuantamHD, 2023-06-30:
Done.

closed 2023-06-30 by QuantamHD